raspberrypi/linux #3710: loopback example SPI not working

sara-bhy, 2020-07-07:
Hello guys, i'm trying to test a simple loopback example using spidev. the problem is that it's not working, i looked everywhere and i found a lot of similar codes to mine i just don't undrestand why it's not working. Could you help me please.

Here is my code

import spidev
import time

speed = 1000000
bits_per_word = 8
delay = 0
buffer_tx = [1,2,3]
ok = True

#settings SPI
spi=spidev.SpiDev()
spi.open(0,0)
spi.max_speed_hz=speed
spi.mode = 0b00
spi.lsbfirst=False
spi.bits_per_word = bits_per_word
spi.threewire = False
spi.cshigh = False
spi.no_cs = False

#loopback example
try:
while True:
buffer_rx = spi.xfer(buffer_tx)
#buffer_rx = spi.xfer2(buffer_tx)
#buffer_rx = spi.xfer3(buffer_tx)
print ("TX:", buffer_tx)
print ("RX:", buffer_rx)
ok = buffer_rx == buffer_tx
if ok:
print ("Success")
else:
print ("Failed")
finally:
spi.close()

And here is the result if i use spi.xfer() or spi.xfer2()

('TX:', [1L, 1L, 1L])
('RX:', [1L, 1L, 1L])
Success

And the result if i use spi.xfer3()

('TX:', [1, 2, 3])
('RX:', (1L, 2L, 3L))
Failed

P33M, 2020-07-07:
This is a bug tracker for our Linux kernel, not a technical support channel.
Ask your question on the community forum (preferably in the Interfacing section): https://raspberrypi.org/forums

closed 2020-07-07 by P33M

pelwell, 2020-07-07:
Hint: It's a Python types difference (list vs tuple), rather than a data transfer problem, so this is definitely the wrong forum.