raspberrypi/linux #3713: rpi-4.19.y-rt update for Pi 4 8GB

gtrainavicius, 2020-07-07:
Hey, looks like a kernel built based on rpi-4.19.y-rt branch boots, but the USB drivers at least don't work, and users report the system after restart is unable to boot.

Switching to mainline kernel and installing all package updates from apt-get fixes the issue.

Any ideas what needs to be done for rpi-4.19.y-rt branch for it to work well with Pi4 8GB?

timg236, 2020-07-07:
It's probably missing the VL805 changes or the PCI fundamental reset patch
https://lkml.org/lkml/2020/5/5/1003

https://lkml.org/lkml/2020/5/7/1056

gtrainavicius, 2020-07-15:
I've tried applying these changes to 4.19.y-rt branch and they do solve the problem. Are these changes in this source repository's 4.19.y branch? I don't see the same patches applied here.

gtrainavicius, 2020-07-20:
ping :)

pelwell, 2020-07-20:
See c74b1b53254016fd83b580b8d49bb02d72ce4836 and https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/pci/controller/pcie-brcmstb.c#L1416.

gtrainavicius, 2020-07-20:
Ok, thank you, I was looking for the changes in raspberrypi.c as in the previous links. :)

pelwell, 2020-07-20:
The fact that the rpi-4.19.y kernel has working USB on an 8GB Pi4 (R6, rev 1.4) is a big clue that this issue has been solved there.

PR #4046 (2021-01-04): Backport VL805 initialization code for RPi4 from 5.4.y (fix #3713, #3976)
Without this patch RPi4 with 4.19.y-rt has no USB at all.
It should fix #3713 and #3976.

------------------------------------------------------------------------------------------------------

The VL805 FW may either be loaded from an SPI EEPROM or alternatively
loaded directly by the VideoCore firmware. A PCI reset will reset
the VL805 XHCI controller on the Raspberry Pi4 requiring the firmware
to be reloaded if an SPI EEPROM is not present.

Use a VideoCore mailbox to trigger the loading of the VL805
firmware (if nece [... cut]