raspberrypi/linux #3718: 2 USB-ethernet adapters on same subnet at same time do not function correctly

j-cripps, 2020-07-10:
**Describe the bug**
Using two USB-ethernet adapters on the same subnet at the same time causes the two USB-ethernet adapters to malfunction. The two adapters seem to conflate properties and ARP probes get broadcast with incorrect details (with IP addresses advertised on the other ones MAC address). This problem disappears when the two adapters are on a different subnet. The physical interfaces are enumerated using udev and renamed using their physical location on the USB hub chip of the Pi. The network config is performed by dhcpcd and both interfaces are assigned static IPs.

**To reproduce**

1. Connect two usb-ethernet dongles (tested a few different providers -asix, realtek etc) to the Pi
2. Assign them static IPs on the same subnet (172.31.250.x/24 in this case)
3. Try and ping both adapters at the same time from a windows 10 host machine (tried macOS - no difference)
4. Remove one cable and see if both IPs are responsive and on what MAC they respond
5. Reinsert cable and see if both IPs are responsive and on what MAC they respond 

**Expected behaviour**
The IP addresses bound to a specific physical interface should stay constant. 172.31.250.9 bound to MAC: xn and 172.31.250.8 bound to MAC: xx should stay like that. 172.31.250.9 should not start to advertise on MAC: xx.

**Actual behaviour**
The IP addresses assigned to a specific physical interface do not remain constant and sometimes advertise on MAC addresses that they were not bound to. Therefore a network interface can become inaccessible from one machine and both become accessible from another machine 

**System**
* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? Custom buildroot image ver 2019.11.1
* Which firmware version (`vcgencmd version`)? 1.20200601-1 (Unsure if this is what is wanted here)
* Which kernel version (`uname -a`)? 4.19.118-v7

**Logs**
No error messages are produced during this behaviour

pelwell, 2020-07-10:
This sounds like it is behaving correctly (albeit confusingly) - even though that is not what you expected. This article on "ARP Flux" and how to avoid it might help: https://netbeez.net/blog/avoiding-arp-flux-in-multi-interface-linux-hosts/

j-cripps, 2020-07-10:
That does seem to be what's happening, thanks! 

I tried the two sysctl commands listed in that article and on the eth2 interface it stopped the pings going to eth1, but they still don't go to the eth2 interface, it still goes to timeout - not unless eth1 is physically unplugged at which point the pings start responding to eth2 immediately. The ARP probes *almost* behave now apart from a weird behaviour in that the eth1 interface seems to advertise for it e.g eth1 is plugged in, eth2 then gets plugged in, eth2 responds with correct ARP, but eth1 also responds with the ARP that should come from eth2, so eth1 is also saying 'Who has x IP, tell IP from eth2'.

pelwell, 2020-07-10:
What you're discovering is that IP networking wasn't designed to be used with multiple interfaces on the same subnet. In particular there's a strange duality where an IP address names an interface, but packets delivered to the interface are actually just delivered to the host - it doesn't really matter they came from. If one of those packets requires a reply, only the destination address is used in working out how to send it back - the source address doesn't control the interface it is sent from.

What's the output from `ip route` (or `route -n`)? The metric indicates the "cost" associated with a route - lower values are preferred.

j-cripps, 2020-07-10:
Right I see, the output of `ip route` is the 'correctly (eth 1)' functioning interface has metric of 202 whereas the 'faulty (eth2)' interface has metric of 204. So if kernel prefers lower route metric and I stopped the cross-map behaviour with the sysctl command, that explains the behaviour of the higher metric interface not responding correctly until the lower metric interface is removed. 

So is it generally good practice to have multiple interfaces as I have in this situation always on different subnets, if I want the behaviour of 'this packet from the host *must* go to this specific interface'

pelwell, 2020-07-10:
Networking has changed a lot since I started in this industry _mumble_ years ago, but as far as I know that is still the only way to specify the interface to use on a packet-by-packet basis with no chance of it going wrong.

j-cripps, 2020-07-10:
Great, thanks for the help!

6by9, 2020-07-10:
Generally it's easier to bind multiple addresses to one interface than have multiple interfaces each with independent addresses.

pelwell, 2020-07-10:
Does that help you if you have two interfaces?

6by9, 2020-07-10:
Not really. One interface on the subnet with all desired addresses bound to it will work, whereas multiple interfaces each with addresses on the same subnet doesn't. The latter doesn't add anything either, except possibly greater bandwidth. Bonding links is a better approach for increasing bandwidth than multiple independent interfaces, but that requires a fair networking knowledge and compatible adapters to achieve.

j-cripps, 2020-07-10:
The two physically separate NICs is a semi-hard requirement of the project so looks like different subnets is the way to go in this case. Not sure getting them to work on the same subnet is worth the time and effort - or even possible - as adding a switch to split the packets isn't possible in this project.

Thanks for the help though! I'll close this out then unless there's anything else you want to add.

closed 2020-07-10 by j-cripps