raspberrypi/linux #3720: System crash after some heavy usb serial usage

Tangerino, 2020-07-13:
**Describe the bug**
System crashes while using ttyUSB0

**To reproduce**
Brand new SD card with Ubuntu Focal 20.04 LTS

All packages up to date

This is my test program
```python
import minimalmodbus

instrument = minimalmodbus.Instrument('/dev/ttyUSB0', 2)
instrument.debug = False
instrument.serial.baudrate = 9600
instrument.serial.timeout = 1
while True:
    instrument.read_long(229)
```

**Expected behaviour**
Keep running forever

**Actual behaviour**
The systems freeze. I was able to verify some behavior only using a terminal and keyboard

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi3B V2.1
* Which OS and version (`cat /etc/rpi-issue`)?
Ubuntu Focal 20.04 LTS

* Which firmware version (`vcgencmd version`)?

* Which kernel version (`uname -a`)?
linux Ubuntu 5.4.0-1013-raspi #13

DTDI USB driver for /dev/ttyUSB0

**Logs**
First error was:

[image]

Second error,  after few seconds

[image]

pelwell, 2020-07-14:
There was a firmware released yesterday - available using "sudo rpi-update" - which fixed a potential problem with USB interrupts. You can update the firmware and leave the kernel if you use "sudo SKIP_KERNEL=1 rpi-update". Alternatively, download the files `start*.elf` and `fixup*.dat` from here: https://github.com/raspberrypi/firmware/tree/master/boot

juergh, 2020-07-14:
This is reported against the Ubuntu kernel so should be reported in launchpad. @Tangerino Can you run 'apport-bug linux' on the affected system?

Tangerino, 2020-07-14:
The system does not exist anymore. I'll rebuild the device and test it again.

Tangerino, 2020-07-14:
We are talking about Ubuntu here, right?

pelwell, 2020-07-14:
1. I thought you would know that, of all people.
2. What @juergh said.
3. Ubuntu don't build VPU firmware, hence my suggestion and the SKIP_KERNL=1 suggestion.

juergh, 2020-07-14:
@Tangerino rpi-update is not available on Ubuntu and even if it were it wouldn't work. Ubuntu uses flash-kernel (and u-boot) which wants the firmware in /boot/firmware and not /boot. I've confirmed the problem (on Ubuntu) and there is already another bug that looks similar/identical: https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+bug/1880388
Let's take the Ubuntu discussion there. Once I've confirmed that the problem is reproducible with Raspbian we can discuss that here.

pelwell, 2020-07-14:
rpi-update is only a curl or wget away:
```
$ sudo curl -L --output /usr/bin/rpi-update https://raw.githubusercontent.com/Hexxeh/rpi-update/master/rpi-update && sudo chmod +x /usr/bin/rpi-update
```
As the [documentation](https://github.com/Hexxeh/rpi-update#root_path-and-boot_path) says:
> ROOT_PATH and BOOT_PATH
>
> sudo ROOT_PATH=/media/root BOOT_PATH=/media/boot rpi-update
>
> Allows you to perform an "offline" update, ie update firmware on an SD card you are not currently booted from. Useful for installing firmware/kernel to a non-RPI customised image. Be careful, you must specify both options or neither. Specifying only one will not work.

juergh, 2020-07-14:
@pelwell Ah, nice. Thanks for the info!

Tangerino, 2020-07-14:
Thank you for the info!
The update is done, the test ran for a while and got his message and then freezes.
[image]

This happens in the console while I was installing other stuff, increasing the CPU usage.

Not sure where to go now.

A piece of additional information is that Raspbian works fine in a number of os devices 24x7.

Cheers

Tangerino, 2020-07-14:
Second try
[image]

Now with no other terminal in parallel.

juergh, 2020-07-15:
I can reproduce this on Ubuntu but not on the Raspberry Pi OS, both running 5.4 kernels and the same firmware version. @pelwell I think you can close this ticket as Raspberry Pi OS doesn't seem to be affected. However, it looks like it's a config option that we have (or not) set on Ubuntu. Do you have any idea which areas I should be focusing on trying to track this down?

pelwell, 2020-07-15:
In rpi-5.4.y we switched from bcm2835-cpufreq to the upstream raspberrypi-cpufreq, on all Pis except those using BCM2835, i.e.
```
CONFIG_CPUFREQ_DT=y
CONFIG_ARM_RASPBERRYPI_CPUFREQ=y
# CONFIG_ARM_BCM2835_CPUFREQ is not set
```

Tangerino, 2020-07-17:
Any action for me here?

juergh, 2020-07-19:
It turns out that the preemption model makes a difference. Raspberry Pi OS uses 'Low-latency Desktop' whereas Ubuntu uses 'Desktop'. If I switch Ubuntu to 'Low-level Desktop' the problem goes away. If I switch the Raspberry Pi OS kernel to 'Desktop' it fails as well. I cannot reproduce this on arm 32-bit.

It looks like something's not right with the calls of local_fiq_{enable,disable} in dwc_otg_read_common_intr(). These calls are noops on arm 32-bit which would explain why the problem doesn't show there.

FWIW, the current Raspberry Pi OS 5.4.51-v8+ kernel spits out the following during boot after enabling DEBUG_ATOMIC_SLEEP. IMO that's a problem that needs fixing.

Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.726241] BUG: sleeping function called from invalid context at kernel/irq/manage.c:137
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.726539] dwc_otg: FIQ enabled
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.733318] in_atomic(): 1, irqs_disabled(): 128, non_block: 0, pid: 0, name: swapper/2
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.733322] Preemption disabled at:
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.733340] [<ffffffc010098c5c>] secondary_start_kernel+0xc4/0x180
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.733357] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.51-v8+ #48
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.740371] dwc_otg: NAK holdoff enabled
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.747486] Hardware name: Raspberry Pi 3 Model B Plus Rev 1.3 (DT)
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.747491] Call trace:
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.747501]  dump_backtrace+0x0/0x148
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.747509]  show_stack+0x24/0x30
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.747523]  dump_stack+0xd8/0x144
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.754599] dwc_otg: FIQ split-transaction FSM enabled
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.761730]  ___might_sleep+0x158/0x188
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.761738]  __might_sleep+0x58/0x90
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.761746]  synchronize_irq+0x54/0xd0
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.761755]  disable_irq+0x38/0x48
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.768983] Module dwc_common_port init
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.776156]  local_fiq_disable+0x20/0x28
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.776166]  dwc_otg_handle_common_intr+0x498/0xdc0
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.776175]  dwc_otg_common_irq+0x20/0x38
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.776184]  __handle_irq_event_percpu+0x78/0x2c0
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.878963]  handle_irq_event_percpu+0x3c/0x98
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.885348]  handle_irq_event+0x4c/0xd0
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.891709]  handle_level_irq+0xe4/0x170
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.898036]  generic_handle_irq+0x34/0x50
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.904345]  bcm2836_chained_handle_irq+0x50/0x70
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.910642]  generic_handle_irq+0x34/0x50
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.916993]  __handle_domain_irq+0x98/0x108
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.923398]  bcm2836_arm_irqchip_handle_irq+0x68/0xd0
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.929802]  el1_irq+0xbc/0x180
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.936144]  arch_cpu_idle+0x34/0x220
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.942471]  default_idle_call+0x24/0x48
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.948775]  do_idle+0x210/0x258
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.955015]  cpu_startup_entry+0x28/0x48
Jul 17 11:34:16 rpi-3b-plus-rev1d3-abfb kernel: [    2.961217]  secondary_start_kernel+0x148/0x180

And with the 'Desktop' preemption model and PROVE_LOCKING enabled I also get:

[    4.937147] ================================
[    4.948024] WARNING: inconsistent lock state
[    4.958927] 5.4.51-v8+ #46 Not tainted
[    4.969265] --------------------------------
[    4.980101] inconsistent {IN-HARDIRQ-W} -> {HARDIRQ-ON-W} usage.
[    4.992707] kworker/1:1/31 [HC0[0]:SC0[0]:HE1:SE1] takes:
[    5.004697] ffffff80347eb698 (&((spinlock_t *)hcd->lock)->rlock){?.-.}, at: DWC_SPINLOCK_IRQSAVE+0x24/0x38
[    5.021160] {IN-HARDIRQ-W} state was registered at:
[    5.032747]   lock_acquire+0xd0/0x210
[    5.043016]   _raw_spin_lock+0x3c/0x50
[    5.053313]   DWC_SPINLOCK+0x20/0x30
[    5.063365]   dwc_otg_handle_common_intr+0x5c/0xdc0
[    5.074726]   dwc_otg_common_irq+0x20/0x38
[    5.085257]   __handle_irq_event_percpu+0xb8/0x368
[    5.096493]   handle_irq_event_percpu+0x3c/0x98
[    5.107430]   handle_irq_event+0x4c/0xd0
[    5.117690]   handle_level_irq+0xcc/0x148
[    5.127971]   generic_handle_irq+0x34/0x50
[    5.138310]   bcm2836_chained_handle_irq+0x50/0x70
[    5.149383]   generic_handle_irq+0x34/0x50
[    5.159733]   __handle_domain_irq+0x6c/0xc0
[    5.170168]   bcm2836_arm_irqchip_handle_irq+0x74/0xe0
[    5.181599]   el1_irq+0xbc/0x140
[    5.191072]   arch_cpu_idle+0x40/0x1c8
[... cut ...]
[    5.806739]  process_one_work+0x284/0x5f8
[    5.815630]  worker_thread+0x50/0x428
[    5.824032]  kthread+0x104/0x130
[    5.831959]  ret_from_fork+0x10/0x18

This is probably a fallout of the above atomic sleep problem and eventually kills the machine.

The above are all reported during boot before any real tests are exercised. To reproduce the hang: Connect a USB/serial adapter to the Pi, run 'minicom -D /dev/ttyUSB0' and send data from another device. With 'Desktop' preemption, the Pi hangs within seconds.

juergh, 2020-07-22:
FWIW, I can get rid of the above warnings when using dwc_otg.fiq_enable=0 and dwc_otg.fiq_fsm_enable=0 and the machine stays up even with 'Desktop' preemption.

pelwell, 2020-07-22:
It's probably worth trying `dtoverlay=dwc2` - this selects the upstream (non-FIQ-accelerated) USB driver.

juergh, 2020-07-24:
dwc2 works as well. What's the difference between upstream dwc2 and Raspberry Pi's dwcotg? FIQ acceleration only?

pelwell, 2020-07-24:
dwc2 has support for gadget mode and benefits from upstream maintenance and bug fixing, dwc_otg has FIQ acceleration.

Tangerino, 2020-07-29:
After:
```
apt update
apt upgrade
```
and add to the file `/boot/firmware/usercfg.txt`
```
dtoverlay=dwc2
```

The tests are running for over 12 hours