raspberrypi/linux #3721: irsend not working - "hardware not supported"

sababb81, 2020-07-13:
Hi and firstly apologies for any points that may have been covered elsewhere - I'm new to Raspberry Pi and linux and am in need of some guidance please.

My setup is as follows:
- Raspberry Pi Model B Rev 2
- Raspbian 10 (Buster) OS (32-bit), Release 3.3 – May 2020
- Linux Raspberry Pi Kernel 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux
- Energenie Raspberry Pi infrared add on board

I am ultimately looking to use the Pi to control an air conditioning unit, however, am attempting to use it to control a TV in the first instance.

To date I have done the following:
- Installed lirc (using APT)
- Edited /boot/config.txt to enable the gpio-ir module.  I am using "dtoverlay=gpio-ir,gpio_out_pin=17,gpio_in_pin=18,gpio_in_pull=up"
- Edited /etc/lirc/lirc_options.conf with "driver = default" and "device = /dev/lirc0"

This has allowed me to successfully receive signals from a remote control.  I've downloaded an lircd.conf file for a TV I am testing on, and with irw get an accurate description of the buttons that are being pressed.

It is at this point, all goes wrong when I attempt to use irsend to send commands to my TV.  To attempt to turn the volume up, I am using the following command - irsend SEND_ONCE Sony_RM-ED035 KEY_VOLUMEUP

This generates the following error messages - "hardware does not support sending" and "Error running command: Input/output error"

I have read so many articles referring to this, in general it appears to be a kernel issue with many possible workarounds and suggestions, none of which have worked for me to date.  Any help would be gratefully appreciated

Thank you in advance

JamesH65, 2020-07-13:
This would appear to need technical support, rather than be reporting a bug. For technical support queries please use the Raspberry pi forums. www.raspberrypi.org/forums

pelwell, 2020-07-13:
It might just be a typo in your report, but the parameters for the gpio-ir overlay are wrong:
```
pi@raspberrypi:~$ dtoverlay -h gpio-ir
Name:   gpio-ir

Info:   Use GPIO pin as rc-core style infrared receiver input. The rc-core-
        based gpio_ir_recv driver maps received keys directly to a
        /dev/input/event* device, all decoding is done by the kernel - LIRC is
        not required! The key mapping and other decoding parameters can be
        configured by "ir-keytable" tool.

Usage:  dtoverlay=gpio-ir,<param>=<val>

Params: gpio_pin                Input pin number. Default is 18.

        gpio_pull               Desired pull-up/down state (off, down, up)
                                Default is "up".

        invert                  "1" = invert the input (active-low signalling).
                                "0" = non-inverted input (active-high
                                signalling). Default is "1".

        rc-map-name             Default rc keymap (can also be changed by
                                ir-keytable), defaults to "rc-rc6-mce"
```
The `gpio-ir` overlay configures an IR _receiver_. If you want a transmitter (as well) you also need `gpio-ir-tx`:
```
pi@raspberrypi:~$ dtoverlay -h gpio-ir-tx
Name:   gpio-ir-tx

Info:   Use GPIO pin as bit-banged infrared transmitter output.
        This is an alternative to "pwm-ir-tx". gpio-ir-tx doesn't require
        a PWM so it can be used together with onboard analog audio.

Usage:  dtoverlay=gpio-ir-tx,<param>=<val>

Params: gpio_pin                Output GPIO (default 18)

        invert                  "1" = invert the output (make it active-low).
                                Default is "0" (active-high).
```
Try this:
```
dtoverlay=gpio-ir-tx,gpio_pin=17
dtoverlay=gpio-ir,gpio_pin=18
```

sababb81, 2020-07-19:
> It might just be a typo in your report, but the parameters for the gpio-ir overlay are wrong:
>
> ```
> pi@raspberrypi:~$ dtoverlay -h gpio-ir
> Name:   gpio-ir
>
> Info:   Use GPIO pin as rc-core style [...]

Hi.  Thank you for the swift response and my apologies for the delay in responding - family bereavement.

I've gone ahead and made changes to /boot/config.txt as you had suggested.

However, with this I was not able to receive signals from the remote control.  If I however comment out dtoverlay=gpio-ir-tx,gpio_pin=17 in config.txt, I am then able to receive signals.  Using the logic that you can either only send or receive, I then instead commented out dtoverlay=gpio-ir,gpio_pin=18.  Whilst irsend does not generate any error messages, no signals are sent.

What I was hoping was going to be a fun and straightforward project has turned out to be a real nightmare, so any final thoughts on what this could be before I put this one down to experience??

Thanks again