raspberrypi/linux #3726: Several drmModeSetCrtc() calls cause video-lockup.

vanfanel, 2020-07-16:
**ABOUT THE TEST SYSTEM**

-Running standard Linux graphics stack in full KMS mode (BRCM legacy libraries removed):

```
/boot/config.txt:
dtoverlay=vc4-kms-v3d

```
-Latest official kernel, 64bit kernel and libs:

```
/boot/config.txt:
arm_64=1

pi@raspberrypi:~/src/LBE_DOCS/DRM_GBM_VIDEOMODE_CHANGES_TEST $ uname -r
5.4.51-v8+
```

-Latest stable libDRM and MESA installed:

```
mesa-20.1.3
libdrm-2.4.102
```

**PROBLEM / BUG**

Calling drmModeSetCrtc() several times causes a video-lockup (monitors report "no-signal"), which causes programs that dynamically change physical video mode to fail.
An error regarding the problem cause is shown in dmesg:

`[ 2974.688398] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:74:crtc-2] flip_done timed out`

**TEST PROGRAM**

I have made a small program that changes the physical display mode using drmModeSetCrtc() repeatedly, thus causing the video lock-up and the aforementioned dmesg error.
It's available here:

https://github.com/vanfanel/LBE_DOCS/tree/master/DRM_GBM_VIDEOMODE_CHANGES_TEST

Relevant program sections are:
1)  The drmModeSetCrtc() call:
https://github.com/vanfanel/LBE_DOCS/blob/82ac9fa3dd2a03b9d5444936ae7caa4dd7af2ea2/DRM_GBM_VIDEOMODE_CHANGES_TEST/context.c#L555

2) The pageflip issuing call:
https://github.com/vanfanel/LBE_DOCS/blob/82ac9fa3dd2a03b9d5444936ae7caa4dd7af2ea2/DRM_GBM_VIDEOMODE_CHANGES_TEST/context.c#L562

3) The pageflip waiting function, called here:
https://github.com/vanfanel/LBE_DOCS/blob/82ac9fa3dd2a03b9d5444936ae7caa4dd7af2ea2/DRM_GBM_VIDEOMODE_CHANGES_TEST/context.c#L571

As you will see, whenever a drmModeSetCrtc() call is done, we have waited for previously issued pageflip to complete. We wait here, in the GBM+EGL surfaces destruction function:
https://github.com/vanfanel/LBE_DOCS/blob/82ac9fa3dd2a03b9d5444936ae7caa4dd7af2ea2/DRM_GBM_VIDEOMODE_CHANGES_TEST/context.c#L234

So, all in all, it seems to be a problem with drmModeSetCrtc() specific to the VC4 libDRM module, or the kernel. It doesn't happen on intel or amdgpu.

**STEPS TO REPRODUCE**

Run the test program (https://github.com/vanfanel/LBE_DOCS/tree/master/DRM_GBM_VIDEOMODE_CHANGES_TEST) repeatedly. It will eventually cause the video lock-up.

6by9, 2020-07-17:
Which variant of Pi?
Full KMS on Pi4 is still very much beta, and the early patchset that is currently applied is known to have issues. It also has the potential to have compromised it on earlier Pis, although most use cases tested have been fine.

I'm in the process of updating the full KMS driver to the latest version sent to dri-devel.

vanfanel, 2020-07-17:
@6by9 : I am doing those tests on a Pi3b+.

HiassofT, 2020-08-20:
It looks like I ran into the same issue when testing LibreELEC master on an RPi3B+ (which is also using the full KMS driver).

I initially noticed it when I had "adjust refresh rate" enabled in player settings (which causes video mode / refresh rate to be adjusted to match those of the video being played), but simply switching back and forth between 1080p 50 and 60Hz in the display settings triggers this, too.

Here's dmesg from current 5.4.59 kernel http://ix.io/2uIA and kodi log http://ix.io/2uIB

In this case I managed to trigger the bug in the first attempt and quite interestingly after about 2 minutes the "flip_done timed out" messages stopped and the display came back.