raspberrypi/linux #3737: Trouble upgrading - apt-get upgrade

shazzr, 2020-07-19:
Get error regarding libavformat58 when upgrading my buster system.
http://paste.debian.net/1157066/

**To reproduce**
sudo apt-get update && sudo apt-get upgrade

http://paste.debian.net/1157066/

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

Pi3B+
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
Apr 15 2020 11:43:08 
Copyright (c) 2012 Broadcom
version 82f9bb929ce2186eb1824178c1ae82902ad6275c (clean) (release) (start)
Linux pitri 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

I have discussed the issue on #raspberrypi on Freenode.

Tenkawa| RoyK: I looked at this the other day and just now.. that 'activa4e-noawaIt' bit looks like a major typo in the package
Tenkawa| I see references to activate-nowait in deb triggers all over the place.. but not what is showing up there

popcornmix, 2020-07-20:
This repo is for tracking linux kernel bugs, so this is in the wrong place.
I'm not quite clear where the problem lies, but this would be a better starting point:
https://github.com/RPi-Distro/repo/issues

closed 2020-07-20 by popcornmix

XECDesign, 2020-07-20:
Also that looks like a locally corrupted file rather than an issue with the package.