raspberrypi/linux #3738: Binding multiple GPIO to one SPI CE?

haolee21, 2020-07-19:
Hi I am currently using raspberry pi with and digital decoder to allow me to use only 4 gpio pins to have 10 CE while using SPI (e.g., 1100 into the decoder will result in 1000000000 as output) while communicating with multiple devices. Yet from the device tree documents, it seems like I can only assign only one gpio as CE, I wonder is there any way to wrap multiple gpio pins?

pelwell, 2020-07-19:
SPI on a Pi uses regular GPIOs driven by the SPI framework. If you wanted some clever encoding like you describe you would need to write a virtual GPIO driver that encodes the 10 one-hot inputs onto 4 GPIO outputs, then use Device Tree to get the SPI interfaces to use these virtual GPIOs for its CE outputs. It wouldn't be very difficult to do, but you might want to make sure your external hardware doesn't glitch on changes since the CE changes aren't clocked.

haolee21, 2020-07-19:
@pelwell would you mind providing more info? I am not familiar with virtual gpio drivers.

pelwell, 2020-07-20:
My use of "virtual gpio drivers" was to describe a general concept rather than a specific thing. The idea is to create a device driver that presents itself to the system as a GPIO controller that manages a set of GPIOs - 10, in your case. At the same time, it is a client of the real GPIO controller, from which it requests four real GPIOs.

The system starts out with all inputs and outputs set to zero/low/off (hmmm, for CE signals you may have to mark the GPIOs as active low, so the level is high but still logically off, or you could build an inversion into the driver). When one of the inputs goes active the driver calculates the correct code and outputs it on the real GPIOs; when the input goes inactive the output returns to zeroes. Multiple simultaneous active inputs is an error.

pelwell, 2020-07-20:
In the Device Tree for a Pi the SPI controllers are declared like this:
```
		spi: spi0: spi@7e204000 {
			#address-cells = <1>;
			#size-cells = <0>;
			clocks = <&clocks BCM2835_CLOCK_VPU>;
			compatible = "brcm,bcm2835-spi";
			cs-gpios = <&gpio 8 1>, <&gpio 7 1>;
			...
```
The `cs-gpios` property declares two GPIOs for use as CS/CE lines - `&gpio` is a reference to the label on the GPIO controller, the first parameter is the GPIO number, and `1` as the second parameter means active-low. For your CE encoding scheme you would need a node to declare your own GPIO controller - let's give it the label `mygpio` - and replace `cs-gpios` with:
```
			cs-gpios = <&mygpio 0 1>, <&mygpio 1 1>, <&mygpio 2 1>, <&mygpio 3 1>, <&mygpio 4 1>,
				   <&mygpio 5 1>, <&mygpio 6 1>, <&mygpio 7 1>, <&mygpio 8 1>, <&mygpio 9 1>;
```

haolee21, 2020-07-21:
@pelwell  Sorry for asking such beginner questions, yet I really don't have experience in developing drivers that can be used in device tree, I just wonder any resources/keyword that you can suggest for researching.

pelwell, 2020-07-22:
A good way to learn about the kernel is to read the source code. Pick a driver that seems close to what you want and try to work out what it is doing. It's still what I do.

This project has quite a bit in common with something else I was wanting to try, but this is much simpler - there may be something for you to test fairly soon.

pelwell, 2020-07-23:
If you're prepared to build your own kernel (and if you're not then I think this is where this ends) then you could try branch `csenc` of https://github.com/pelwell/linux - or at least read the top three commits of https://github.com/pelwell/linux/commits/csenc.