raspberrypi/linux #3741: 5.4.51-v7+ raspberrypi-kernel:armhf (1.20200717-1) causes DRBD to stop working

MJP007, 2020-07-21:
I just upgraded a working two node raspberry pi drbd cluster from  kernel 1.20200601-1 to 1.20200717-1.  The boot loader also went up to the same version.  The drbd service now fails to mount the device.  The DRBD service throws this error on startup "create res: r0:failed(new-resource:20)".
Any assistance gratefully received.

MJP007, 2020-07-22:
Update:
After getting some help from the pi forum I used
rpi-udate e1050e94821a70b2e4c72b318d6c6c968552e9a2 
to take the machines back to a kernel 4.19.118.  That appears to have worked.  Which rather tends to suggest that the current state kernel development is incompatible with the version of drbd-utils available in the apt-get repository.  Can someone look into bringing the two things back into line?

XECDesign, 2020-07-22:
I can take a look. Is there a quick way to test whether it works without needing to learn about it and do a lot of setup?

MJP007, 2020-07-22:
Well it is a clustered/distributed raid control package which requires multiple nodes, but, each node has to be configured in isolation and the failures are occurring when it starts to bring a device up, i.e. before it connects to the other nodes.  If you imagine a scenario with a two node cluster, one node is dead and the other node has just been rebooted, if you can manually issue the commands to bring the device up on the rebooted node and get it to a state where it is waiting for the dead node to connect, then it may well be that we would have cracked the issue we have.  So no need to have two nodes talking to each other.

This is a really simple config for DRBD https://www.osradar.com/install-and-configure-drbd-on-debian-10/.  Just configure 1 node, ignore anything on the other node.  You will need a disk partition to point it at. If you get as far as issuing drbdadm attach r0, without any errors, then we will be in a much better place.

If I can be of any help, just ask.  I am far from expert, but feel free to ask anyway.

MorrieMalbec, 2020-08-05:
I had the same problem using Gentoo and found the solution was to patch the user/libgenl.h source file as follows:-

`--- drbd-utils-8.9.6.orig/user/shared/libgenl.h
+++ drbd-utils-8.9.6/user/shared/libgenl.h
@@ -843,7 +843,7 @@
 {
 	struct nlattr *start = (struct nlattr *)msg->tail;
 
-	if (nla_put(msg, attrtype, 0, NULL) < 0)
+	if (nla_put(msg, attrtype | NLA_F_NESTED, 0, NULL) < 0)
 		return NULL;
 
 	return start;
`
I haven't managed to successfully compile from source on the raspberry pi though but hopefully this may help if a package maintainer reads it

MM

Sebastian0816, 2020-09-28:
Above patched fixed the issue for me on Raspberry running raspian/debian 10.4 with kernel Linux rpi-cl2a 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux and drbd module:
root@rpi-cl2a:/etc# modinfo drbd
filename:       /lib/modules/5.4.51-v7+/kernel/drivers/block/drbd/drbd.ko
alias:          block-major-147-*
license:        GPL
version:        8.4.11

 I used drbd-utils-9.5.0 installed via apt-get source and patched it.

kanutope, 2020-12-23:
Running raspian/debian 10.7 with Linux '5.4.79-v7+ #1373 SMP', I encountered the same problem as @MJP007 . Got is solved by

- removing default package by
`sudo apt remove drbd-utils`
- getting latest 'drbd-utils' from linbit.com [drbd-utils-9.15.1.tar](https://www.linbit.com/downloads/drbd/utils/drbd-utils-9.15.1.tar.gz)
- unpacking
- executing
`./configure`
`make`
`sudo make install`

During `./configure` I had to install a few missing package, specific to my system(s): automake, flex, xsltproc, po4a.
During `make`, I got errors when documentation was built, though that didn't break things (apparently).

MJP007, 2021-01-08:
I have applied the work around from kanutope (thanks) and it is up and running.  If we can get the repository updated then great but in the mean time, panic over.