raspberrypi/linux #3742: 5.4.51 breaks cifs: mv cannot overwrite files

mrankine, 2020-07-21:
Since upgrading the kernel to 5.4.51, mv fails to overwrite existing files on a CIFS filesystem.

Expected behaviour:
$ touch foo bar
$ mv -f foo bar
[mv succeeds; "foo" is renamed "bar", replacing the existing file "bar"]

Actual behaviour:
$ touch foo bar
$ mv -f foo bar
mv: cannot move 'foo' to 'bar': File exists

4.19.118 and 5.4.50 were OK.

**Backstory**
* Advised by dom to create an issue here.
* Pi forums thread: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=280530&e=1
* Suggestion that there is a recent kernel regression. I haven't located the bug report mentioned. https://unix.stackexchange.com/a/599433/423705

**System**
* Which model of Raspberry Pi? Pi 4B
* Which firmware version (`vcgencmd version`)? 21a15cb094f41c7506ad65d2cb9b29c550693057 (Jul 17 2020 10:59:37)
* Which kernel version (`uname -a`)? Linux percival 5.4.51-v7l+ #1326 SMP Fri Jul 17 10:51:18 BST 2020 armv7l GNU/Linux

popcornmix, 2020-07-21:
This looks related: https://github.com/raspberrypi/linux/commit/71a20b798da3964b669f51f22cd0b3856549a571

mrankine, 2020-07-21:
For anyone else affected, here's the bug report on the linux-cifs mailing list:
https://marc.info/?t=159526521700006&r=1&w=2

popcornmix, 2020-07-21:
Okay, we'll keep and eye on that and either revert https://github.com/raspberrypi/linux/commit/71a20b798da3964b669f51f22cd0b3856549a571 or pick up a better fix before the next kernel update.

mrankine, 2020-07-22:
The 5.4.51 regression only affects the SMB v1 protocol.

Configuring client and server to use SMB v2 or v3 seems to resolve.

popcornmix, 2020-07-23:
Latest rpi-update firmware contains the revert of the bad cifs commit. Please confirm it fixes your issue.

ingleneuk, 2020-07-23:
Yes, the latest rpi-update fixes the issue. Thanks!

mrankine, 2020-07-23:
Yes, latest rpi-update fixes the issue. Thanks!

popcornmix, 2020-07-23:
That's for confirming. If there's no new issues reported we'll bump the apt kernel later today.

closed 2020-07-27 by mrankine