raspberrypi/linux #3743: Upgrade to 5.4.51 broke snd-rpi-hifiberry-dacplus

lyn1337, 2020-07-21:
Recent upgrade broke snd-rpi-hifiberry-dacplus (and hifiberry amp 2 head compatibility).

dmesg:
`[    4.722147] snd-rpi-hifiberry-dacplus soc:sound: ASoC: failed to init link HiFiBerry DAC+: -517`

pelwell, 2020-07-21:
That message isn't in itself an indication of an error - it should really be suppressed - but if the ALSA device doesn't appear then it is. What does `aplay -l` report?

@j-schambacher Have you tried the latest release?

lyn1337, 2020-07-21:
@pelwell My bad! The card didn't show up trying that, just forgot to mention.

j-schambacher, 2020-07-21:
I've tried with 5.4.50. I will have a look to it.

@lyn1337 is there any other error message than the -517 in the dmesg output?

lyn1337, 2020-07-21:
@j-schambacher None, I believe just the one over and over.

j-schambacher, 2020-07-21:
@lyn1337 I've x-checked and it works with a couple of different DACplus boards here. I've also done an 'update/upgrade' without problems. I can see some -517 (it's on our backlog to be fixed), but as Phil said it's not nice but the nature of deferred probing of kernel modules. Do you have any other hint? What PI version? ....

lyn1337, 2020-07-21:
@j-schambacher Hmm, let's see...
```
cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.2
```

lyn1337, 2020-07-21:
Seems working now despite the message. I left out specifying `dtoverlay=hifiberry-dacplus` in the /boot/config.txt, which wasn't needed before.
Thanks everyone involved!

closed 2020-07-21 by lyn1337

mishka81, 2020-07-22:
@lyn1337 I got the same issue, had to add dtoverlay=hifiberry-dacplus in my /boot/config.txt
But now I have no HDMI sound output ! 
```
pi@rpi4:~ $ cat /proc/asound/cards
 0 [sndrpihifiberry]: HifiberryDacp - snd_rpi_hifiberry_dacplus
                      snd_rpi_hifiberry_dacplus
```
Same for you ?