raspberrypi/linux #3746: Upgrade to 5.4.51 breaks ath9k, kernel patched 11657669

Deenniss, 2020-07-21:
**Describe the bug**
Upgrade to kernel 5.4.51 breaks ath9k module, on all platforms (not only Raspberry Pi), and there is a patch available.
As a consequence now, for me and others, all WiFi devices using ath9k do not work anymore.

**To reproduce**
Update the raspberry pi to latest stable kernel (apt update && apt upgrade).

**Expected behaviour**
ath9k should work as it has for years (client & AP).

**Actual behaviour**
ath9k is broken, WiFi device is showing up as USB device, but no interface is showing up (i.e. wlan0)
WiFi device works fine with previous kernel, and on other GNU/Linux systems and hardware.
LOG on RaspberryPi, stable kernel 5.4.51:
[   37.693289] ath9k_htc 1-2.2:1.0: ath9k_htc: HTC initialized with 33 credits
[  148.871980] ath9k_htc: Failed to initialize the device
[  148.872155] usb 1-2.2: ath9k_htc: USB layer deinitialized

**Additional context**
This is a known kernel bug, referenced here:
https://bugzilla.kernel.org/show_bug.cgi?id=208251

The bug has been patched there:
https://patchwork.kernel.org/patch/11657669/

Best regards!

popcornmix, 2020-07-21:
Okay, we'll pick up that patch before the next kernel update.

6by9, 2020-07-21:
Comment https://bugzilla.kernel.org/show_bug.cgi?id=208251#c27 says it's in 5.4.52 which was released 6 days ago. Is it easier to do the bump instead of reverting locally?

kilrah, 2020-07-22:
Can you put back the previous package on the repo at least so it's easy to downgrade with apt in the meantime?

popcornmix, 2020-07-22:
@XECDesign is the older kernel/firmware not available from apt?

popcornmix, 2020-07-23:
Latest rpi-update firmware contains the revert of the bad ath9k commit. Please confirm it fixes your issue.

rmrten, 2020-07-23:
> 
> 
> Latest rpi-update firmware contains the revert of the bad ath9k commit. Please confirm it fixes your issue.

had the same issue. Did rpi-update and now WN722N working fine.

Thanks

kilrah, 2020-07-23:
Updated via apt after confirming the new package was from today and it's fixed indeed. Same wifi adapter as above. Thanks

popcornmix, 2020-07-23:
@Deenniss please test and close if you are happy.

max17048, 2020-07-23:
Did also rpi-update on a 3B and now WN722N working fine again.

Deenniss, 2020-08-15:
Did the upgrade today: problem solved on 5.4.51-v7+.
I close this bug.
Thanks everyone!

closed 2020-08-15 by Deenniss

Procuria, 2020-09-08:
Sorry to reopen this but I upgraded to 5.4.51-v7+ on a raspi CM3 and the problem persists. Can anyone confirm this and/or point me the direction to solve this issue please ?

pelwell, 2020-09-08:
How did you upgrade? What does `uname -a` report?

Procuria, 2020-09-08:
[image]

Thank you for the fast reply ! I updated via apt update && apt upgrade. Does it help if I send a screenshot of the dmesg-erorr too ?

pelwell, 2020-09-08:
> Does it help if I send a screenshot of the dmesg-erorr too ?

It might.

Procuria, 2020-09-08:
[image]

There we go :) I already reinstalled the firmware ath9k_htc but this doesn't solved the problem.

max17048, 2020-09-08:
> *Edit: This is one cycle of the 'loop'. I tested it against an USB-WLAN Adapter with a Realtek Chipset, which works like a charm.

On the 3B, the AR9271 (WN722N) works well.

Procuria, 2020-09-09:
I can confirm to, that it works on a PI3 but not on a CM3+. I investigated the Spec's of both platforms to check if there are any differences that could cause this problem, but couldn't find any. Does somebody maybe have a hint for me regarding this starnge behaviour ? I already contacted 'Alfa Networks', the manufacturer of this USB WLAN device, and hope to get some info's of them.