raspberrypi/linux #3749: Upgrade to 5.4.51-v7+ breaks SPI

ludionisio, 2020-07-23:
**Describe the bug**
I can no longer do an `echo "7" > /sys/class/gpio/export` and `-bash: echo: write error: Device or resource busy` is thrown.

I tried disabling and reenabling SPI through `raspi-config` and even though there is effectively a `/dev/spidev0.0` and `/dev/spidev0.1` the fore mentioned command can no longer be executed.

**To reproduce**
After an upgrade through `sudo apt upgrade` from kernel 4.19.118-v7+ I am unable to use SPI

**System**

Result from `sudo cat /sys/kernel/debug/gpio`:
`gpiochip0: GPIOs 0-53, parent: platform/3f200000.gpio, pinctrl-bcm2835:`
 `gpio-7   (                    |spi0 CS1            ) out hi ACTIVE LOW`
 `gpio-8   (                    |spi0 CS0            ) out hi ACTIVE LOW`
 `gpio-29  (                    |led0                ) out lo `

* Which model of Raspberry Pi? Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)? Raspberry Pi reference 2020-05-27 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
* Which firmware version (`vcgencmd version`)? Jul 17 2020 11:04:27 version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)
* Which kernel version (`uname -a`)? Linux raspberrypi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux

**Logs**
`dmesg`:
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.51-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1327 SMP Thu Jul 23 10:58:46 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 1896 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49792 r8192 d23936 u81920
[    0.000000] pcpu-alloc: s49792 r8192 d23936 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240792
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=df2829cd-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 882468K/970752K available (8192K kernel code, 681K rwdata, 2532K rodata, 1024K init, 824K bss, 22748K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 27966 entries in 55 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x328/0x4e8 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000292] Console: colour dummy device 80x30
[    0.000871] printk: console [tty1] enabled
[    0.000937] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000985] pid_max: default: 32768 minimum: 301
[    0.001334] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001380] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002570] CPU: Testing write buffer coherency: ok
[    0.003139] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004100] Setting up static identity map for 0x100000 - 0x10003c
[    0.004324] rcu: Hierarchical SRCU implementation.
[    0.005018] smp: Bringing up secondary CPUs ...
[    0.006155] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.007409] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.008559] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.008711] smp: Brought up 1 node, 4 CPUs
[... 221 lines cut ...]
[    7.188839] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt failed with error -2
[    7.405405] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.419727] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    7.447401] random: crng init done
[    7.447414] random: 7 urandom warning(s) missed due to ratelimiting
[    8.809686] uart-pl011 3f201000.serial: no DMA platform data
[    8.898374] 8021q: 802.1Q VLAN Support v1.8
[    9.236256] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    9.259628] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    9.516893] 8021q: adding VLAN 0 to HW filter on device eth0
[    9.520097] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   10.798619] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.eth0.retrans_time - use net.ipv6.neigh.eth0.retrans_time_ms instead
[   15.852055] Bluetooth: Core ver 2.22
[   15.852131] NET: Registered protocol family 31
[   15.852136] Bluetooth: HCI device and connection manager initialized
[   15.852151] Bluetooth: HCI socket layer initialized
[   15.852160] Bluetooth: L2CAP socket layer initialized
[   15.852175] Bluetooth: SCO socket layer initialized
[   15.918870] Bluetooth: HCI UART driver ver 2.3
[   15.918882] Bluetooth: HCI UART protocol H4 registered
[   15.918929] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.919079] Bluetooth: HCI UART protocol Broadcom registered
[   17.101439] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   17.101453] Bluetooth: BNEP filters: protocol multicast
[   17.101474] Bluetooth: BNEP socket layer initialized`

pelwell, 2020-07-23:
Why do you need GPIO access to GPIO 7? Why does not having that prevent you from using SPI?

ludionisio, 2020-07-23:
Being able to compare the updated system with the previous one I identified with the command `cat /sys/kernel/debug/gpio` that before the GPIO07 was:
`gpiochip0: GPIOs 0-53, parent: platform/3f200000.gpio, pinctrl-bcm2835:`
 `gpio-7   (                    |sysfs               ) in  lo`    
 `gpio-29  (                    |led0                ) out lo`

Investigating further It seems that the current configuration actually reserves pins 7 and 8 for the SPI Chip Select. Which seems the right thing to do, even though this wasn't being done before.

But I am working with LoRa Gateway PiHats (different brands) that share a common design following SemTech's one. They all use GPIO07 to reset the concentrator chip and it is not possible to change it (unless some trail cutting and some hack wiring takes place). I realize now that the update does not affect SPI, but only Hats that use SPI and expect GPIO07 to be free.

I am not sure now what would be the proper solution because it is not clear what exactly changed. Is the latest kernel using a different device tree overlay? I saw that many others were having similar problems with other peripherals as well after the upgrade.

Would you have any suggestion?

pelwell, 2020-07-24:
If you have a spare GPIO somewhere then the spi0-cs overlay will let you reassign either or both CS pins:
```
pi@raspberrypi:~ $ dtoverlay -h spi0-cs
Name:   spi0-cs

Info:   Allows the (software) CS pins for SPI0 to be changed

Usage:  dtoverlay=spi0-cs,<param>=<val>

Params: cs0_pin                 GPIO pin for CS0 (default 8)
        cs1_pin                 GPIO pin for CS1 (default 7)
```
You would need to add `dtoverlay=spi0-cs,cs1_pin=99`, where you replace 99 with the number of the spare GPIO.

If you don't have one spare then it wouldn't be difficult to modify this overlay or create a new one that makes CS1 optional. At the same time I'd probably also make the use of the MISO optional because users seem to have borrowed GPIO 9 for their own purposes.

ludionisio, 2020-07-24:
Excelent!
Thanks a lot!

closed 2020-07-24 by ludionisio

sp193, 2020-11-28:
> If you have a spare GPIO somewhere then the spi0-cs overlay will let you reassign either or both CS pins:
>
> ```
> pi@raspberrypi:~ $ dtoverlay -h spi0-cs
> Name:   spi0-cs
>
> Info:   Allows the (software) [...]

Sorry to bring up this issue again, but is this really all that needs to be done? For example, if GPIO4 is unused and I would like to reassign the kernel's claim on GPIO7 to GPIO4, do I just enter this command as the ``pi`` user?
``dtoverlay=spi0-cs,cs1_pin=4``

There's no output from dtoverlay after entering the command. GPIO7 remained unusable, even after rebooting the RPi. I'm not sure what else to check, since I'm also quite new to the RPi.

Unlike the OP, I'm trying out a RisingHF RHF0M301 with RHF4T002 bridge. The RHF4T002 also has the reset pin connected to GPIO7.

pelwell, 2020-11-28:
Since the original issue was opened, a new overlay - spi0-1cs - had been added. This configures SPI0 to use only one CS line:
```
dtoverlay=spi0-1cs,cs0_pin=4
```

sp193, 2020-11-28:
Thanks. I eventually also figured out that I needed to add the line to ``/boot/config.txt``.
Once added and the RPi is rebooted, the GPIO configuration gets changed to:
```
pi@raspberrypi:sudo cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 0-53, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-4   (                    |spi0 CS0            ) out hi ACTIVE LOW
 gpio-35  (                    |led1                ) in  lo 
 gpio-47  (                    |led0                ) out lo
```
I am now able to use GPIO7. Once again, thanks.

sp193, 2020-12-01:
Pardon me for asking, but why would I get a difference in behaviour between using these two, given that I do not use the CS0/CS1 signals and only CS1 (GPIO7) is actually in the way:
```
dtoverlay=spi0-1cs,cs0_pin=5
```
```
dtoverlay=spi0-cs,cs1_pin=5
```
With the latter, the LoRa gateway library appears to work fine. But with the former, the library seems to read all registers as 0x00 values:
```
pi@raspberrypi:~/lora_gateway/libloragw $ ./test_loragw_reg
Beginning of test for loragw_reg.c
Note: SPI port opened and configured ok
Note: SPI read success
lgw_connect:532: INFO: no FPGA detected or version not supported (v0)
Note: SPI read success
lgw_connect:555: ERROR: NOT EXPECTED CHIP VERSION (v0)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
IMPLICIT_PAYLOAD_LENGHT = 0 (should be 197)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
FRAME_SYNCH_PEAK2_POS = 0 (should be 11)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
PREAMBLE_SYMB1_NB = 0 (should be 49253)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
ADJUST_MODEM_START_OFFSET_SF12_RDX4 = 0 (should be 3173)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
IF_FREQ_1 = 0 (should be -1947)
ERROR: CONCENTRATOR UNCONNECTED
ERROR: CONCENTRATOR UNCONNECTED
Note: SPI port closed
Note: success disconnecting the concentrator
End of test for loragw_reg.c
```

The healthy output is actually quite long, but the chip version should be detected (no error logged) and fields like IMPLICIT_PAYLOAD_LENGHT, FRAME_SYNCH_PEAK2_POS and IF_FREQ_1 should have their written values reflected correctly (not 0). My board has no FPGA, so the log message about the FPGA is expected.

Previously, I asked about how to map CS1 to GPIO4. I'm now using GPIO5 because this pin is actually used for the LCD panel's backlight.

Once again, thanks in advance.

pelwell, 2020-12-01:
```
dtoverlay=spi0-1cs,cs0_pin=5
```
configures GPIO5 to be CE/CS line 0, and
```
dtoverlay=spi0-cs,cs1_pin=5
```
(which gets converted to "spi0-2cs,...") configures GPIO8 as CE/CS line 0 and GPIO5 as CE/CS line 1.

If you only want to use 1 CS line then it must be CS0, so you'll need to reconfigure the rest of your overlay/software to take account of that.

sp193, 2020-12-01:
Thanks. I understand the problem now. I incorrectly assumed it never used a CS line, due to a lack of experience with SPI and documentation. You are right, it appears to be using CS0 (GPIO8) for its CSN pin input, while CS1 (GPIO7) is used for reset. So I need to keep CS0 as GPIO8, otherwise the board stops working since it never gets enabled.