raspberrypi/linux #3751: Not able to fetch linux-raspberrypi-1_5.4.40+gitAUTOINC+aafb8f095e_machine-r0

Siddharthav1, 2020-07-24:
Hello, 

        From past two days I am trying to build the xen image for RPI-4. But I am not able to fetch the linux raspberrypi-5.4 kernel. Earlier I thought it might be my network issue but its not. It looks like there is an issue with the github may be. Please suggest me what can I do fetch the linux raspberrypi-5.4 kernel.

Regards,
Siddhartha V

pelwell, 2020-07-24:
Where did you get that name from, and what do you think it refers to - a binary or source package?

Siddharthav1, 2020-07-24:
Hello,    

  Actually I am building the Xen hypervisor image for RPI4 by using yocto. I am using **"bitbake -k xen-image-minimal "** to build the image. but unfortunaltly from past two days I am getting the below error

```
ERROR: linux-raspberrypi-1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus,guid=5444b1d0deb73f37f3b69e8f5f18fb4a"; export SSH_AGENT_PID="1622"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/siddhu/Documents/yocto/zindagi/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/siddhu/Documents/yocto/zindagi/sources1/poky/scripts:/home/siddhu/Documents/yocto/zindagi/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux:/home/siddhu/Documents/yocto/zindagi/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0/recipe-sysroot/usr/bin/crossscripts:/home/siddhu/Documents/yocto/zindagi/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0/recipe-sysroot-native/usr/sbin:/home/siddhu/Documents/yocto/zindagi/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0/recipe-sysroot-native/usr/bin:/home/siddhu/Documents/yocto/zindagi/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0/recipe-sysroot-native/sbin:/home/siddhu/Documents/yocto/zindagi/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0/recipe-sysroot-native/bin:/home/siddhu/Documents/yocto/zindagi/sources/poky/bitbake/bin:/home/siddhu/Documents/yocto/zindagi/build/tmp/hosttools"; export HOME="/home/siddhu"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror git://github.com/raspberrypi/linux.git /home/siddhu/Documents/yocto/zindagi/build/downloads/git2/github.com.raspberrypi.linux.git --progress failed with exit code 128, output:
Cloning into bare repository '/home/siddhu/Documents/yocto/zindagi/build/downloads/git2/github.com.raspberrypi.linux.git'...
remote: Enumerating objects: 47, done.        
remote: Counting objects: 100% (47/47), done.        
remote: Compressing objects: 100% (41/41), done.        
fatal: read error: Connection reset by peer01.66 MiB | 450.00 KiB/s    
fatal: early EOF
fatal: index-pack failed

ERROR: linux-raspberrypi-1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0 do_fetch: Fetcher failure for URL: 'git://github.com/raspberrypi/linux.git;branch=rpi-5.4.y'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/siddhu/Documents/yocto/zindagi/build/tmp/work/raspberrypi4_64-poky-linux/linux-raspberrypi/1_5.4.47+gitAUTOINC+aafb8f095e_machine-r0/temp/log.do_fetch.3251
ERROR: Task (/home/siddhu/Documents/yocto/zindagi/sources1/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_5.4.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4100 tasks of which 3941 didn't need to be rerun and 1 failed.
```
Later I tried to fetch only linux-raspberrypi using **"bitbake -c fetch linux-raspberryp"** but eventhen the result was same. 

 As I mentioned earlier I thought it might be due to my network issue. But it doesn't look so. It looks like issue is with the github repo. So I thought of seeking help here.

I kindly request you suggest me how can I fetch the repo please.

Regards,
Siddhartha v

pelwell, 2020-07-24:
This forum article probably applies: https://www.raspberrypi.org/forums/viewtopic.php?t=276975

But in general, we can't support bitbake.

Barhoumi-Nawres, 2024-11-30:
how did you solve the problem now i have the same problem but i don't know how to solve it thanks in advance