raspberrypi/linux #3754: unable to use g_cdc/g_ether with gadget boot without initrd

H-Sachse, 2020-07-25:
**Describe the bug**
usb gadget booting into NFS root fails.
can not load g_ether/g_cdc drivers without resorting to initrd, because drivers are modules.

**To reproduce**
gadget boot recent RaspberryPiOS lite

**Expected behaviour**
raspberrypi kernel can be gadget booted with g_ether/g_cdc functionality 

**Actual behaviour**
boot stalls with no network connectivity

**System**
raspberrypi zero-w connected via USB to raspberrypi 3B+
current rpiboot from github 
OS: Stretch updated to Buster 10.4
Kernel 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020

**Additional context**
I tried setting up NFS root for a gadget booted RPi-zero, but the boot stalls due to missing gadget modules in the kernel.

ED6E0F17, 2020-07-27:
Yes. You will need kernel modules for USB, or firmware for the wifi, before you can get a network connection to continue the boot process, so you will need an initrd.

H-Sachse, 2020-07-27:
Wifi has nothing to do with it, and if you want to do USB gadget boot, needing an initrd is an unnecessary hassle.

the whole of all gadget modules and function for the RPI0/1 kernel is only 224k in size and could be included into the kernel

```16K	function/u_audio.ko
24K	function/u_ether.ko
20K	function/u_serial.ko
16K	legacy/g_acm_ms.ko
12K	legacy/g_audio.ko
12K	legacy/g_cdc.ko
16K	legacy/g_ether.ko
12K	legacy/g_hid.ko
16K	legacy/g_mass_storage.ko
12K	legacy/g_midi.ko
16K	legacy/g_multi.ko
12K	legacy/g_printer.ko
12K	legacy/g_serial.ko
12K	legacy/g_webcam.ko
16K	legacy/g_zero.ko
224K	total
```