raspberrypi/linux #3756: Instabilities/crashes due to kernel update 5.4.51

ysard, 2020-07-26:
**Describe the bug**

On an originally stable system (2 voluntary restarts in 20 months).

- Instabilities/crashes due to migration from "4.19.75-v7+ #1270" to successive versions of kernel 5.4.51 (1.20200717-1 and 1.20200723-1).
- Difficulties installing packages requiring kernel headers

---

# Unable to compile kernel modules vs. kernel headers:

**Describe the bug**

I use the xtables-addons-common package (v3.2-1 from the repository) to complete my server's iptables firewall.
This important package requires the current kernel headers. However, multiple errors appear when compiling it
using dkms (xtables-addons-dkms package).

**To reproduce**

    $ sudo apt-get install xtables-addons-common
    or
    $ dkms install -m xtables-addons -k 4.19.75-v7+ -v 3.2
    ...
    run-parts: executing /etc/kernel/header_postinst.d/dkms 5.4.51-v7+.
    Error!  Build of xt_ACCOUNT.ko failed for: 5.4.51-v7+ (armv7l)
    Consult the make.log in the build directory
    /var/lib/dkms/xtables-addons/3.2/build/ for more information.

In the make.log file I distinguish 2 types of errors:

## Kernel independent errors (4.19.75 and 5.4.51) in the precompiled binaries in the `/srcripts/` directory .

**To reproduce**

    ./scripts/recordmcount: Exec format error
    ./scripts/basic/fixdep: Exec format error
    ./scripts/basic/mod/modpost: Exec format error
    ./scripts/basic/mod/mk_elfconfig: Exec format error

If manual execution is attempted:

    /usr/src/linux-headers-4.19.75+ $ scripts/basic/fixdep
    -bash: scripts/basic/fixdep: impossible d'exécuter le fichier binaire : Erreur de format pour exec()

I thought at first I'd have to recompile the kernel. In fact It was a waste of time since they are actually binaries that don't match the platform of the Raspberry Pi.

Tests performed in /usr/src/linux-headers-*/

test on binary from the raspberrypi-kernel-headers package:

    $ file scripts/mod/mk_elfconfig
    mk_elfconfig: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, for GNU/Linux 3.7.0, BuildID[sha1]=e8f88d80903552cea01900283c192455de82ddba, not stripped

test on compiled cross binary:

    make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- scripts
    $ file scripts/mod/mk_elfconfig
    scripts/mod/mk_elfconfig: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=d9a2cb16f1ef2700a3726e8105cf15f3961e40b4, not stripped

**Expected behaviour**

test binary compiled on site:

    $ sudo make scripts
    $ file scripts/mod/mk_elfconfig
    scripts/mod/mk_elfconfig: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 3.2.0, BuildID[sha1]=c658f0306135e453aaa4f5151c93e5d7aca79f21b, not stripped

**Actual behaviour**

- the make command does not take into account the destination platform
[... 276 lines cut ...]
Jul 21 13:27:46 localhost kernel: [ 2846.106211] Backtrace:
Jul 21 13:27:46 localhost kernel: [ 2846.108540] [<8020b204>] (trace_hardirqs_on) from [<801259f4>] (__local_bh_enable_ip+0x74/0xd0)
Jul 21 13:27:46 localhost kernel: [ 2846.113175]  r8:be82f000 r7:00002ce9 r6:be581300 r5:00000200 r4:b991e000 r3:00000000
Jul 21 13:27:46 localhost kernel: [ 2846.117904] Backtrace aborted due to bad frame pointer <d21c29c4>
Jul 21 13:27:46 localhost kernel: [ 2846.120332] Code: 5f617077 6c727463 3534315f 312d3631 (00058b9c)
Jul 21 13:27:46 localhost kernel: [ 2846.122709] ---[ end trace 425b335f65cb5021 ]---

[  374.966284] Internal error: Oops: d [#1] SMP ARM
[  374.978415] Process kworker/3:0 (pid: 26, stack limit = 0xf1b66816)
[  374.980722] Stack: (0xb9977ea0 to 0xb9978000)
[  374.982966] 7ea0: b9977ee4 b9977eb0 7f123220 808d2e14 8013bd80 be851e90 8013c5fc be85104c
[  374.987614] 7ec0: b980ba00 be5bd300 be5c0900 00000000 b9977ee4 900526a0 7f123224 808cd9fc
[  374.992420] 7ee0: 7f123670 7f123160 be85104c b980ba00 b9977f34 b9977f00 8013bd94 7f123658
[  374.997408] 7f00: be5bd300 be5bd300 00000008 be5bd300 b980ba14 00000008 be5bd318 80d03d00
[  375.002548] 7f20: b9976038 b980ba00 b9977f7c b9977f38 8013c5fc 8013bc24 b9977f5c 00000000
[  375.007904] 7f40: 80d03d00 be5bd318 80d03d00 80d993f7 b9977f7c b98ce300 b98ce280 00000000
[  375.013375] 7f60: b980ba00 8013c5a8 b98fbda4 b98ce31c b9977fac b9977f80 801429e0 8013c5b4
[  375.019074] 7f80: ffffffff b98ce280 801428a4 00000000 00000000 00000000 00000000 00000000
[  375.024986] 7fa0: 00000000 b9977fb0 801010ac 801428b0 00000000 00000000 00000000 00000000
[  375.031173] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  375.037641] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  375.051021] Code: e52de004 e8bd4000 e1a0300d e3c33d7f (e3c3303f)
```

Best regards!

ysard, 2020-07-26:
Probably same context as #3757

vincent-olivert-riera, 2020-07-27:
I also reported an issue related to the kernel update breaking a dkms package: https://github.com/RPi-Distro/repo/issues/188
And also reported it in the forums: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=243717&p=1702605#p1702480

I was expecting RaspberryPi OS Buster to follow what Debian Buster does (they stick with 4.19 kernels). It seems diverting from them and update to 5.4 kernels wasn't the best decision, and it was done without proper testing, otherwise the dkms failures would have been detected.

JamesH65, 2020-07-27:
We need to move to 5.4 for many reasons, the newer Mesa stuff which is needed for our graphics stack is one major one, and that cannot be backported in any feasible way. So the move to 5.4 was unavoidable.

This 5.4 branch has been available for testing for 4 months in github, and in rpi-update for some weeks. So anyone could have tried it out and reported any issues. We do rely on third parties for testing as we are a very small team with lots of other work to do. Unfortunately, testing doesn't catch everything.

vincent-olivert-riera, 2020-07-27:
Sorry @JamesH65 , I wasn't aware of that.

So, now that a kernel 5.4 is necessary and we know is breaking other packages, would it be possible to get those packages fixed? For instance, for the `v4l2loopback-dkms` there is a fix upstream and in fact the package in Bullseye works fine.

JamesH65, 2020-07-27:
We are working right now on fixing the issues that have surfaced. There are some high priority ones to sort out before this one I suspect. We would also take PR's if others can fix the issues before we get round to them.

scotstan, 2020-08-04:
@vincent-olivert-riera thanks for mentioning the upstream `vl2loopback-dkms`...I came here looking around specifically for a solution to this. I have ~500 Raspberry Pi 4 units I'm preparing for a project...currently on 4.19, but hoping to go-forward quickly with 5.4. This was one of the blockers.

vincent-olivert-riera, 2020-08-04:
> @vincent-olivert-riera thanks for mentioning the upstream `vl2loopback-dkms`...I came here looking around specifically for a solution to this. I have ~500 Raspberry Pi 4 units I'm preparing for a project...currently on 4.19, but hoping to go-forward quickly with 5.4. This was one of the blockers.

You won't be able to install `v4l2loopback-dkms` even if you stay on 4.19, because the `linux-headers` package for 4.19 is no longer available from the packages list.
A solution for you is to do the same as I did. Upgrade to 5.4, but when you install `v4l2loopback-dkms` use the `.deb` from Bullseye:  http://raspbian.raspberrypi.org/raspbian/pool/main/v/v4l2loopback/v4l2loopback-dkms_0.12.5-1_all.deb