raspberrypi/linux #3757: kernel panic with 5.4.51-v7+ after short time, in particular during downloads

jan-h-1970, 2020-07-26:
I have got lots of kernel panic with 5.4.51-v7+ after short time, in particular during downloads

affected models are 2Bv1.2, 3Bv1.2, 3B+ and probably more
using genuine rpi3 2.5A power supply
kernel 4.19.118-v7+ never had such issues

rpi is used as a (internet dsl) router.
I tried these as the second nic, both with errors
0b95:7e2b ASIX Electronics Corp. AX88772B Fast Ethernet Controller (module asix)
0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter (module r8152)

kernel 1.20200723-1
uname -a
Linux rpi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux

vcgencmd version
Jul 17 2020 11:04:27 
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)

cat /etc/rpi-issue
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2

- - - - - - - - - - - 

errors are:
Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: return_address+0x88/0x88
Unable to handle kernel paging request at virtual address 48200240
Kernel panic - not syncing: Attempted to kill the idle task!
Unable to handle kernel NULL pointer dereference at virtual address 0000012c
WARNING: CPU: 1 PID: 4431 at drivers/firmware/raspberrypi.c:63 rpi_firmware_transaction+0x108/0x128
Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: __netif_receive_skb_core+0xc2c/0xc34
Unhandled prefetch abort: unknown 1 (0x001) at 0x00000002
Bad mode in prefetch abort handler detected
Internal error: Oops - bad mode: 0 [#1] SMP ARM
Unable to handle kernel paging request at virtual address fffffffc
Internal error: Oops: 837 [#1] SMP ARM

- - - - - - - - - - -

model 2Bv1.2 examples

rpi login: 
rpi login: [  348.318763] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: return_address+0x88/0x88
[  348.336178] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.51-v7+ #1327
[  348.351425] Hardware name: BCM2835
[  348.362018] Backtrace: 
[  348.371543] [<8010d458>] (dump_backtrace) from [<8010d750>] (show_stack+0x20/0x24)
[  348.386439]  r6:80d00000 r5:00000000 r4:80d93ff4 r3:e7eed6d5
[  348.399416] [<8010d730>] (show_stack) from [<808b2324>] (dump_stack+0xe0/0x124)
[  348.414216] [<808b2244>] (dump_stack) from [<8011ff28>] (panic+0x12c/0x34c)
[  348.428582]  r8:b909d000 r7:80daaa20 r6:80a99374 r5:00000000 r4:80daaa66 r3:00000001
[  348.443737] [<8011fdfc>] (panic) from [<8011fa98>] (print_tainted+0x0/0xa8)
[  348.458170]  r3:e7eed6d5 r2:80d04f48 r1:8010bc44 r0:80a99374
[  348.471336]  r7:b97b72c4
[  348.481347] [<8011fa74>] (__stack_chk_fail) from [<8010bc44>] (c_start+0x0/0x2c)
[  348.496478] [<8010cfa8>] (walk_stackframe) from [<8010bc18>] (return_address+0x5c/0x88)
[  348.512258]  r6:80d00000 r5:00000000 r4:80d04f48 r3:8010bbbc
[  348.525656] [<8010bbbc>] (return_address) from [<8020b3fc>] (trace_hardirqs_off+0x84/0x170)
[  348.541932]  r5:00000001 r4:80125d00
[  348.553437] [<8020b378>] (trace_hardirqs_off) from [<80125d00>] (tasklet_action_common+0x24/0xec)
[  348.570576]  r7:b97b72c4 r6:80d00000 r5:00000000 r4:80d03080
[  348.584581] [<80125cdc>] (tasklet_action_common) from [<80125df0>] (tasklet_hi_action+0x28/0x30)
[  348.601974]  r9:00000001 r8:b909d000 r7:00000100 r6:80d00000 r5:00000000 r4:80d03080
[  348.618468] [<80125dc8>] (tasklet_hi_action) from [<8010249c>] (__do_softirq+0x1a4/0x418)
[  348.635446] [<801022f8>] (__do_softirq) from [<80125be4>] (irq_exit+0x100/0x110)
[  348.651639]  r10:b9fff840 r9:80d00000 r8:b909d000 r7:00000001 r6:00000000 r5:00000000
[  348.668344]  r4:80ca22a4
[  348.679602] [<80125ae4>] (irq_exit) from [<80180ad4>] (__handle_domain_irq+0x70/0xc4)
[  348.696341] [<80180a64>] (__handle_domain_irq) from [<80102228>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
[... 784 lines cut ...]
[  277.543816] Backtrace: 
[  277.552087] [<8010d458>] (dump_backtrace) from [<8010d750>] (show_stack+0x20/0x24)
[  277.565604]  r6:b913c000 r5:00000000 r4:80d93ff4 r3:387f5dfd
[  277.577185] [<8010d730>] (show_stack) from [<808b2324>] (dump_stack+0xe0/0x124)
[  277.590472] [<808b2244>] (dump_stack) from [<8011033c>] (handle_IPI+0x334/0x3bc)
[  277.603763]  r8:80da4b40 r7:00000000 r6:00000003 r5:80ca22a4 r4:80daa8e8 r3:387f5dfd
[  277.617401] [<80110008>] (handle_IPI) from [<80102268>] (bcm2836_arm_irqchip_handle_irq+0xa0/0xa4)
[  277.632323]  r10:00000000 r9:b913c000 r8:80d9943c r7:b913df6c r6:ffffffff r5:60000013
[  277.646132]  r4:00000003
[  277.654520] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[  277.669385] Exception stack(0xb913df38 to 0xb913df80)
[  277.680390] df20:                                                       80109b94 00000000
[  277.694575] df40: 60000093 8011a2e0 80d04f68 b913c000 80d04fb0 00000008 80d9943c 410fd034
[  277.708745] df60: 00000000 b913df94 80d0535c b913df88 00000000 80109b98 60000013 ffffffff
[  277.722882]  r4:80109b98 r3:387f5dfd
[  277.732352] [<80109b64>] (arch_cpu_idle) from [<808d2c84>] (default_idle_call+0x34/0x48)
[  277.746481] [<808d2c50>] (default_idle_call) from [<80153494>] (do_idle+0xec/0x170)
[  277.760193] [<801533a8>] (do_idle) from [<801537f4>] (cpu_startup_entry+0x28/0x2c)
[  277.773880]  r8:0000406a r7:80daa8f8 r6:10c0387d r5:00000003 r4:0000008a r3:387f5dfd
[  277.787726] [<801537cc>] (cpu_startup_entry) from [<8010fd94>] (secondary_start_kernel+0x130/0x13c)
[  277.802931] [<8010fc64>] (secondary_start_kernel) from [<001027ac>] (0x1027ac)
[  277.816296]  r5:00000055 r4:3912806a
[  277.826003] ---[ end Kernel panic - not syncing: Attempted to kill the idle task! ]---

- - - - - - - - - - -

brabl2, 2020-07-27:
I had same behavior on my Raspberry Pi 2B v1.1.
After reverting back to 4.19.118-v7+ everything is stable again.

popcornmix, 2020-07-27:
Are you able to test with latest 5.4 kernel and older firmware?
You can get the previous firmware here:
https://github.com/Hexxeh/rpi-firmware/raw/2d76ecb08cbf7a4656ac102df32a5fe448c930b1/start.elf
https://github.com/Hexxeh/rpi-firmware/raw/2d76ecb08cbf7a4656ac102df32a5fe448c930b1/fixup.dat

brabl2, 2020-07-27:
I installed all raspberrypi related packages from repository to version 1.20200723-1, 
then manually installed raspberrypi-bootloader_1.20200601-1_armhf.deb package and rebooted.
It is running 5.4.51-v7+ #1327 and looking stable.

brabl2, 2020-07-27:
I find that problem of my system was caused by the raspberrypi/firmware#1431.
After commenting out the '#arm_freq_min=400' in the /boot/config.txt
the system is stable even with latest raspberrypi-bootloader version 1.20200723-1.

jan-h-1970, 2020-07-27:
I replaced the bootloader with the old one coming along with 4.19.118-v7+ from source: 
$ wget https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20200601-1_armhf.deb
$ sudo dpkg -i raspberrypi-bootloader_1.20200601-1_armhf.deb
dpkg: warning: downgrading raspberrypi-bootloader from 1.20200723-1 to 1.20200601-1
That does the trick!
I will do now a long term test.
Thanks a lot for quick support!

I also use "arm_freq_min=400", seems to be a popular setting.
I will try tomorrow commenting out that line and try again with bootloader 1.20200723-1.

jan-h-1970, 2020-07-27:
A first test shows, the combination of #arm_freq_min=400 and bootloader 1.20200723-1 works for me, too.

kiff777, 2020-08-02:
I tried commenting out arm_freq_min=300 in config and it didn't hang, but there seems to be another problem.

The performance of Ip transfer, which is a major element as a router, is considerably degraded as follows.

4.19.118-v7+
DL: 83.13 Mbps
UL: 83.95 Mbps

5.4.51-v7+
DL: 73.60 Mbps
UL: 65.05 Mbps

by, 2020-08-02:
@kiff777 : Don't you need at least arm_freq_min=1000 to guarantee the expected performance of Gigabit Ethernet? – BTW, underclocking the CPU apparently doesn't seem to conserve only a minimal amount of power.

kiff777, 2020-08-03:
@by : 
I use Pi2B as a router because the network service I contract at home is 100Mbps service.

The config has been as follows until now, but it seems that the CPU temperature is lower than doing nothing.
arm_freq_min=300
gpu_freq=200
sdram_freq=400
temp_limit=50

However, arm_freq_min=300 may not make much sense.

With the update from 4.19.118-v7+ to 5.4.51-v7+, I don't know the reason why Ip transfer throughput is so low, so I will keep it at 4.19.118-v7+ for a while.

4.19.118-v7+
DL: 83.13 Mbps
UL: 83.95 Mbps

5.4.51-v7+
DL: 73.60 Mbps
UL: 65.05 Mbps

I'm not good at English and use google translation, so I'm sorry it's hard to read.

popcornmix, 2020-08-04:
We have a workaround in latest rpi-update firmware that will disallow arm_freq_min below 600.
A fuller fix is being worked on but will need more work/testing so will wait until after the stable release is settled.

vmachiel, 2020-08-11:
> We have a workaround in latest rpi-update firmware that will disallow arm_freq_min below 600.
> A fuller fix is being worked on but will need more work/testing so will wait until after the stable release is settled.

But that fix will allow for arm_freq_min <600 again right?

popcornmix, 2020-08-11:
Yes, the future fix will allow lower frequencies again.
But the benefit (to either power or temperature) is negligible.

vmachiel, 2020-08-12:
Awesome thanks. I'll wait till it all appears via apt.

vmachiel, 2020-11-02:
Is the kernel version that’s out now the one that fixes this?

MichaIng, 2020-12-04:
Yes, the firmware packages version `1.20201022-1` have the issue mostly solved (only: on my board, `over_voltage` could not be lowered as much as before). I'm just trying out the today's revision `1.20201201-1`, as they might simply have lower ARM frequencies disabled: https://github.com/raspberrypi/firmware/issues/1431#issuecomment-732214981

... and yes indeed, `arm_freq_min` is now simply ignored 😢. I hope that is not a give-up on this topic as it already worked quite stable.