raspberrypi/linux #3758: 5.4.x kernel loses i2c-0 on Machine model: Raspberry Pi Model B Rev 1

dflogeras, 2020-07-26:
Hi, I'm running Gentoo on a bunch of RPi's of various denominations, which means I compile my kernel from source.

Today I tried to upgrade from tag raspberrypi-kernel_1.20200601-1 (last 4.19.x) to raspberrypi-kernel_1.20200723-1 (newly released 5.4.x).  This went well on all except my very first pi, a Model B Rev 1 (the first batch ones with 256MB RAM).

On this device, when it boots, i2c is missing. When I look under /sys/class/i2c-adapter, I see it is being shown as i2c-11, not as expected i2c-0.  Perhaps something missed in the device tree?  Same config works fine on a Model B Rev 2 (same vintage but with 512MB RAM).  I tried probing a known RTC chip on the "11" bus, but it doesn't find it.

I downgraded to 4.19.x and i2c works again as expected (even keeping the firmware blobs from 20200723).

The only things in my config.txt are

gpu_mem_512=15
gpu_mem_256=16
dtparam=i2c

I'm happy to compile patches if that helps.

Thank you.

pelwell, 2020-07-28:
Have you updated your firmware recently? A commit was added in April to give the Rev 1 Model B a dedicated DTB file (`bcm2708-rpi-b-rev1.dtb`). This was required to allow the I2C multiplexer to be added in the correct place - which is on I2C1 in this case.

dflogeras, 2020-07-29:
Ahaa, yes that was exactly it, totally operator error.

Related side-question:  is there a document anywhere on the naming scheme of the dts files?  I've never understood if the Pi (let's say zero-w) uses bcm2708-rpi-zero-w.* or bcm2835-rpi-zero-w.* or both (And similarly for the other boards).

pelwell, 2020-07-30:
Both sets of part numbers were used in Broadcom - BCM27xx are the bare dies, while BCM283x are the packaged parts with memory (where appropriate) - or I could have them backwards. When we developed Device Tree files for the Pis we used the `bcm27xx` naming scheme, then when the upstream devs added official support they chose `bcm283x` instead, which is quite convenient because it allows both sets of files to coexist.

If you go to kernel.org you'll only see the `bcm283x` files (apart from `bcm2711` - there is no `bcm2838`). Don't try mixing and matching a downstream and upstream kernels and dtbs - it probably won't work, and an incompatible dtb file can prevent booting.

dflogeras, 2020-07-31:
OK, that sounds pretty straight-forward.  I only every use these kernel sources, not upstream, so stick with the bcm27xy-* files.

Thanks!

closed 2020-07-31 by dflogeras