raspberrypi/linux #3760: Wifi Connection lost

ajseesink, 2020-07-27:
My Unifi network is scheduled to check WiFi channells for the best performce. 
Once my access points get the instruction to change channel, the Raspberry Pi (3B+) is loosing it's WiFi connection. 
The only thing I can do is reboot the Pi again. 

The Pi is the only device in the netwerk that looses its connection. All other wireless devices keep working. 

Current Raspberry Pi:
* Pi3B+, Buster 10.4
* /etc/rpi-issue 
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage5

* vcgencmd version
Jul 17 2020 11:04:27 
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)

* Kernel version Linux pi 5.4.51-v7+ #1326 SMP Fri Jul 17 10:45:25 BST 2020 armv7l GNU/Linux

pelwell, 2020-07-27:
That is interesting - it might help to explain a problem seen by another user.
* Which Unifi AP is this?
* Can you tell how often it checks the best channel?

ajseesink, 2020-07-27:
I have the Unifi AP-AC-PRO. 
I start the check every night at 3. Depending on the result it switches to another channel. I attached a screen print of it. 

I have created a workaround at the moment that checks the Wifi connection every 5 minutes. If no connection, then reboot.
Here's part of the logging from my check script:
No connecion on : do 16 jul 2020  3:05:01 CEST 
No connecion on : zo 19 jul 2020  3:05:01 CEST 
No connecion on : ma 20 jul 2020  3:05:03 CEST 
No connecion on : di 21 jul 2020  3:05:01 CEST 
No connecion on : wo 22 jul 2020  3:05:01 CEST 
No connecion on : ma 27 jul 2020  3:05:01 CEST

 

[image]