raspberrypi/linux #3761: 5.x+ kernel and LXC 3.0.3 doesn't work

bderenzo, 2020-07-27:
**Describe the bug**
With 5.x kernel there is a problem with LXC 3.0.3 (lastest version available in raspbian repository).

The error message is : 
```
Failed to setup limits for the "devices" controller. 
The controller seems to be unused by "cgfsng" cgroup driver or not enabled on the cgroup hierarchy
```

It seem's related to 
* https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=944389
* https://github.com/lxc/lxc/issues/2991
* https://github.com/lxc/lxc/issues/3183

**To reproduce**
* Install lxc
* Set `lxc.cgroup.memory.limit_in_bytes = 134217728`  in container configuration file.
* Start container

pelwell, 2020-07-27:
Same route cause as (so effectively a duplicate of) #3644.

closed 2020-07-27 by pelwell

bderenzo, 2020-07-27:
@pelwell Thanks for your quick answer and sorry for the duplicate.