raspberrypi/linux #3766: Unable to apply max98357a dtoverlay

timothyjward, 2020-07-29:
I'm running the Raspberry Pi Kernel 4.19.118 (specifically `Linux pi-middle-bedroom 4.19.118+ #1311 Mon Apr 27 14:16:15 BST 2020 armv6l GNU/Linux`) on my Pi Zero W. This includes an overlay for the MAXIM 98357A I2S DAC which can be enabled in config or via `dtoverlay`. Unfortunately this doesn't work (the overlay isn't active).

One issue seems to be that there is no kernel module for the `max98357a`. I downloaded the kernel sources and found that I  had to modify `sound/soc/codecs/Kconfig` to add an entry which selected `SND_SOC_MAX98357A` (the existing entry doesn't show up in menuconfig). I did try backporting the change from commit 2eb6074f15b4b61f11852e7944ca02aa77e95ff4 but that still didn't result in the kernel module being built.

Once I had successfully managed to compile and load the `SND_SOC_MAX98357A` kernel module I was able to play audio through my DAC, however the dtoverlay still doesn't apply (or show up as active). The error I see is simply:

    * Failed to apply overlay '0_max98357a' (kernel)"

I'm not sure how to debug this further. Although I can use my DAC, I would prefer to be able to configure it properly (e.g. to move the sd gpio output to a different pin). I would also like not to have to manually load the kernel module in `/etc/modules`

Reproducing this bug is easy, simply try to enable the dtoverlay using the `dtoverlay` command, or by setting `dtoverlay=max98357a` in `/boot/config.txt`. The result will be that the overlay is not active. Given that the overlay is part of the default distribution I would expect it to work properly.

pelwell, 2020-07-29:
A user submitted that overlay last year, not mentionimg that the module wasn't present. Most of the overlays we ship have been provided by users for devices that we don't have. Apart from checking that the overlays look sensible, compile, apply, and have an entry in the README and Makefile, we don't do any other testing. Automated checking that there is a supporting driver is an interesting idea, but it isn't there at the moment.

The rpi-5.4.y kernel includes the required module, and the overlay applies without errors, but I can't test it beyond that.

timothyjward, 2020-07-29:
> The rpi-5.4.y kernel includes the required module, and the overlay applies without errors, but I can't test it beyond that.

This is good news - I'm stuck on 4.19.y for now, but moving to 5.4.y is a possibility in a few months.

Do you have any debugging advice for me if I keep trying to get it working while I'm using 4.19.y?