raspberrypi/linux #3767: kernel 5.4.x breaks backlight control on PiTFT(STMPE610)

vorbeiei, 2020-07-30:
**Describe the bug**
After the update of the Raspberry Pi OS to kernel 5.4 (from 4.19) the backlight control no longer works. When trying to turn on/off the backlight as described [here](https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/backlight-control#on-slash-off-using-stmpe-gpio-2859122-5) the backlight simply stays on with no error output.

This issue seems to be fixed [in kernel 5.5 by this commit.](https://github.com/raspberrypi/linux/commit/706dc68102bc7421a9e6573d149ab6d769d71cc7#diff-1cefba26c5d09fe4fcd3c13424a2376a) Compiling the current rpi-5.4.y branch with the gpio-backlight module from rpi-5.5.y results in the backlight control working again.

**System**

* Which model of Raspberry Pi?  
        Raspberry Pi 2B
* Which OS and version (`cat /etc/rpi-issue`)?   
        Raspberry Pi reference 2020-02-13
* Which firmware version (`vcgencmd version`)?
        Jul 17 2020 11:04:27 
        Copyright (c) 2012 Broadcom
        version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)
* Which kernel version (`uname -a`)?
        Linux raspberrypi 5.4.51-v7+ #1 SMP Thu Jul 30 15:26:28 CEST 2020 armv7l GNU/Linux(mind you that this is my own       build of the latest rpi-5.4.y branch with a fix applied)

pelwell, 2020-07-30:
That commit has been back-ported into rpi-5.4.y (42a52f1a) and will be in the next firmware release.

popcornmix, 2020-07-31:
Please test latest rpi-update kernel and report if issue is resolved

vorbeiei, 2020-07-31:
Yes, works fine. Thanks.

closed 2020-07-31 by vorbeiei