raspberrypi/linux #3768: Pi unable to use VHT with built in WiFi chip

AleXSR700, 2020-08-01:
Hi everyone,

I set up my internal wifi to act as an AP according to https://www.raspberrypi.org/documentati ... bridged.md

I then adjusted to run on 5 Ghz network and wanted to ensure high transfer rates so I wanted to set it to ac mode and use the 80 Mhz channels.

But as soon as I set
vht_oper_chwidth=1
my AP does not start anymore. With otherwise exactly the same settings but without the vht_oper_chwidth line it works, but I get a maximum of 72 Mbps.
I would really like to try and achieve 450 or 867 MBps here. This should be possible as far as the specs show.

Is there a bug in the kernel or something?

I am running a Raspberry Pi 4 Model B with 8 GB of RAM.

Could someone help me out here?

Thank you
Alex

AleXSR700, 2020-08-04:
Update:
Any setting to do with VHT leads to an inability of the AP to start.
I even tried setting the country code to US. But still not working.

There seems to be a bug on the RPi 4.

Please also see my response to a similar problem on the RPi 3. But the solutions stated there do not seem to work on the RPi4.

https://github.com/raspberrypi/linux/issues/2619

pelwell, 2020-08-05:
There is an alternative clm_blob - not yet part of the standard release - that may help. You can download it from here: https://drive.google.com/file/d/1Qoc90FCTO17d69PbBqUhkJKgqDMmdOui/view?usp=sharing

Install it using (be careful with the tricky punctuation in the first line - it saves a lot of typing):
```
$ sudo cp /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob{,.orig}
$ sudo cp 43455_raspberry_3p_v1_190515.clm_blob /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
```
Revert to the old one with:
```
$ sudo mv /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob{.orig,}
```

AleXSR700, 2020-08-06:
> There is an alternative clm_blob - not yet part of the standard release - that may help. You can download it from here: [...]

Thank you for your reply.
I gave it a shot but unfortunately it did not change anything. Still not working.

I am now trying my external wifi dongle as AP (8822bu) but I don' think that will work either. I am currently resetting everything and starting from scratch.
Seems although 5 Ghz has been around and standard for a long time, the RPi is not good at handling it.

AleXSR700, 2020-08-07:
I think that there is something wrong with the driver for the built in wifi chip.

It is unable to properly connect to a 5 Ghz ac network. The driver also does not support monitor mode.
If I have wlan0 and wlan1 active and they both connect to the same hotspot (which is a pure 5 Ghz hotspot), only the external wlan1 will actually connect properly using ac mode and reach the full connection speed.

```
wlan0     IEEE 802.11  ESSID:"Network1"  
          Mode:Managed  Frequency:5.18 GHz  Access Point: xx:FF:4D:EB:xx:xx   
          Bit Rate=200 Mb/s   Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
          Link Quality=70/70  Signal level=-25 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan1     IEEE 802.11AC  ESSID:"Network1"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:5.18 GHz  Access Point: xx:FF:4D:EB:xx:xx   
          Bit Rate:867 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=75/100  Signal level=72/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

AleXSR700, 2020-08-07:
```
pi@piclient:/proc/sys/net/ipv4 $ sudo ifconfig wlan0 down
pi@piclient:/proc/sys/net/ipv4 $ sudo iwconfig wlan0 mode monitor
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Operation not supported.
pi@piclient:/proc/sys/net/ipv4 $ 
```

pelwell, 2020-11-18:
There's now an updated clm_blob that should give access to the 80MHz channels: https://drive.google.com/file/d/1AN7lC_kMJGGg5AJLhSRtlTRgIh9qJlaI/view?usp=sharing

rmaksimov, 2021-01-24:
On the latest Raspberry OS (`Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux`) using the config mentioned here https://github.com/raspberrypi/linux/issues/2619#issuecomment-410703338, hostapd and the driver work as expected out of the box

The firmware-brcm80211 version is
```
ii  firmware-brcm80211             1:20190114-1+rpt10                  all          Binary firmware for Broadcom/Cypress 802.11 wireless cards
```

The hostapd version is
```
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
```

Also it works perfectly on the testing branch of Raspberry OS

But I had the same issue with setting up an AP on both Debian Buster and Bullseye (https://raspi.debian.net/)
```
Linux rpi4-20201112 5.9.0-0.bpo.5-arm64 #1 SMP Debian 5.9.15-1~bpo10+1 (2020-12-31) aarch64 GNU/Linux
```

The following errors occur every time the hostapd service starts
```
nl80211: Scan trigger failed: ret=-52 (Invalid exchange)
```
and
```
Failed to request a scan of neighboring BSSes ret=-52 (Invalid exchange)
```

Also there are some error messages in dmesg
```
[   25.812004] ieee80211 phy0: brcmf_cfg80211_start_ap: Set Channel failed: chspec=57386, -52
[   86.433605] ieee80211 phy0: brcmf_run_escan: error (-52)
[   86.436688] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[   86.443699] ieee80211 phy0: brcmf_run_escan: error (-52)
[   86.446806] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[  211.466640] ieee80211 phy0: brcmf_run_escan: error (-52)
[  211.469720] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
[  211.478222] ieee80211 phy0: brcmf_run_escan: error (-52)
[  211.481293] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)
```

The problem is related to a lack of the .clm_blob file
```
[   10.291641] brcmfmac mmc0:0001:1: firmware: failed to load brcm/brcmfmac43455-sdio.clm_blob (-2)
[   10.358229] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
...
[   11.878522] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
```

So, to resolve the problem you can use one of the following links to get the necessary file (which are the same)
```
https://github.com/raspberrypi/linux/issues/3768#issuecomment-729642147
```

or download it from a cypress.com website 
```
https://community.cypress.com/t5/Resource-Library/Cypress-Linux-WiFi-Driver-Release-FMAC-2020-09-25/ta-p/251089
(direct link: https://community.cypress.com/gfawx74859/attachments/gfawx74859/resourcelibrary/1030/1/cypress-fmac-v5.4.18-2020_0925.zip)
```
decompress it and copy the file to `/lib/firmware/brcm/` directory
```
unzip cypress-fmac-v5.4.18-2020_0925.zip cypress-firmware-v5.4.18-2020_0925.tar.gz
tar xzf cypress-firmware-v5.4.18-2020_0925.tar.gz firmware/cyfmac43455-sdio.clm_blob
cp firmware/cyfmac43455-sdio.clm_blob /lib/firmware/brcm/brcmfmac43455-sdio.clm_blob
```