raspberrypi/linux #3770: Cannot enable. Maybe the USB cable is bad?

hardy2408, 2020-08-02:
I hope it's the right place to report this bug, IMHO.

**Describe the bug**
I'm getting this error message after I issued the command 
`sudo apt-get upgrade`

**To reproduce**
I'm running an RPi 3 and installed a fresh buster raspbian.
After installing a few packages for setting up a web server with wordpress I issued the command
`sudo apt-get upgrade`
After a reboot the system boot was stopping completely. 

**Expected behaviour**
Run as before.
Currently I cannot upgrade the system. I never had that during 7 years working with a RPi.

**Actual behaviour**
`pi@raspberrypi3:~ $ sudo apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
The following packages will be upgraded:
  ca-certificates firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek
  libgnutls30 libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 raspberrypi-bootloader
  raspberrypi-kernel raspi-config rpi-eeprom rpi-eeprom-images
16 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 127 MB of archives.
After this operation, 6,477 kB disk space will be freed.
Do you want to continue? [Y/n]`

Error messages while booting:
`usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
usb usb1-port1: attempt power cycle`

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW: Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?
pi@raspberrypi3:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
pi@raspberrypi3:~ $ vcgencmd version
Apr 15 2020 11:43:08
Copyright (c) 2012 Broadcom
version 82f9bb929ce2186eb1824178c1ae82902ad6275c (clean) (release) (start)
pi@raspberrypi3:~ $ uname -a
Linux raspberrypi3 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
It seems that the code is written in 
https://github.com/raspberrypi/linux/blob/1c64f4bc22811d2d371b271daa3fb27895a8abdd/drivers/usb/core/hub.c#L2952

P33M, 2020-08-02:
Is the failure repeatable? What happens if you re-use a fresh SD card with a download of current Raspberry Pi OS and reboot instead of power cycling?
It's unlikely that an apt package update caused USB to stop working on Pi 3s.

hardy2408, 2020-08-03:
Yes, it's totally reproducible, I did it a few times. Without the upgrade
command it works now, the server is stable.
I tried it also with different hard drives and had the same effect.
What do you mean with power cycle?

Rebooting with a fresh raspbian download works. The problem comes with the
upgrade command.

I can't test it on an SD card as the RPi 3 is configured to boot from
external USB card. This is a one time configuration. You can't reverse it.

Regards

    Gerhard

pelwell, 2020-08-03:
"Power cycle" is just another way to say "turn off and on again", which it sounds like you will have done already.

P33M, 2020-08-03:
I've performed the same steps that I think you've taken, and my Pi 3 B reboots successfully with latest packages updated.
- Image an SD card with Raspberry Pi OS
- Boot a Pi 3 B with this SD card, then follow the guide for enabling USB mass storage boot: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md
- Shut down the Pi, then move the SD card to a USB-SD adapter. Boot from USB.
- Run `sudo apt update && sudo apt upgrade` and reboot
- Pi boots to login prompt.

Does your Pi's red LED (PWR) ever go out or flicker when rebooting?

hardy2408, 2020-08-03:
Hi,

These steps I did, yes, long ago.

You mean, if the red led is flickering now? Because now I did not issue the
*upgrade* command.

I needed my live system back.
Now I ordered a pi 4. I can only reproduce it when my new system is able to
run.

Regards

     Gerhard

P33M, 2020-08-05:
On your 3B+, can you test rpi-update firmwares in turn, and report the first failing one?

Do `sudo rpi-update <hash>` where <hash> is found in this list of commits: https://github.com/Hexxeh/rpi-firmware/commits/master

Start at e1050e94821a70b2e4c72b318d6c6c968552e9a2 (which should be the firmware in the shipped image) and try increasingly recent versions. To speed up the process, try a50c7d5eebb351d16665eabcedad992cdc167537 first as it's halfway between latest and last-shipped.

hardy2408, 2020-08-05:
I understand the process.
Give me some days as I'm quite busy and want to use a new hard drive as a
backup/test device.

Regards

    Gerhard

hardy2408, 2020-08-11:
Hi,

I did as advised.

e1050e94821a70b2e4c72b318d6c6c968552e9a2 from 27.04.2020: successfully
rebooted
a50c7d5eebb351d16665eabcedad992cdc167537 from 10.06.2020 : successfully
rebooted
70598414a73afab7f8e521c358a7cfd5ffb65d3e from 17.07.2020: successfully
rebooted
ffc3ed09c14f0b829dd3ca00ed72a6eaa1abfd72 from 29.07.2020: successfully
rebooted
668419cb9876474053574b2b1dc20ca996e14cc8 from 30.07.2020: successfully
rebooted
d764cbe6dda70aa1e05d00b0dff20d7f65f210d7 from 31.07.2020: successfully
rebooted
ce0d21fbb73fd788b5b6409105b2523aebbf235c from 04.08.2020: successfully
rebooted
97d1bef547bdaba10ede86192f39ce3e83a54a76 from 04.08.2020: successfully
rebooted

Any further advice?

I understand that 16 packages were upgraded:
The following packages will be upgraded: ca-certificates firmware-atheros
firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek
libgnutls30 libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc
libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel raspi-config
rpi-eeprom rpi-eeprom-images 16 upgraded, 0 newly installed, 0 to remove
and 0 not upgraded.

About which one(s) are we talking with the rpi-update command?

By the way, in the meantime I have configured a new system with a RPi4 +
SSD, and the sudo update && upgrade worked smoothly.
Means also that I have a test system now ;-)

Regards

     Gerhard

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virenfrei.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

hardy2408, 2020-08-16:
I executed the same command `sudo apt-get upgrade` today again and it works now

closed 2020-08-16 by hardy2408