raspberrypi/linux #3771: udev-rule referenced to a wrong device since last kernel release

Blackbox-git, 2020-08-02:
Hi, guys.

I have a reproducible problem on Raspberry Pi since the last kernel release.

I have written a UDEV rule which includes an original Arduino as symlink /dev/keypad.

In the old kernel version (Raspbian Pi OS from 2020-05-27) the symlink is correctly referenced for example to "/dev/ttyUSB0". But in the new version it is referenced to "/dev/gpiochip1".

If you need more Information, i will get them for you.

pelwell, 2020-08-02:
Of course we need information - start with the udev rule.

Blackbox-git, 2020-08-02:
This is my original Arduino nano:

`lpi@raspberrypi:~$ lsusb` 
**` Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC`** 
` Bus 001 Device 006: ID 1781:24ab Multiple Vendors` 
` Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter` 
` Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub` 
` Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub` 

` pi@raspberrypi:~ $ lsusb -vs 001:004` 
` Bus 001 Device 004: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC`
`Device Descriptor:`
  **`idVendor           0x0403 Future Technology Devices International, Ltd` 
`  idProduct          0x6001 FT232 Serial (UART) IC`** 
 ` bcdDevice            6.00` 
`  iManufacturer           1 FTDI` 
`  iProduct                2 FT232R USB UART` 
 **` iSerial                 3 A600MYLH`** 
`    MaxPower               90mA` 
`    Interface Descriptor:` 
`      bLength                 9` 
  `     bInterfaceClass       255 Vendor Specific Class` 
 `      bInterfaceSubClass    255 Vendor Specific Subclass` 
  `     bInterfaceProtocol    255 Vendor Specific Protocol` 
  `     iInterface              2 FT232R USB UART` 
`lpi@raspberrypi:~$` 

here is the content of the rule (the rule is stored in the directory "/lib/udev/rules/91-keypad.rules"):

`
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="A600MYLH", SYMLINK+="keypad", ACTION=="add"
`

The rule will be execute and the symlink "/dev/keypad" is created in the old and the new kernel. The difference is depence on where to which device the rule refrence.

Here is the listing of the old kernel:

`pi@raspberrypi:/dev $ uname -a `
 `Linux raspberrypi 4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020 armv7l GNU/Linux `
 `pi@raspberrypi:/dev $ ls -l /dev/keypad `
 **`lrwxrwxrwx 1 root root 7 Aug  2 09:56 /dev/keypad -> ttyUSB0 `**

If i do sudo apt update && sudo dist-upgrade the listiting looks like that:

`pi@raspberrypi:/dev $ uname -a`
`Linux raspberrypi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux`
`pi@raspberrypi:/dev $ ls -l /dev/keypad`
**`lrwxrwxrwx 1 root root 9 Aug  2 10:14 /dev/keypad -> gpiochip1`**

You can see the refrence is changed to "/dev/ gpiochip1". You need more information?

pelwell, 2020-08-02:
I can reproduce the problem (although the link is to gpiochip2 here). Leave it with us.

pelwell, 2020-08-02:
There's a typo in your rule (`SUBSYSTEMS`) and an incorrect value (`usb`) - try this:
```
SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="A600MYLH", SYMLINK+="keypad", ACTION=="add"
```
And I think it makes more sense with ACTION first, although it makes no difference:
```
ACTION=="add",SUBSYSTEM=="tty", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6001", ATTRS{serial}=="A600MYLH", SYMLINK+="keypad"
```

Blackbox-git, 2020-08-02:
hello @pelwell,

thanks for your answer. The adjustment of the udev-rule works for me.

So i think its no kernel issue. The problem is in front of the monitor. I just wonder why it worked before. 

You made my day.

cheers

closed 2020-08-02 by Blackbox-git

pelwell, 2020-08-02:
The effect of the mistakes is to allow the rule to match more events. I suspect the order of events has changed with the new kernel, so that what used to work no longer does.