raspberrypi/linux #3774: I must press CTRL+C when logging in to the Raspberry Pi system using SSH.

Dolphinzzx, 2020-08-03:
* Board: Raspberry Pi4
* Image version: Raspberry Pi reference 2020-05-27
* kernel version: Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux

**Describe the bug**
When I remotely log in to the Raspberry Pi system, it will freeze at this position.
As shown below.
[image]

Connecting to 192.168.1.244:22...
Connection established.
To escape to local shell, press 'Ctrl+Alt+]'.
Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Aug  3 11:02:40 2020 from 192.168.1.119

Then, i need to pree ctrl+ c to stop this process. Finally, i can log into Raspberry Pi system.
[image]

Connecting to 192.168.1.244:22...
Connection established.
To escape to local shell, press 'Ctrl+Alt+]'.
Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.
Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Aug  3 11:02:40 2020 from 192.168.1.119
^Cpi@raspberrypi:~ $

kripton, 2020-08-03:
I am a bit unsure about the `XShell` and `NetSarang` prompts. Is this some software running on your host computer? Would you mind trying another software, such as PuTTY (assuming you are on Windows, https://www.putty.org/), to connect to the Pi via the SSH protocol?

pelwell, 2020-08-03:
Something must have been added to one of the login scripts - `.bashrc` or `.profile` - that is not terminating. Ctrl-C kills it and allows you to start interacting with the shell.

This isn't a Linux issue and it's not a Raspberry Pi OS issue - something else has caused this.

closed 2020-08-03 by pelwell