raspberrypi/linux #3777: official 7-inch touchscreen polling rate too low

ardera, 2020-08-06:
**Describe the bug**
The rate of input events arriving at userspace when using the Official 7inch touchscreen is too low. Input events arrive at user space at about 33Hz, resulting in perceived lagginess of touch applications, even though the application is running at 60fps.

This bug was present once before, see #3227, so this is kinda a regression. I'm not sure exactly what commit/update caused it to go back down to 33Hz though.

**To reproduce**
See #3227 

**Expected behaviour**
Expected is at least 60Hz, possibly even higher (since touchscreen controller seems to support 100Hz, and higher polling rates result in less input lag. All iPhones past iPhone X for example poll at 120Hz, some android gaming phones at 240Hz)

**Actual behaviour**
The frequency of input events arriving at userspace is 33Hz.

**System**
```
System Information
------------------

Raspberry Pi 4 Model B Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

Linux hpi4 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux
Revision        : b03111
Serial          : 100000001afe70ee
Model           : Raspberry Pi 4 Model B Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jul 17 2020 10:59:17
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       30391116 8174312  20946604  29% /
devtmpfs          732824       0    732824   0% /dev
tmpfs             864920       0    864920   0% /dev/shm
tmpfs             864920    8960    855960   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             864920       0    864920   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   54461    203634  22% /boot
tmpfs             172984       0    172984   0% /run/user/1000

Filename                                Type            Size    Used    Priority
/var/swap                               file            102396  0       -2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installed: 1.20200611
raspberrypi-sys-mods:
  Installed: 20200729
openbox:
  Installed: 3.6.1-8+rpt4
lxpanel:
[... 572 lines cut ...]
raspberrypi-ts: Latest    33Hz, Average    28Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    16Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    28Hz
raspberrypi-ts: Latest    33Hz, Average    28Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    30Hz
raspberrypi-ts: Latest    16Hz, Average    30Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    29Hz
raspberrypi-ts: Latest    33Hz, Average    30Hz
raspberrypi-ts: Latest    33Hz, Average    30Hz
raspberrypi-ts: Latest    33Hz, Average    30Hz
raspberrypi-ts: Latest    16Hz, Average    30Hz
```

**kinda offtopic**
Is there any way the (actual) touchscreen driver can be moved out of the firmware into the linux kernel, and be modified to use interrupts if the user wants so? The delay is pretty bad to be honest, less input lag would make the Raspberry Pi actually very usable for modern UI (using flutter and my flutter-pi project).

ardera, 2020-09-20:
Any news on this? Basically every other generic touchscreen is working better right now (both in terms of polling rate and latency) than this officially supported one.

6by9, 2020-09-21:
5.4 has adopted the mainline driver from drivers/input/touchscreen/raspberrypi-ts.c rather than the downstream drivers/input/touchscreen/rpi-ft5406.c

It appears to have defined [60fps polling](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/drivers/input/touchscreen/raspberrypi-ts.c#L31), but it offloads the poll process to the framework by the looks of it.

There is one commit in later kernels (>=5.5) that changes this driver, but I haven't looked to see if the required framework change are in 5.4 or not. 
https://github.com/raspberrypi/linux/commit/bd88ce25335d23a9967e6d3d1efdc320dbd0a3a4

ardera, 2020-09-22:
I can confirm it seems like its the linux-side again, the firmware-side is polling the TS at approx 57Hz according to my i2c sniffer.

The intervals between events are most commonly 30ms, sometimes 60ms and very rarely 40ms.

> There is one commit in later kernels (>=5.5) that changes this driver, but I haven't looked to see if the required framework change are in 5.4 or not.
> [bd88ce2](https://github.com/raspberrypi/linux/commit/bd88ce25335d23a9967e6d3d1efdc320dbd0a3a4)

I did a little bit of research and it seems the actual implementation didn't change much. `input_dev_poller` looks basically exactly the same as `input_polled_device` (/ `input_polldev`). They still use the same mechanism for the delaying the work / polling, and they have exactly the same delay calculation.

popcornmix, 2020-09-22:
While the driver is different, the cause of the slow polling may be the same as https://github.com/raspberrypi/linux/issues/3227#issuecomment-532265372

6by9, 2020-09-22:
I've had a very brief look at the framework for input_poller and it looks plausible. It's using ```queue_delayed_work``` to retrigger every poll period.
AFAIK that is accurate at <20ms, although it does appear to be using timers (not hrtimers) for sleeping, so it may well be the same issue. That ought to be reported upstream as it surely affects all input devices using that input-poller mechanism.

ardera, 2020-09-22:
> That ought to be reported upstream as it surely affects all input devices using that input-poller mechanism.

Do you want to do that? I can do it, if you want me to. I kinda have the feeling though they'll say we should just increase the `HZ`.

6by9, 2020-09-23:
@pelwell Do you have a feel for the delay accuracy on ```queue_delayed_work```?
```HZ``` appears to be set to 100 for our configs, so a 10ms tick. That would give a moderate level of inaccuracy on the polling, but I wouldn't expect so close to 30Hz.
I find it odd that upstream have switched to this polling mechanism on a number of drivers if the accuracy isn't there on the poll rate.

> Is there any way the (actual) touchscreen driver can be moved out of the firmware into the linux kernel, and be modified to use interrupts if the user wants so? The delay is pretty bad to be honest, [...]

There is some work in mainline to write drivers for the Toshiba TC358752 DSI to DPI bridge and the touchscreen controller. One issue is that they share an I2C controller with the camera module, so if you shift the touchscreen to the kernel then you can not use the firmware camera stack (you can use libcamera which is all ARM side).

pelwell, 2020-09-24:
I'd be tempted to create a delayed_work item that just requeues itself 1 jiffy later for 100 iterations, and log the times.

ardera, 2020-09-24:
So, according to [this doc on linux work queues](http://books.gigatux.nl/mirror/kerneldevelopment/0672327201/ch07lev1sec4.html), `queue_delayed_work` waits for _at least_ `delay` jiffies. That means `queue_delayed_work` alone is introducing 10-20ms of latency. The polling work is executed right after a kernel tick ocurred, and then `queue_delayed_work` with the polling work is executed. `queue_delayed_work` then needs to wait for 2 kernel ticks, since the dt between the invocation of `queue_delayed_work` and the first kernel tick is not _at least 10ms_ (more like 9.99ms). That means the delay introduced by `queue_delayed_work` alone is most likely closer to 20ms.

I don't know where the other 10ms come from. Maybe the worker thread is not directly woken up by the work being queued after these 20ms, but woken up by the scheduler on the next kernel tick?

Either way, I'm amazed by how inaccurate `input_polled_dev` is. I mean not only the problem we have right now, but generally, using relative timing/delays to implement reocurring work is a bad idea. If the actual poll would be a bit more time consuming, the timing would be off. Absolute timing works better in most cases. Or did I miss something?

pelwell, 2020-09-24:
The kernel devs have been introducing sleep ranges and other "inaccuracies", allowing power savings by batching up activity to increase the length of the low power states. This could be consequence of those changes.

ardera, 2020-09-24:
> The kernel devs have been introducing sleep ranges and other "inaccuracies", allowing power savings by batching up activity to increase the length of the low power states. This could be a consequence of those changes.

Could be. Though I just looked it up and `input-polldev` was introduced in 2007 in 2.6.22 and hasn't changed much since.

ardera, 2021-12-17:
just for completeness: with the new `edt-ft5x06`  driver the polling rate is still pretty low, about 33Hz at max:

```
pi@hpi4:~/devel/evhz $ sudo ./evhz 
Press CTRL-C to exit.

event0: Razer Razer Viper
event1: Razer Razer Viper Keyboard
event2: Razer Razer Viper Consumer Control
event3: Razer Razer Viper System Control
event4: Razer Razer Viper
event5: Razer Razer Viper
event6: vc4
event7: vc4
event8: generic ft5x06 (00)

generic ft5x06 (00): Latest    33Hz, Average     0Hz
generic ft5x06 (00): Latest    33Hz, Average    16Hz
generic ft5x06 (00): Latest    33Hz, Average    22Hz
generic ft5x06 (00): Latest    33Hz, Average    24Hz
generic ft5x06 (00): Latest    33Hz, Average    26Hz
generic ft5x06 (00): Latest    33Hz, Average    27Hz
generic ft5x06 (00): Latest    33Hz, Average    28Hz
generic ft5x06 (00): Latest    33Hz, Average    28Hz
generic ft5x06 (00): Latest    16Hz, Average    29Hz
generic ft5x06 (00): Latest     1Hz, Average    28Hz
generic ft5x06 (00): Latest    16Hz, Average    25Hz
generic ft5x06 (00): Latest    33Hz, Average    24Hz
generic ft5x06 (00): Latest    33Hz, Average    25Hz
generic ft5x06 (00): Latest    33Hz, Average    25Hz
generic ft5x06 (00): Latest    33Hz, Average    26Hz
generic ft5x06 (00): Latest    33Hz, Average    26Hz
generic ft5x06 (00): Latest    33Hz, Average    27Hz
generic ft5x06 (00): Latest    33Hz, Average    27Hz
```

dlukach, 2022-01-12:
Any updates on this issue? This issue is making it hard to use the pi display for our product. Even button click events cause a lag that is hard to not notice

lbt, 2022-04-06:
based on @ardera's comment about rounding up the jiffies I changed
RPI_TS_POLL_INTERVAL to 7 [in raspberrypi-ts.c](https://github.com/raspberrypi/linux/blob/rpi-5.15.y/drivers/input/touchscreen/raspberrypi-ts.c#L30) for 5.15.32 and got:

raspberrypi-ts: Latest    50Hz, Average    47Hz
raspberrypi-ts: Latest    50Hz, Average    47Hz
raspberrypi-ts: Latest    25Hz, Average    46Hz
raspberrypi-ts: Latest    50Hz, Average    46Hz
raspberrypi-ts: Latest    50Hz, Average    47Hz

Then setting HZ to 250 in the config:

raspberrypi-ts: Latest    50Hz, Average    56Hz
raspberrypi-ts: Latest    62Hz, Average    56Hz
raspberrypi-ts: Latest    62Hz, Average    56Hz
raspberrypi-ts: Latest    50Hz, Average    56Hz
raspberrypi-ts: Latest    41Hz, Average    56Hz
raspberrypi-ts: Latest    83Hz, Average    56Hz
raspberrypi-ts: Latest    35Hz, Average    56Hz
raspberrypi-ts: Latest   125Hz, Average    57Hz
raspberrypi-ts: Latest    50Hz, Average    57Hz

This led to VASTLY improved gesture support on the screen with Qt/QML code

Of course the correct solution is to have the input-poller code get the timing correct but this solution may help others