raspberrypi/linux #3782: Kernel panic, bcm2836_arm_irqchip_handle_irq

vasyugan, 2020-08-10:
Almost each night, when my raspi receives btrfs snapshot backups from a remote server, it is having kernel panics, 

See screenshot below.

[image]
[image]

**System**
System Information
------------------

Raspberry Pi 3 Model B Rev 1.2\00
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2016-02-26
Generated using Pi-gen, https://github.com/RPi-Distro/Pi-gen, stage4

Linux raspbian 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux
Revision	: a02082
Serial		: 00000000cdcc9893
Model		: Raspberry Pi 3 Model B Rev 1.2
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Jul 17 2020 11:04:27 
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Dateisystem     1K-Blöcke    Benutzt  Verfügbar Verw% Eingehängt auf
/dev/root         7344832    3159444    3854196   46% /
devtmpfs           441236          0     441236    0% /dev
tmpfs              474516         92     474424    1% /dev/shm
tmpfs              474516      13464     461052    3% /run
tmpfs                5120          4       5116    1% /run/lock
tmpfs              474516          0     474516    0% /sys/fs/cgroup
/dev/mmcblk0p1     258096      54460     203636   22% /boot
/dev/sda1      5858423808 3966408376 1886047912   68% /var/backups
/dev/sda1      5858423808 3966408376 1886047912   68% /var/log
/dev/sda1      5858423808 3966408376 1886047912   68% /var/www
/dev/sda1      5858423808 3966408376 1886047912   68% /root
/dev/sda1      5858423808 3966408376 1886047912   68% /tmp
/dev/sda1      5858423808 3966408376 1886047912   68% /var/lib/mysql
/dev/sda1      5858423808 3966408376 1886047912   68% /var/cache
/dev/sda1      5858423808 3966408376 1886047912   68% /home
/dev/sda1      5858423808 3966408376 1886047912   68% /srv
tmpfs               94900          0      94900    0% /run/user/0
tmpfs               94900          0      94900    0% /run/user/1000

Filename				Type		Size	Used	Priority
/dev/sda2                               partition	2097148	776	-2

Package version information
---------------------------
raspberrypi-ui-mods:
  Installiert:           (keine)
raspberrypi-sys-mods:
  Installiert:           20200514
openbox:
[... 540 lines cut ...]
[   41.746019] uart-pl011 3f201000.serial: no DMA platform data
[   41.839232] 8021q: 802.1Q VLAN Support v1.8
[   43.586544] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   44.338442] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   45.494387] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   45.800787] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
[   46.097430] Bluetooth: Core ver 2.22
[   46.097543] NET: Registered protocol family 31
[   46.097553] Bluetooth: HCI device and connection manager initialized
[   46.097586] Bluetooth: HCI socket layer initialized
[   46.097604] Bluetooth: L2CAP socket layer initialized
[   46.097635] Bluetooth: SCO socket layer initialized
[   46.111882] Bluetooth: HCI UART driver ver 2.3
[   46.111906] Bluetooth: HCI UART protocol H4 registered
[   46.112055] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   46.112373] Bluetooth: HCI UART protocol Broadcom registered
[   46.615484] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   46.615512] Bluetooth: BNEP filters: protocol multicast
[   46.615552] Bluetooth: BNEP socket layer initialized
[   46.962929] ICMPv6: process `dhcpcd' is using deprecated sysctl (syscall) net.ipv6.neigh.wlan0.retrans_time - use net.ipv6.neigh.wlan0.retrans_time_ms instead
[   53.054846] NFSD: Using UMH upcall client tracking operations.
[   53.054870] NFSD: starting 90-second grace period (net f00004c1)
[   53.085964] tun: Universal TUN/TAP device driver, 1.6
**Logs**
If applicable, add the relevant output from `dmesg` or similar.

vasyugan, 2020-08-10:
I have now downgraded the kernel to the package of 20200717, and so far it seems like the kernel panic doesn't occur there. Which would match the chronology, when this problem started occuring

vasyugan, 2020-08-10:
hm, what really seems to be dragging the raspi kernel down is pbzip2. When it receives plain btrfs subvolumes, it doesn't crash, but when it receives pbzip2 compressed subvolumens which it has to decompress, it crashes.

pelwell, 2023-10-25:
I don't think there's any useful information in that photo - it just shows the other cores not stopping, not the initial cause of the problem.