raspberrypi/linux #3785: Please enable CONFIG_UEVENT_HELPER for 5.4.y

srcshelton, 2020-08-12:
**Describe the bug**
`mdev` is the minimal, lightweight device manager from BusyBox.  In order to be able to respond to hot-plug events, it requires the kernel be configured with `CONFIG_UEVENT_HELPER=y`.  If this could be enabled in the stock kernel then I wouldn't have to build a new kernel from source simply to gain this functionality.

Thanks for your consideration!

pelwell, 2020-08-18:
The relevant Kconfig file says:
```
config UEVENT_HELPER
	bool "Support for uevent helper"
	help
	  The uevent helper program is forked by the kernel for
	  every uevent.
	  Before the switch to the netlink-based uevent source, this was
	  used to hook hotplug scripts into kernel device events. It
	  usually pointed to a shell script at /sbin/hotplug.
	  This should not be used today, because usual systems create
	  many events at bootup or device discovery in a very short time
	  frame. One forked process per event can create so many processes
	  that it creates a high system load, or on smaller systems
	  it is known to create out-of-memory situations during bootup.
```
which doesn't sound very appealing. However, it also says:
```
config UEVENT_HELPER_PATH
	string "path to uevent helper"
	depends on UEVENT_HELPER
	default ""
	help
	  To disable user space helper program execution at by default
	  specify an empty string here. This setting can still be altered
	  via /proc/sys/kernel/hotplug or via /sys/kernel/uevent_helper
	  later at runtime.
```
which suggests that it will do nothing by default unless something writes to one of those pseudo-files.

The code in kobject_uevent_env checks for a non-NUL first character of uevent_helper before starting any additional processing, which is a fairly low overhead on each uevent.

Any objections, @popcornmix?

popcornmix, 2020-08-18:
It sounds okay if it's dormant by default.

pelwell, 2020-08-18:
Thanks - see 9bbdaa0.

closed 2020-08-19 by pelwell