raspberrypi/linux #3786: pizero kernel does not build after update

Claudenw, 2020-08-12:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**

From the build:
```
drivers/net/usb/lan78xx.c: In function 'lan78xx_probe':
drivers/net/usb/lan78xx.c:3777:15: error: 'struct lan78xx_net' has no member named 'ep_intr'
   period = dev->ep_intr->desc.bInterval;
               ^
```

It looks like ep_intr was removed from the lan78xx.net structure.  However line 3777 of lan78xx.c requires it.  Not sure what should be done here.

**To reproduce**

```
git clone --depth=1 --branch rpi-${version}.y https://github.com/raspberrypi/linux.git
export PATH=$PATH:/opt/tools/arm-bcm2708/arm-linux-gnueabihf/bin 
export KERNEL=kernel 
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- bcmrpi_defconfig 
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -j 4 zImage modules dtbs 

**Expected behaviour**
The make should succeed

**Actual behaviour**
the make failed with error noted above

**System**

Compiling for PIZeroW.
cross compiling on
```
$uname -a
Linux zoltan-the-great 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

Claudenw, 2020-08-12:
Checking out 4b945d5fb69233f4ddf66d67c3d35145f1039e01 the build works so only the changes in a6825094038507f5eafdca6328ac6ddeb44e3502 contain the error.

hvenev, 2020-08-12:
It looks like upstream commit ea060b352654a8de1e070140d25fe1b7e4d50310 changed the driver a bit.

Dark-Sky, 2020-08-13:
I know 5.4-51 @ 1baaf03127861d00b69e3eba4a32a119ffe37e01 compiles but I have not compiled the latest 5.4-58 commit today.

pelwell, 2020-08-13:
The upstream commit broke a downstream patch, but the fix is easy (`dev->ep_intr` becomes `ep_intr`). There's another issue in the ath9k driver that needs fixing, but then I'll merge them both.

popcornmix, 2020-08-13:
I've pushed a fixed.

ncopa, 2020-08-13:
> There's another issue in the ath9k driver that needs fixing, but then I'll merge them both.

As mentioned in #3788 , this can be fixed by cherry-picking e6eb815beccca7018ce7b7c4b52059712dd1232d

pelwell, 2020-08-13:
Which I've done - just building now.

ncopa, 2020-08-13:
ah, i didnt notice that @pelwell and @popcornmix are two different persons :smile:  sorry.

Thank you for your quick response! :heart:

pelwell, 2020-08-13:
With the ath9k cherry-pick included it all builds again. That was an upstream error.

pelwell, 2020-08-13:
@popcornmix is up earlier than I expected.

Exaga, 2020-08-13:
Thanks guys. I have been head-scratching over the past 12 hours with this  too.

Dark-Sky, 2020-08-13:
I can confirm 5.4-58 @ a6825094038507f5eafdca6328ac6ddeb44e3502 now compiles with the fixes.

Claudenw, 2020-08-14:
The fixed applied resolved this problem.

closed 2020-08-14 by Claudenw