raspberrypi/linux #3787: Kernel 5.7.15 fails to compile

Dark-Sky, 2020-08-12:
Kernel 5.7.15 fails to compile. I aborts trying to compile drivers/net/usb/lan78xx.c

```
  CC      drivers/net/usb/lan78xx.o
drivers/net/usb/lan78xx.c: In function ‘lan78xx_probe’:
drivers/net/usb/lan78xx.c:3769:15: error: ‘struct lan78xx_net’ has no member named ‘ep_intr’
 3769 |   period = dev->ep_intr->desc.bInterval;
      |               ^~
  GZIP    kernel/config_data.gz
make[3]: *** [scripts/Makefile.build:266: drivers/net/usb/lan78xx.o] Error 1
make[2]: *** [scripts/Makefile.build:488: drivers/net/usb] Error 2
make[1]: *** [scripts/Makefile.build:488: drivers/net] Error 2
make: *** [Makefile:1735: drivers] Error 2
make: *** Waiting for unfinished jobs....
```

Claudenw, 2020-08-12:
this is the same error as in as #3786

popcornmix, 2020-08-13:
Pushed a fix.

Dark-Sky, 2020-08-13:
> Pushed a fix.

Thanks popcornmix. Compiled ok!

closed 2020-08-13 by Dark-Sky