raspberrypi/linux #3788: build of rpi-5.4.y (5.4.58) fails on 32 bit

ncopa, 2020-08-13:
**Describe the bug**
Build rpi-5.4.y (commit  a6825094038507f5eafdca6328ac6ddeb44e3502) with 32bit config fails.

**To reproduce**
build the rpi-5.4.y kernel with either `bcmrpi_defconfig` `bcm2709_defconfig` or `bcm2711_defconfig` (not sure exactly which one triggered it)

**Expected behaviour**
build success

**Actual behaviour**
build fails:
```
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/wireless/ath/ath9k/hif_usb.c: In function 'ath9k_hif_usb_reg
_in_cb':                                                    
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/wireless/ath/ath9k/hif_usb.c:735:3: error: 'rx_buf' undeclar
ed (first use in this function); did you mean 'tx_buf'?
  735 |   rx_buf->skb = nskb;                               
      |   ^~~~~~                                            
      |   tx_buf                                            
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/wireless/ath/ath9k/hif_usb.c:735:3: note: each undeclared id
entifier is reported only once for each function it appears in
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

This build happens on an arm server from packet net. (but its irrelevant)
```
$ uname -a
Linux ncopa-edge-armv7 5.4.34-0-lts #1-Alpine SMP Wed, 22 Apr 2020 19:26:07 UTC armv8l Linux
```

**Logs**
```
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/wireless/ath/ath9k/hif_usb.c: In function 'ath9k_hif_usb_reg
_in_cb':                                                    
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/wireless/ath/ath9k/hif_usb.c:735:3: error: 'rx_buf' undeclar
ed (first use in this function); did you mean 'tx_buf'?
  735 |   rx_buf->skb = nskb;                               
      |   ^~~~~~                                            
      |   tx_buf                                            
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/wireless/ath/ath9k/hif_usb.c:735:3: note: each undeclared id
entifier is reported only once for each function it appears in
  LD [M]  drivers/media/usb/gspca/gspca_finepix.o
```
**Additional context**
I believe commit e6eb815beccca7018ce7b7c4b52059712dd1232d needs to be cherry-picked to `rpi-5.4.y`.

What happened upstream in linux-5.4.y was:

- commit b5c8896bc14f54e5c4dd5a6e42879f125b8abd2d Fix general protection fault in ath9k_hif_usb_rx_cb
- this apparently caused problems so it got reverted in 5a046d75acf74b7fd889dca11120e42733f6f46d
- a fix those problems was made so the problematic commit was re-applied in  e6eb815beccca7018ce7b7c4b52059712dd1232d.
- The fix for the problems was applied on top in commit c15d59b9451129395ebeb6984409c3017fd42342 ath9k: Fix regression with Atheros 9271

I think the rpi branch has both the revert commit (5a046d75acf74b7fd889dca11120e42733f6f46d) and the fix (c15d59b9451129395ebeb6984409c3017fd42342 ath9k: Fix regression with Atheros 9271) for the reverted commit but not the re-applied e6eb815beccca7018ce7b7c4b52059712dd1232d.

To get a clear view of the history:
```
git log origin/linux-5.4.y -- drivers/net/wireless/ath/ath9k/hif_usb.
```

ncopa, 2020-08-13:
After e6eb815 got reapplied the build failed with this:
```
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/usb/lan78xx.c: In function 'lan78xx_probe':
/home/ncopa/aports/main/linux-rpi/src/linux-5.4/drivers/net/usb/lan78xx.c:3777:15: error: 'struct lan78xx_net' has no member named 'ep_intr'
 3777 |   period = dev->ep_intr->desc.bInterval;
      |               ^~
make[4]: *** [/home/ncopa/aports/main/linux-rpi/src/linux-5.4/scripts/Makefile.build:266: drivers/net/usb/lan78xx.o] Error 1
```

ncopa, 2020-08-13:
looks like the `lan78xx.c` build error comes from the merge commit a6825094038507f5eafdca6328ac6ddeb44e3502

pelwell, 2020-08-13:
Duplicate of #3786.

closed 2020-08-13 by pelwell

ncopa, 2020-08-13:
Hum... github does not give a nice view of it, but `git show a6825094038507f5eafdca6328ac6ddeb44e3502` give a better view:
```diff
commit a6825094038507f5eafdca6328ac6ddeb44e3502 (tag: 5.4.58-rpi-alpine, rpi/rpi-5.4.y, rpi-5.4.y)
Merge: 4b945d5fb692 cad17feaf0d0
Author: popcornmix <popcornmix@gmail.com>
Date:   Wed Aug 12 19:30:50 2020 +0100

    Merge remote-tracking branch 'stable/linux-5.4.y' into rpi-5.4.y

diff --cc drivers/net/usb/lan78xx.c
index 2b1abc0c8cdd,71cc5b63d8ce..88a95ce02ffb
--- a/drivers/net/usb/lan78xx.c
+++ b/drivers/net/usb/lan78xx.c
@@@ -3814,23 -3726,7 +3773,13 @@@ static int lan78xx_probe(struct usb_int
        netdev->max_mtu = MAX_SINGLE_PACKET_SIZE;
        netif_set_gso_max_size(netdev, MAX_SINGLE_PACKET_SIZE - MAX_HEADER);
  
-       dev->ep_blkin = (intf->cur_altsetting)->endpoint + 0;
-       dev->ep_blkout = (intf->cur_altsetting)->endpoint + 1;
-       dev->ep_intr = (intf->cur_altsetting)->endpoint + 2;
- 
-       dev->pipe_in = usb_rcvbulkpipe(udev, BULK_IN_PIPE);
-       dev->pipe_out = usb_sndbulkpipe(udev, BULK_OUT_PIPE);
- 
-       dev->pipe_intr = usb_rcvintpipe(dev->udev,
-                                       dev->ep_intr->desc.bEndpointAddress &
-                                       USB_ENDPOINT_NUMBER_MASK);
 -      period = ep_intr->desc.bInterval;
 +      if (int_urb_interval_ms <= 0)
 +              period = dev->ep_intr->desc.bInterval;
 +      else
 +              period = int_urb_interval_ms * INT_URB_MICROFRAMES_PER_MS;
 +
 +      netif_notice(dev, probe, netdev, "int urb period %d\n", period);
 +
        maxp = usb_maxpacket(dev->udev, dev->pipe_intr, 0);
        buf = kmalloc(maxp, GFP_KERNEL);
        if (buf) {
diff --cc drivers/usb/host/xhci-pci.c
index e5712e1f9754,bbd616324faa..5238fa29ea97
--- a/drivers/usb/host/xhci-pci.c
+++ b/drivers/usb/host/xhci-pci.c
@@@ -244,18 -247,16 +247,22 @@@ static void xhci_pci_quirks(struct devi
                        pdev->device == 0x3432)
                xhci->quirks |= XHCI_BROKEN_STREAMS;
  
 +      if (pdev->vendor == PCI_VENDOR_ID_VIA &&
 +                      pdev->device == 0x3483) {
 +              xhci->quirks |= XHCI_LPM_SUPPORT;
 +              xhci->quirks |= XHCI_EP_CTX_BROKEN_DCS;
 +      }
 +
        if (pdev->vendor == PCI_VENDOR_ID_ASMEDIA &&
-                       pdev->device == 0x1042)
+               pdev->device == PCI_DEVICE_ID_ASMEDIA_1042_XHCI)
                xhci->quirks |= XHCI_BROKEN_STREAMS;
        if (pdev->vendor == PCI_VENDOR_ID_ASMEDIA &&
-                       pdev->device == 0x1142)
+               pdev->device == PCI_DEVICE_ID_ASMEDIA_1042A_XHCI)
                xhci->quirks |= XHCI_TRUST_TX_LENGTH;
+       if (pdev->vendor == PCI_VENDOR_ID_ASMEDIA &&
+           (pdev->device == PCI_DEVICE_ID_ASMEDIA_1142_XHCI ||
+            pdev->device == PCI_DEVICE_ID_ASMEDIA_2142_XHCI))
+               xhci->quirks |= XHCI_NO_64BIT_SUPPORT;
  
        if (pdev->vendor == PCI_VENDOR_ID_ASMEDIA &&
                pdev->device == PCI_DEVICE_ID_ASMEDIA_1042A_XHCI)
```

ncopa, 2020-08-13:
> Duplicate of #3786.

The second problem (with lan78xx.c) is a duplicate, but the original  problem with `drivers/net/wireless/ath/ath9k/hif_usb.c` is not mentioned in #3786. They will bump into the ath9k error (this issue) once they solve the lan78xx.c problem.

pelwell, 2020-08-13:
They're both trivial problems - there's almost more work updating the issues, so I'd rather only do it once.