raspberrypi/linux #3789: display_rotate display_hdmi_rotate etc -- nothing works any more

pinheadmz, 2020-08-13:
Just recently updated OS on a Pi4B 4GB model.

```
$ uname -a
Linux zippi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux
```

I have this HDMI screen attached to HDMI port 0: https://www.ebay.com/itm/221952432326

I have always run this system headless, no GUI.

For some reason with a recent update I am unable to rotate the screen after trying all these options in `/boot/config.txt`:

```
display_rotate=2
display_hdmi_rotate=2
display_lcd_rotate=2
lcd_rotate=2
disable_fw_kms_setup=1
```

I'm seeing lots of forum posts about this:

https://raspberrypi.stackexchange.com/questions/82793/does-display-rotate-still-work
https://raspberrypi.stackexchange.com/questions/104073/display-rotate-not-working-in-raspberry-pi-64-bit
https://www.raspberrypi.org/forums/viewtopic.php?t=246146
...and several more.

I didn't see any open or closed issues after searching for `display_rotate` -- kinda stumped here. Any thoughts?

JamesH65, 2020-08-13:
https://www.raspberrypi.org/documentation/configuration/display_rotation.md

pinheadmz, 2020-08-13:
@JamesH65 Thank you sir!

Solution for me was adding this to the end of the line in `/boot/cmdline.txt`:

` video=HDMI-A-1:800x480@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0,rotate=180`

closed 2020-08-13 by pinheadmz

colesnicov, 2024-05-05:
> @JamesH65 Thank you sir!
> 
> Solution for me was adding this to the end of the line in `/boot/cmdline.txt`:
> 
> ` video=HDMI-A-1:800x480@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0,rotate=180`

Hi team, yes 180°. But how about 90˚ or 270˚??

colesnicov, 2024-05-05:
Hello people. I've been trying to figure out something similar for a few days now here https://github.com/raysan5/raylib/issues/3958 .

I try to rotate the display _by 90˚ or 270˚ degrees_, but it **only works 180˚**. I'm starting to give up. Would anyone know? It is this display

popcornmix, 2024-05-05:
Screen rotation by 90/270 can be handled from desktop where 3d hardware is used for composition.
There is no general solution for this when running outside of desktop.

colesnicov, 2024-05-05:
Good. I know it. But how does the desktop do it? I know it's possible, I had a desktop installed on it, but I don't need a desktop. Because of the kiosk.. It might be possible, via openbox, but I don't have time for that right now. I am programming a 3D kiosk with a 2D GUI. The old goal. Classic...

colesnicov, 2024-05-05:
I think, from what I have found, the desktop makes a composition before rendering, where it immediately rotates the image. And the desktop does not necessarily need to use 3D for this.