raspberrypi/linux #3791: videobuf2_common: driver bug: stop_streaming operation is leaving buf 0b70dd47 in active state

adalava, 2020-08-13:
I hit this "driver bug" after stopping an IP camera stream that was left open overnight. The software uses OpenCV to read camera frames using a gstreamer pipeline like this: "rtspsrc protocols=tcp tcp-timeout=2000000 retry=2 do-retransmission=false location=rtsp://192.168.42.10:554/user=admin&password=&channel=1&stream=0.sdp ! rtpjitter
buffer ! rtph264depay ! queue ! h264parse ! v4l2h264dec ! queue ! appsink sync=1"

Model: Raspberry Pi 3B 

$ cat /etc/rpi-issue
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2

$ vcgencmd version
Jul 17 2020 11:04:27 
Copyright (c) 2012 Broadcom
version 21a15cb094f41c7506ad65d2cb9b29c550693057 (clean) (release) (start)

$ uname -a
Linux br0001 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux

[Thu Aug 13 10:12:33 2020] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 11 outstanding
[Thu Aug 13 10:12:33 2020] ------------[ cut here ]------------
[Thu Aug 13 10:12:33 2020] WARNING: CPU: 0 PID: 13924 at drivers/media/common/videobuf2/videobuf2-core.c:1882 __vb2_queue_cancel+0x210/0x288 [videobuf2_common]
[Thu Aug 13 10:12:33 2020] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc brcmfmac brcmutil sha256_generic libsha256 cfg80211 raspberrypi_hwmon rfkill i2c_bcm2835 bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem snd_bcm2835(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops snd_pcm videobuf2_v4l2 videobuf2_common snd_timer videodev snd mc vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
[Thu Aug 13 10:12:33 2020] CPU: 0 PID: 13924 Comm: python Tainted: G        WC        5.4.51-v7+ #1327
[Thu Aug 13 10:12:33 2020] Hardware name: BCM2835
[Thu Aug 13 10:12:33 2020] Backtrace:
[Thu Aug 13 10:12:33 2020] [<8010d458>] (dump_backtrace) from [<8010d750>] (show_stack+0x20/0x24)
[Thu Aug 13 10:12:33 2020]  r6:8bcf6000 r5:00000000 r4:80d93ff4 r3:82a41469
[Thu Aug 13 10:12:33 2020] [<8010d730>] (show_stack) from [<808b2324>] (dump_stack+0xe0/0x124)
[Thu Aug 13 10:12:33 2020] [<808b2244>] (dump_stack) from [<8011fd24>] (__warn+0xec/0x104)
[Thu Aug 13 10:12:33 2020]  r8:0000075a r7:00000009 r6:7f0ca7c4 r5:00000000 r4:00000000 r3:82a41469
[Thu Aug 13 10:12:33 2020] [<8011fc38>] (__warn) from [<8011fdf4>] (warn_slowpath_fmt+0xb8/0xc0)
[Thu Aug 13 10:12:33 2020]  r9:7f0ca7c4 r8:0000075a r7:7f0c519c r6:00000009 r5:00000000 r4:80d04f48
[Thu Aug 13 10:12:33 2020] [<8011fd40>] (warn_slowpath_fmt) from [<7f0c519c>] (__vb2_queue_cancel+0x210/0x288 [videobuf2_common])
[Thu Aug 13 10:12:33 2020]  r9:afa8b40c r8:40045613 r7:7f165ee0 r6:afa8b40c r5:00000009 r4:afa8b40c
[Thu Aug 13 10:12:33 2020] [<7f0c4f8c>] (__vb2_queue_cancel [videobuf2_common]) from [<7f0c6b00>] (vb2_core_streamoff+0x44/0xb4 [videobuf2_common])
[Thu Aug 13 10:12:33 2020]  r10:b1bd27c8 r9:00000020 r8:40045613 r7:7f165ee0 r6:afa8b40c r5:00000009
[Thu Aug 13 10:12:33 2020]  r4:afa8b40c r3:00000009
[Thu Aug 13 10:12:33 2020] [<7f0c6abc>] (vb2_core_streamoff [videobuf2_common]) from [<7f0e6084>] (vb2_streamoff+0x40/0x60 [videobuf2_v4l2])
[Thu Aug 13 10:12:33 2020]  r4:afa8b400 r3:00000000
[Thu Aug 13 10:12:33 2020] [<7f0e6044>] (vb2_streamoff [videobuf2_v4l2]) from [<7f1d8c08>] (v4l2_m2m_streamoff+0x3c/0xf4 [v4l2_mem2mem])
[Thu Aug 13 10:12:33 2020] [<7f1d8bcc>] (v4l2_m2m_streamoff [v4l2_mem2mem]) from [<7f1d8ce0>] (v4l2_m2m_ioctl_streamoff+0x20/0x24 [v4l2_mem2mem])
[Thu Aug 13 10:12:33 2020]  r10:b1bd27c8 r9:00000020 r8:40045613 r7:7f165ee0 r6:80d04f48 r5:00000000
[Thu Aug 13 10:12:33 2020]  r4:b1bd24a0 r3:97229000
[Thu Aug 13 10:12:33 2020] [<7f1d8cc0>] (v4l2_m2m_ioctl_streamoff [v4l2_mem2mem]) from [<7f165f08>] (v4l_streamoff+0x28/0x2c [videodev])
[Thu Aug 13 10:12:33 2020] [<7f165ee0>] (v4l_streamoff [videodev]) from [<7f16acb4>] (__video_do_ioctl+0x240/0x470 [videodev])
[Thu Aug 13 10:12:33 2020] [<7f16aa74>] (__video_do_ioctl [videodev]) from [<7f16cd54>] (video_usercopy+0x220/0x60c [videodev])
[Thu Aug 13 10:12:33 2020]  r10:7f16aa74 r9:40045613 r8:00000000 r7:8bcf7e04 r6:00000004 r5:00000000
[Thu Aug 13 10:12:33 2020]  r4:80d04f48
[Thu Aug 13 10:12:33 2020] [<7f16cb34>] (video_usercopy [videodev]) from [<7f16d160>] (video_ioctl2+0x20/0x24 [videodev])
[Thu Aug 13 10:12:33 2020]  r10:00000036 r9:8bcf6000 r8:00000026 r7:8e89a540 r6:b1371690 r5:021f1200
[Thu Aug 13 10:12:33 2020]  r4:80d04f48
[Thu Aug 13 10:12:33 2020] [<7f16d140>] (video_ioctl2 [videodev]) from [<7f164144>] (v4l2_ioctl+0x4c/0x60 [videodev])
[Thu Aug 13 10:12:33 2020] [<7f1640f8>] (v4l2_ioctl [videodev]) from [<802f8540>] (do_vfs_ioctl+0xbc/0x804)
[Thu Aug 13 10:12:33 2020] [<802f8484>] (do_vfs_ioctl) from [<802f8ccc>] (ksys_ioctl+0x44/0x6c)
[Thu Aug 13 10:12:33 2020]  r10:00000036 r9:8bcf6000 r8:00000026 r7:40045613 r6:8e89a540 r5:021f1200
[Thu Aug 13 10:12:33 2020]  r4:8e89a541
[Thu Aug 13 10:12:33 2020] [<802f8c88>] (ksys_ioctl) from [<802f8d0c>] (sys_ioctl+0x18/0x1c)
[Thu Aug 13 10:12:33 2020]  r8:801011c4 r7:00000036 r6:4ca637d0 r5:4ca63000 r4:6a308648 r3:76d72510
[Thu Aug 13 10:12:33 2020] [<802f8cf4>] (sys_ioctl) from [<80101000>] (ret_fast_syscall+0x0/0x28)
[Thu Aug 13 10:12:33 2020] Exception stack(0x8bcf7fa8 to 0x8bcf7ff0)
[Thu Aug 13 10:12:33 2020] 7fa0:                   6a308648 4ca63000 00000026 40045613 021f1200 76d72510
[Thu Aug 13 10:12:33 2020] 7fc0: 6a308648 4ca63000 4ca637d0 00000036 6a307300 6a4785f0 706e492c 74c73ce8
[Thu Aug 13 10:12:33 2020] 7fe0: 708cfc60 7eb630b4 4ca2b954 76d7251c
[Thu Aug 13 10:12:33 2020] ---[ end trace abd9ddf252533015 ]---
[Thu Aug 13 10:12:33 2020] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0b70dd47 in active state
[Thu Aug 13 10:12:33 2020] videobuf2_common: driver bug: stop_streaming operation is leaving buf b0d5ce36 in active state
[Thu Aug 13 10:12:33 2020] videobuf2_common: driver bug: stop_streaming operation is leaving buf da61dea1 in active state
[Thu Aug 13 10:12:33 2020] videobuf2_common: driver bug: stop_streaming operation is leaving buf d4e41263 in active state
[Thu Aug 13 10:12:33 2020] videobuf2_common: driver bug: stop_streaming operation is leaving buf db4367ac in active state
[Thu Aug 13 10:12:33 2020] videobuf2_common: driver bug: stop_streaming operation is leaving buf 08347a89 in active state

6by9, 2020-08-13:
Hopefully solved by #3790

popcornmix, 2020-08-14:
Fix should be in latest rpi-update kernel.

adalava, 2020-08-15:
Thank you! I updated to 5.4.58-v7+ #1335 SMP Thu Aug 13 22:17:06 BST 2020 armv7l GNU/Linux and will let you known if it appears again.

adalava, 2020-08-24:
I've being using  5.4.58-v7+ for more than a week, it looks good, so I'm closing this issue. Thank you!

timemaster5, 2020-08-26:
I think the problem is still there. My use case is slightly different. I am playing a youtube video in the wpewebkit browser, and still reaching these messages in dmesg :( 

Sorry..

adalava, 2020-08-26:
:-(   @timemaster5, it's still good here so far. Anyway I reopened the issue and would be nice if you could give more information for the devs, like your dmesg and "uname -a " output. Maybe there's something in your output that could help.

timemaster5, 2020-08-27:
hm, maybe it has multiple causes.. here is the dmesg. The second line is uname -a basically. Hope that helps.

`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.59-v7 (oe-user@oe-host) (gcc version 9.3.0 (GCC)) #1 SMP Tue Aug 25 23:21:36 UTC 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x1ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 196608
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49868 r8192 d23860 u81920
[    0.000000] pcpu-alloc: s49868 r8192 d23860 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 194880
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=HDMI-A-1:1680x1050M@60 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait loglevel=1 quiet consoleblank=0
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 500792K/786432K available (9216K kernel code, 691K rwdata, 2752K rodata, 1024K init, 828K bss, 23496K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 29395 entries in 58 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x324/0x504 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000006] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000017] Switching to timer-based delay loop, resolution 52ns
[    0.000292] Console: colour dummy device 80x30
[    0.000323] printk: console [tty0] enabled
[    0.000378] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000399] pid_max: default: 32768 minimum: 301
[    0.000799] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000822] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002270] Disabling memory control group subsystem
[    0.002376] CPU: Testing write buffer coherency: ok
[    0.002990] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003952] Setting up static identity map for 0x100000 - 0x10003c
[    0.004154] rcu: Hierarchical SRCU implementation.
[    0.004873] smp: Bringing up secondary CPUs ...
[    0.006066] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.007385] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.008599] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.008755] smp: Brought up 1 node, 4 CPUs
[    0.008771] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.008781] CPU: All CPU(s) started in HYP mode.
[    0.008790] CPU: Virtualization extensions available.
[    0.009787] devtmpfs: initialized
[    0.026317] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.026599] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.026629] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.038245] pinctrl core: initialized pinctrl subsystem
[    0.039506] NET: Registered protocol family 16
[    0.043967] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.050185] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.050198] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.050438] Serial: AMBA PL011 UART driver
[    0.053018] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.080213] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-07-13 13:57, variant start
[    0.090226] raspberrypi-firmware soc:firmware: Firmware hash is adcebbdb7b415c623931e80795ba3bae68dcc4fa
[    0.141592] bcm2835-dma 3f007000.dma: DMA legacy API manager, dmachans=0x1
[    0.143899] SCSI subsystem initialized
[    0.144174] usbcore: registered new interface driver usbfs
[    0.144238] usbcore: registered new interface driver hub
[... 403 lines cut ...]
[  228.399107] [<7f117dd0>] (v4l_streamoff [videodev]) from [<7f11cbec>] (__video_do_ioctl+0x240/0x470 [videodev])
[  228.399114]  r5:40045613 r4:9cc3d8a0
[... cut ...]
[  228.399971] ffc0: 019d4300 6a4ff000 6a4ff7e4 00000036 019d9450 019e2388 00001407 4ae74678
[  228.399979] ffe0: 4ae733bc 7eb3ab24 6a4d06a5 4a676098
[  228.399991] ---[ end trace 566a229dbfdfd722 ]---
[  228.400007] videobuf2_common: driver bug: stop_streaming operation is leaving buf d7c9d963 in active state
[  228.400026] videobuf2_common: driver bug: stop_streaming operation is leaving buf 439ca233 in active state
[  267.118243] bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Timeout waiting for buffers to be returned - 6 outstanding`

timemaster5, 2020-09-20:
there has been some progress in suspiciously similar issue https://github.com/raspberrypi/linux/issues/3325#issuecomment-684040830, is it helpful?

hailfinger, 2021-04-02:
@timemaster5 This might be fixed in firmware release edf2e9c318863999c97c50cdb74eee235ede3af5 which will hopefully reach your Pi with a regular update soon.

timemaster5, 2021-04-02:
thank you for letting me know.. I use a custom distribution build with Yocto, so will try to pull this version manually

timemaster5, 2021-04-02:
however, struggling with finding this.. is it a commit id in the raspberry/firmware repo?

pelwell, 2021-04-02:
That's probably the build version (commit hash for the source code) - the correct hash for the firmware repo is https://github.com/raspberrypi/firmware/commit/20081d8e86119e95e516526700be62049850c01a, or https://github.com/Hexxeh/rpi-firmware/commit/c0cf93a133dab106439c208f03d32155bc19e432 for rpi-update.

alecov, 2021-05-25:
Facing the same issue here, the system crashes when streaming video via any browser. Sometimes there is no log available.

This is from my crash:
```
May 25 20:29:37 raspberrypi kernel: [ 1178.638469] ------------[ cut here ]------------
May 25 20:29:37 raspberrypi kernel: [ 1178.638491] WARNING: CPU: 0 PID: 5964 at drivers/media/common/videobuf2/videobuf2-core.c:1945 __vb2_queue_cancel+0x1e0/0x258 [videobuf2_common]
May 25 20:29:37 raspberrypi kernel: [ 1178.638496] Modules linked in: binfmt_misc rfcomm cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc joydev v3d brcmfmac brcmutil gpu_sched sha256_generic cfg80211 rfkill sg vc4 cec raspberrypi_hwmon bcm2835_codec(C) v4l2_mem2mem drm_kms_helper bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc vc_sm_cma(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd_bcm2835(C) mc drm rpivid_mem drm_panel_orientation_quirks snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops backlight uio_pdrv_genirq nvmem_rmem uio ip_tables x_tables ipv6
May 25 20:29:37 raspberrypi kernel: [ 1178.638785] CPU: 0 PID: 5964 Comm: WebKitWebProces Tainted: G         C        5.10.17-v7l+ #1414
May 25 20:29:37 raspberrypi kernel: [ 1178.638788] Hardware name: BCM2711
May 25 20:29:37 raspberrypi kernel: [ 1178.638792] Backtrace: 
May 25 20:29:37 raspberrypi kernel: [ 1178.638810] [<c0b63050>] (dump_backtrace) from [<c0b633e4>] (show_stack+0x20/0x24)
May 25 20:29:37 raspberrypi kernel: [ 1178.638815]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e69fc
May 25 20:29:37 raspberrypi kernel: [ 1178.638821] [<c0b633c4>] (show_stack) from [<c0b67774>] (dump_stack+0xcc/0xf8)
May 25 20:29:37 raspberrypi kernel: [ 1178.638829] [<c0b676a8>] (dump_stack) from [<c0220bac>] (__warn+0xfc/0x114)
May 25 20:29:37 raspberrypi kernel: [ 1178.638834]  r10:bf2328dc r9:00000009 r8:bf0dc6b4 r7:00000799 r6:00000009 r5:bf0dc6b4
May 25 20:29:37 raspberrypi kernel: [ 1178.638838]  r4:bf0e1f90 r3:c1205094
May 25 20:29:37 raspberrypi kernel: [ 1178.638844] [<c0220ab0>] (__warn) from [<c0b63b54>] (warn_slowpath_fmt+0x70/0xd8)
May 25 20:29:37 raspberrypi kernel: [ 1178.638848]  r7:00000799 r6:bf0e1f90 r5:c1205048 r4:00000000
May 25 20:29:37 raspberrypi kernel: [ 1178.638860] [<c0b63ae8>] (warn_slowpath_fmt) from [<bf0dc6b4>] (__vb2_queue_cancel+0x1e0/0x258 [videobuf2_common])
May 25 20:29:37 raspberrypi kernel: [ 1178.638865]  r9:bf240098 r8:c272abe0 r7:00000001 r6:c480e014 r5:c480e014 r4:c480e014
May 25 20:29:37 raspberrypi kernel: [ 1178.638881] [<bf0dc4d4>] (__vb2_queue_cancel [videobuf2_common]) from [<bf0dd6b4>] (vb2_core_streamoff+0x28/0xbc [videobuf2_common])
May 25 20:29:37 raspberrypi kernel: [ 1178.638886]  r10:bf2328dc r9:bf240098 r8:c272abe0 r7:00000001 r6:00000009 r5:c480e014
May 25 20:29:37 raspberrypi kernel: [ 1178.638889]  r4:c480e014 r3:00000009
May 25 20:29:37 raspberrypi kernel: [ 1178.638906] [<bf0dd68c>] (vb2_core_streamoff [videobuf2_common]) from [<bf17b0f8>] (vb2_streamoff+0x24/0x64 [videobuf2_v4l2])
May 25 20:29:37 raspberrypi kernel: [ 1178.638910]  r5:c480e014 r4:c480e000
May 25 20:29:37 raspberrypi kernel: [ 1178.638926] [<bf17b0d4>] (vb2_streamoff [videobuf2_v4l2]) from [<bf224f08>] (v4l2_m2m_streamoff+0x40/0x10c [v4l2_mem2mem])
May 25 20:29:37 raspberrypi kernel: [ 1178.638940] [<bf224ec8>] (v4l2_m2m_streamoff [v4l2_mem2mem]) from [<bf224ff4>] (v4l2_m2m_ioctl_streamoff+0x20/0x24 [v4l2_mem2mem])
May 25 20:29:37 raspberrypi kernel: [ 1178.638945]  r10:bf2328dc r9:bf240098 r8:c272abe0 r7:00000001 r6:c1205048 r5:40045613
May 25 20:29:37 raspberrypi kernel: [ 1178.638948]  r4:bf224fd4
May 25 20:29:37 raspberrypi kernel: [ 1178.638989] [<bf224fd4>] (v4l2_m2m_ioctl_streamoff [v4l2_mem2mem]) from [<bf2400c4>] (v4l_streamoff+0x2c/0x30 [videodev])
May 25 20:29:37 raspberrypi kernel: [ 1178.639040] [<bf240098>] (v4l_streamoff [videodev]) from [<bf245114>] (__video_do_ioctl+0x234/0x460 [videodev])
May 25 20:29:37 raspberrypi kernel: [ 1178.639044]  r5:40045613 r4:c272a8a0
May 25 20:29:37 raspberrypi kernel: [ 1178.639092] [<bf244ee0>] (__video_do_ioctl [videodev]) from [<bf245b28>] (video_usercopy+0x128/0x5ac [videodev])
May 25 20:29:37 raspberrypi kernel: [ 1178.639097]  r10:c6f05e3c r9:79124e18 r8:00000000 r7:c663ec00 r6:c1205048 r5:40045613
May 25 20:29:37 raspberrypi kernel: [ 1178.639100]  r4:40045613
May 25 20:29:37 raspberrypi kernel: [ 1178.639147] [<bf245a00>] (video_usercopy [videodev]) from [<bf245fcc>] (video_ioctl2+0x20/0x24 [videodev])
May 25 20:29:37 raspberrypi kernel: [ 1178.639152]  r10:00000054 r9:c6f04000 r8:00000000 r7:c663ec00 r6:c663ec01 r5:c1205048
May 25 20:29:37 raspberrypi kernel: [ 1178.639156]  r4:bf245fac
May 25 20:29:37 raspberrypi kernel: [ 1178.639205] [<bf245fac>] (video_ioctl2 [videodev]) from [<bf23e1f4>] (v4l2_ioctl+0x4c/0x60 [videodev])
May 25 20:29:37 raspberrypi kernel: [ 1178.639237] [<bf23e1a8>] (v4l2_ioctl [videodev]) from [<c045178c>] (sys_ioctl+0x1d4/0x8ec)
May 25 20:29:37 raspberrypi kernel: [ 1178.639242]  r5:c1205048 r4:40045613
May 25 20:29:37 raspberrypi kernel: [ 1178.639248] [<c04515b8>] (sys_ioctl) from [<c0200040>] (ret_fast_syscall+0x0/0x28)
May 25 20:29:37 raspberrypi kernel: [ 1178.639252] Exception stack(0xc6f05fa8 to 0xc6f05ff0)
May 25 20:29:37 raspberrypi kernel: [ 1178.639257] 5fa0:                   a75d6b90 78dde000 00000054 40045613 79124e18 b0368510
[... cut ...]
May 25 20:29:37 raspberrypi kernel: [ 1178.639293] videobuf2_common: driver bug: stop_streaming operation is leaving buf ca7b0c4f in active state
May 25 20:29:37 raspberrypi kernel: [ 1178.639300] videobuf2_common: driver bug: stop_streaming operation is leaving buf a1e29602 in active state
```

uname:
```
Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
```

firmware:
```
Apr 30 2021 13:45:52 
Copyright (c) 2012 Broadcom
version d7f29d96450abfc77cd6cf011af1faf1e03e5e56 (clean) (release) (start)
```

FWIW, I've tested the aarch64 (beta) release from RaspiOS and the issue is still there.

adalava, 2021-05-28:
I'm still getting this error while decoding h264 from IP cameras using gstreamer, no luck so far.  Kernel "4.19.118-v7+ #1311 SMP Mon Apr 27 14:21:24 BST 2020" seems stable, I'm not hitting this problem after several weeks.

timemaster5, 2021-12-04:
Do we have any progress or clues here? Or even workaround from the others.
I want to help somehow, but I am out of ideas. I feel it like the vc4 opensource driver is just not ready yet.

Last week, I did some tests with the userland driver and was able to play videos in a loop without significant issues. Nothing in the kernel log at least :) Only GStreamer had something to tell, but I guess it wasn't substantial enough to break something.

I tried even the newest bullseye firmware, but the situation is the same. The only noticeable difference between the old kernel 4.19 and 5.4 and 5.10 was that in 4.19, after the video crashed, the new one didn't start. On the newer kernel, the new loop starts playing, but kernel log keeps growing of those messages :(

hailfinger, 2021-12-04:
Try adding `over_voltage=2` to config.txt. If that doesn't help, try increasing until `over_voltage=6`. For my fleet of ~150 devices, a majority will behave with over_voltage=2, those which are still not fixed get over_voltage=4, and the still unfixed remainder gets over_voltage=6 (roughly 5% of my devices need that). Three devices still don't work reliably even with over_voltage=6, but I'll remove those from production soon.

Despite claims to the contrary, these buffers will accumulate (every time the error message gets repeated, the reported number of buffers increases) and eventually playing videos will fail. Since my devices are in production, those which cannot be fixed even with over_voltage=6 get regular scheduled reboots. That's a horrible workaround, but at least it's a workaround.

timemaster5, 2021-12-04:
Hi @hailfinger, thank you for your response. I found your thread somewhere else some time ago and immediately thought, hey, that's my missing bit, but it wasn't :( I tried over_voltage up to 6, then even more, but the problem is still there. And it is immediate. I mean, this error log appears the first time the video plays and then with every loop. I don't need to wait hours for it :( That is why I am quite surprised no one can sort it. 

If I'm correct, you were on raspberry pi 4, is that right?. Here I am on the Compute Module 3+.

hailfinger, 2021-12-04:
Ah yes, that might explain the difference. I'm on a Raspberry Pi 4B.

timemaster5, 2021-12-06:
I believe it is an issue with the hardware itself. All is fine with userland on CM3+ or Rpi3b+, so the decoder is not "broken", but as userland is the closed source, I guess the vc4 driver does something potentially right from the general point of view, but BCM requires something specific, maybe more non-standard to work correctly in this case.

The reason why this is haven't been fixed, I guess, is because of this difference in board versions. Many people successfully resolved this issue with the newer firmware release, but they were on Rpi 4. On Rpi 3, this has never been fixed and maybe not even looked into.

6by9, 2021-12-06:
> I believe it is an issue with the hardware itself.

It's not.
You say it yourself: 

> All is fine with userland on CM3+ or Rpi3b+

The V4L2 driver uses exactly the same parts of the firmware, but there is a path that I believe I know what the issue is, but haven't had time to investigate (partly as I don't have a reliable route to reproduce).

I've said it multiple times, ```videobuf2_common: driver bug: stop_streaming operation is leaving buf XXXXXX in active state``` is not a fatal issue.

timemaster5, 2021-12-06:
> > I believe it is an issue with the hardware itself.
> 
> It's not. You say it yourself:
> 
> > All is fine with userland on CM3+ or Rpi3b+

I am sorry for the misunderstanding. The context is further in the following sentence. I believe it is hardware-specific, not a hardware bug, just a specific way we need to treat BCM here. I don't know much about documentation available for this specific bit, but in general, I know it is let's say more problematic with Broadcom.

> I've said it multiple times, `videobuf2_common: driver bug: stop_streaming operation is leaving buf XXXXXX in active state` is not a fatal issue.

I have to disagree here. This cause system lockup and the only variable is time. If you have a short 10s video you want to play in a loop, the number of buffers left in the active state increase every 10s. It is worse on the 4.19 kernel, where the new video loop doesn't even start, and you can't reboot the system to fix it because it will not boot again. The only way to fix it is to power cycle.

It is much better on 5.4. and up, where the video can play, but at some point, the number of buffers left is just too high. I don't know whether the memory runs out or there is no new buffer available for the next video loop.

> The V4L2 driver uses exactly the same parts of the firmware, but there is a path that I believe I know what the issue is, but haven't had time to investigate (partly as I don't have a reliable route to reproduce).

I am happy to help. If you want, I can build you an image with all the tools you need and some example video so you can just flash and immediately see what is wrong.

tysonite, 2022-06-14:
Same on `Linux 5.10.92-v7l+ #1514 SMP Mon Jan 17 17:38:03 GMT 2022 armv7l GNU/Linux` with GStreamer and web camera.

```
[363583.786353] restoring control 00000000-0000-0000-0000-000000000101/3/7
[363583.786358] restoring control 00000000-0000-0000-0000-000000000101/6/10
[363583.786364] restoring control 00000000-0000-0000-0000-000000000101/9/4
[363583.786369] restoring control 00000000-0000-0000-0000-000000000101/12/11
[364321.215193] restoring control 00000000-0000-0000-0000-000000000001/1/2
[364321.215203] restoring control 00000000-0000-0000-0000-000000000001/3/4
[364321.215209] restoring control 00000000-0000-0000-0000-000000000001/5/6
[364321.215215] restoring control 00000000-0000-0000-0000-000000000001/17/8
[364321.215615] restoring control 00000000-0000-0000-0000-000000000101/0/2
[364321.215622] restoring control 00000000-0000-0000-0000-000000000101/1/3
[364321.215627] restoring control 00000000-0000-0000-0000-000000000101/3/7
[364321.215634] restoring control 00000000-0000-0000-0000-000000000101/6/10
[364321.215639] restoring control 00000000-0000-0000-0000-000000000101/9/4
[364321.215645] restoring control 00000000-0000-0000-0000-000000000101/12/11
[364325.758121] restoring control 00000000-0000-0000-0000-000000000001/1/2
[364325.758129] restoring control 00000000-0000-0000-0000-000000000001/3/4
[364325.758135] restoring control 00000000-0000-0000-0000-000000000001/5/6
[364325.758141] restoring control 00000000-0000-0000-0000-000000000001/17/8
[364325.758517] restoring control 00000000-0000-0000-0000-000000000101/0/2
[364325.758523] restoring control 00000000-0000-0000-0000-000000000101/1/3
[364325.758528] restoring control 00000000-0000-0000-0000-000000000101/3/7
[364325.758534] restoring control 00000000-0000-0000-0000-000000000101/6/10
[364325.758539] restoring control 00000000-0000-0000-0000-000000000101/9/4
[364325.758545] restoring control 00000000-0000-0000-0000-000000000101/12/11
[364591.527886] bcm2835-codec bcm2835-codec: bcm2835_codec_flush_buffers: Timeout waiting for buffers to be returned - 23 outstanding
[364591.528097] ------------[ cut here ]------------
[364591.528120] WARNING: CPU: 1 PID: 27736 at drivers/media/common/videobuf2/videobuf2-core.c:1956 __vb2_queue_cancel+0x1e0/0x258 [videobuf2_common]
[364591.528126] Modules linked in: rfcomm cmac algif_hash aes_arm_bs crypto_simd cryptd algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc spidev tc358743 snd_soc_hdmi_codec brcmfmac brcmutil snd_usb_audio snd_hwdep snd_usbmidi_lib cfg80211 rfkill uvcvideo snd_rawmidi snd_seq_device bcm2835_codec(C) v4l2_mem2mem v3d spi_bcm2835 gpu_sched snd_soc_simple_card snd_soc_simple_card_utils snd_soc_spdif_rx raspberrypi_hwmon bcm2835_isp(C) bcm2835_unicam bcm2835_v4l2(C) videobuf2_dma_contig v4l2_dv_timings bcm2835_mmal_vchiq(C) v4l2_fwnode videobuf2_vmalloc i2c_mux_pinctrl i2c_brcmstb videobuf2_memops i2c_mux videobuf2_v4l2 i2c_bcm2835 videobuf2_common videodev mc snd_soc_bcm2835_i2s vc4 snd_bcm2835(C) vc_sm_cma(C) cec drm_kms_helper rpivid_mem snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer uio_pdrv_genirq uio ir_rc6_decoder snd syscopyarea sysfillrect sysimgblt fb_sys_fops gpio_ir_tx nvmem_rmem gpio_ir_recv i2c_dev drm drm_panel_orientation_quirks
[364591.528520]  backlight fuse ip_tables x_tables ipv6
[364591.528552] CPU: 1 PID: 27736 Comm: gst-launch-1.0 Tainted: G         C        5.10.92-v7l+ #1514
[364591.528555] Hardware name: BCM2711
[364591.528559] Backtrace:
[364591.528574] [<c0b87728>] (dump_backtrace) from [<c0b87abc>] (show_stack+0x20/0x24)
[364591.528580]  r7:ffffffff r6:00000000 r5:60000013 r4:c12e6fd8
[364591.528587] [<c0b87a9c>] (show_stack) from [<c0b8be9c>] (dump_stack+0xcc/0xf8)
[364591.528595] [<c0b8bdd0>] (dump_stack) from [<c0220d2c>] (__warn+0xfc/0x114)
[364591.528600]  r10:ffffe000 r9:00000009 r8:bf40f6b4 r7:000007a4 r6:00000009 r5:bf40f6b4
[364591.528604]  r4:bf414f90 r3:c1205094
[364591.528610] [<c0220c30>] (__warn) from [<c0b8821c>] (warn_slowpath_fmt+0x70/0xd8)
[364591.528614]  r7:000007a4 r6:bf414f90 r5:c1205048 r4:00000000
[364591.528627] [<c0b881b0>] (warn_slowpath_fmt) from [<bf40f6b4>] (__vb2_queue_cancel+0x1e0/0x258 [videobuf2_common])
[364591.528631]  r9:00000000 r8:c25bb5d8 r7:c28fa790 r6:c4e27414 r5:c4e27474 r4:c4e27414
[364591.528649] [<bf40f4d4>] (__vb2_queue_cancel [videobuf2_common]) from [<bf411cf8>] (vb2_core_queue_release+0x28/0x48 [videobuf2_common])
[364591.528654]  r10:ffffe000 r9:00000000 r8:c25bb5d8 r7:c28fa790 r6:c2f663e0 r5:c4e27474
[364591.528657]  r4:c4e27414 r3:1c02cd6b
[364591.528674] [<bf411cd0>] (vb2_core_queue_release [videobuf2_common]) from [<bf1a4850>] (vb2_queue_release+0x18/0x1c [videobuf2_v4l2])
[364591.528678]  r5:c4c59000 r4:c4e27400
[364591.528693] [<bf1a4838>] (vb2_queue_release [videobuf2_v4l2]) from [<bf39b37c>] (v4l2_m2m_ctx_release+0x24/0x38 [v4l2_mem2mem])
[364591.528707] [<bf39b358>] (v4l2_m2m_ctx_release [v4l2_mem2mem]) from [<bf3ac7f4>] (bcm2835_codec_release+0x7c/0xc8 [bcm2835_codec])
[364591.528711]  r5:c4c59000 r4:c2f66040
[364591.528749] [<bf3ac778>] (bcm2835_codec_release [bcm2835_codec]) from [<bf30c6e0>] (v4l2_release+0xd8/0xe4 [videodev])
[364591.528753]  r7:c28fa790 r6:000e001f r5:c4f2a0c0 r4:c2f660a0
[364591.528786] [<bf30c608>] (v4l2_release [videodev]) from [<c0440af0>] (__fput+0x90/0x25c)
[364591.528789]  r5:c3499b68 r4:c4f2a0c0
[364591.528795] [<c0440a60>] (__fput) from [<c0440cd4>] (____fput+0x18/0x1c)
[... cut ...]
[364591.529029] videobuf2_common: driver bug: stop_streaming operation is leaving buf 87bbcbf6 in active state
[364591.529037] videobuf2_common: driver bug: stop_streaming operation is leaving buf b3ddc833 in active state
[364591.529044] videobuf2_common: driver bug: stop_streaming operation is leaving buf 5705413d in active state
[364591.529051] videobuf2_common: driver bug: stop_streaming operation is leaving buf 051591ab in active state
[364591.529058] videobuf2_common: driver bug: stop_streaming operation is leaving buf ae93b7f0 in active state
```

nicoske, 2023-02-27:
Facing the same issue when looping a 30s video from WPEWebkit on latest kernel 6.1.13-v7l.

`Feb 27 11:47:01 pi-ibbdis-865a7b kernel: ------------[ cut here ]------------
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: WARNING: CPU: 3 PID: 343 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x23c/0x2b8 [videobuf2_common]
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: Modules linked in: vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper brcmfmac brcmutil snd_soc_core bcm2835_codec(C) bcm2835_isp(C) snd_compress bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem snd_pcm_dmaengine videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 cfg80211 raspberrypi_hwmon videobuf2_common syscopyarea snd_bcm2835(C) sysfillrect sysimgblt fb_sys_fops rfkill snd_pcm videodev i2c_bcm2835 snd_timer snd vc_sm_cma(C) uio_pdrv_genirq uio ipt_REJECT nf_reject_ipv4 xt_limit xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype xt_tcpudp iptable_filter sch_fq_codel drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: CPU: 3 PID: 343 Comm: WPEWebProcess Tainted: G        WC         6.1.13-v7l #1
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: Hardware name: BCM2835
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  unwind_backtrace from show_stack+0x18/0x1c
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  show_stack from dump_stack_lvl+0x68/0x8c
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  dump_stack_lvl from __warn+0x84/0x11c
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  __warn from warn_slowpath_fmt+0x80/0xbc
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  warn_slowpath_fmt from __vb2_queue_cancel+0x23c/0x2b8 [videobuf2_common]
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  __vb2_queue_cancel [videobuf2_common] from vb2_core_streamoff+0x20/0xb4 [videobuf2_common]
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  vb2_core_streamoff [videobuf2_common] from v4l2_m2m_streamoff+0x44/0x108 [v4l2_mem2mem]
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  v4l2_m2m_streamoff [v4l2_mem2mem] from __video_do_ioctl+0x240/0x490 [videodev]
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  __video_do_ioctl [videodev] from video_usercopy+0x2a0/0x610 [videodev]
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  video_usercopy [videodev] from sys_ioctl+0x110/0xa14
Feb 27 11:47:01 pi-ibbdis-865a7b kernel:  sys_ioctl from __sys_trace_return+0x0/0x10
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: Exception stack(0xf0a9dfa8 to 0xf0a9dff0)
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: dfa0:                   a3c798ec a3c7a14c 00000027 40045613 a6a80788 beb5e870
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: dfc0: a3c798ec a3c7a14c a24ea420 00000036 a223a1c0 a244fb50 b166cc5d b40c64b4
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: dfe0: 00000036 beb5e858 b3d96a49 b3d124e6
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: ---[ end trace 0000000000000000 ]---
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf ad8b427c in active state
Feb 27 11:47:01 pi-ibbdis-865a7b kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 999aa935 in active state
`

tetsuoMikami, 2023-09-03:
I found a similar issue.
I observed a video buffer crash when converting a UVC camera's MJPEG to H.264 and delivering it via RTSP.
The time from the start of RTSP distribution until the crash occurs is approximately 1 to 3 hours.
Crash happened in both RPi 4B and RPi 3B.

kernel 6.1.21-v8+ on RPi 4B

```
[ 6246.641801] ------------[ cut here ]------------
[ 6246.641822] WARNING: CPU: 1 PID: 894 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 6246.641881] Modules linked in: cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep snd_usb_audio uvcvideo snd_hwdep snd_usbmidi_lib cdc_ether snd_rawmidi snd_seq_device hci_uart btbcm bluetooth ecdh_generic ecc libaes 8021q garp stp llc brcmfmac brcmutil vc4 cfg80211 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper rfkill drm_kms_helper bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_codec(C) rpivid_hevc(C) v3d bcm2835_mmal_vchiq(C) v4l2_mem2mem raspberrypi_hwmon snd_soc_core videobuf2_vmalloc videobuf2_dma_contig gpu_sched videobuf2_memops snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm drm_shmem_helper videobuf2_v4l2 snd_timer videobuf2_common i2c_brcmstb snd syscopyarea videodev vc_sm_cma(C) sysfillrect sysimgblt mc fb_sys_fops uio_pdrv_genirq nvmem_rmem uio drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[ 6246.642197] CPU: 1 PID: 894 Comm: pool-test-launc Tainted: G         C         6.1.21-v8+ #1642
[ 6246.642210] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[ 6246.642217] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 6246.642229] pc : __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 6246.642266] lr : __vb2_queue_cancel+0x3c/0x220 [videobuf2_common]
[ 6246.642301] sp : ffffffc00949bae0
[ 6246.642307] x29: ffffffc00949bae0 x28: ffffffdcbb0a2938 x27: 0000000000000000
[ 6246.642327] x26: 0000000040045613 x25: ffffff8043985528 x24: 0000000000000000
[ 6246.642346] x23: ffffff80516aa600 x22: ffffffdd346ad858 x21: ffffff8048e16028
[ 6246.642364] x20: ffffff8048e16028 x19: ffffff8048e16028 x18: ffffff80468c0028
[ 6246.642382] x17: 0000000133ae8f08 x16: ffffffdd34369bb0 x15: ffffff808f43b978
[ 6246.642400] x14: 0000000000000000 x13: ffffff80468c0028 x12: ffffff808f43b908
[ 6246.642418] x11: 0000000000000000 x10: ffffff80468c0030 x9 : ffffffdcbb20b24c
[ 6246.642435] x8 : ffffff808f43b930 x7 : 0000000000000000 x6 : 0000000000000001
[ 6246.642452] x5 : ffffff804398581c x4 : ffffff8040fddc40 x3 : ffffff80468c0008
[ 6246.642470] x2 : 0000000000000002 x1 : 0000000000000000 x0 : 0000000000000006
[ 6246.642487] Call trace:
[ 6246.642492]  __vb2_queue_cancel+0x1b8/0x220 [videobuf2_common]
[ 6246.642529]  vb2_core_streamoff+0x2c/0xc8 [videobuf2_common]
[ 6246.642564]  vb2_streamoff+0x24/0x78 [videobuf2_v4l2]
[ 6246.642594]  v4l2_m2m_streamoff+0x44/0x130 [v4l2_mem2mem]
[ 6246.642628]  v4l2_m2m_ioctl_streamoff+0x20/0x30 [v4l2_mem2mem]
[ 6246.642659]  v4l_streamoff+0x2c/0x38 [videodev]
[ 6246.642743]  __video_do_ioctl+0x184/0x3d8 [videodev]
[ 6246.642822]  video_usercopy+0x2d8/0x740 [videodev]
[ 6246.642899]  video_ioctl2+0x20/0x38 [videodev]
[ 6246.642977]  v4l2_ioctl+0x48/0x68 [videodev]
[ 6246.643054]  __arm64_sys_ioctl+0xb0/0xf0
[ 6246.643074]  invoke_syscall+0x4c/0x110
[ 6246.643090]  el0_svc_common.constprop.3+0xfc/0x120
[ 6246.643104]  do_el0_svc+0x34/0xd0
[ 6246.643117]  el0_svc+0x30/0x88
[ 6246.643131]  el0t_64_sync_handler+0x98/0xc0
[ 6246.643143]  el0t_64_sync+0x18c/0x190
[ 6246.643154] ---[ end trace 0000000000000000 ]---
[ 6246.643278] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000013046bea in active state
[ 6246.643300] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000267e1628 in active state
[ 6246.643366] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000f387843c in active state
[ 6246.643382] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000008d73a95c in active state
[ 6246.643396] videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000009965ce5a in active state
[ 6246.643411] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000b599f765 in active state

```

nicoske, 2023-09-08:
Still have the issue as well on latest kernel 6.1.51-v8 with gstreamer 1.22.2 and wpewebkit 2.40.3.
I'm just playing a 1080p video in loop..

```
Sep 08 12:06:19 pi-ibbdis-769213 kernel: ------------[ cut here ]------------
Sep 08 12:06:19 pi-ibbdis-769213 kernel: WARNING: CPU: 0 PID: 427 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x200/0x280 [videobuf2_common]
Sep 08 12:06:19 pi-ibbdis-769213 kernel: Modules linked in: cfg80211 rfkill vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper snd_soc_core snd_compress v3d snd_bcm2835(C) raspberrypi_hwmon snd_pcm_dmaengine snd_pcm bcm2835_v4l2(C) bcm2835_codec(C) gpu_sched rpivid_hevc(C) bcm2835_isp(C) i2c_brcmstb drm_shmem_helper bcm2835_mmal_vchiq(C) snd_timer v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops syscopyarea snd vc_sm_cma(C) sysfillrect videobuf2_v4l2 videobuf2_common sysimgblt fb_sys_fops videodev nvmem_rmem uio_pdrv_genirq uio ipt_REJECT nf_reject_ipv4 xt_limit xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_addrtype xt_tcpudp iptable_filter sch_fq_codel drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Sep 08 12:06:19 pi-ibbdis-769213 kernel: CPU: 0 PID: 427 Comm: WPEWebProcess Tainted: G        WC         6.1.51-v8 #1
Sep 08 12:06:19 pi-ibbdis-769213 kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
Sep 08 12:06:19 pi-ibbdis-769213 kernel: pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Sep 08 12:06:19 pi-ibbdis-769213 kernel: pc : __vb2_queue_cancel+0x200/0x280 [videobuf2_common]
Sep 08 12:06:19 pi-ibbdis-769213 kernel: lr : __vb2_queue_cancel+0x38/0x280 [videobuf2_common]
Sep 08 12:06:19 pi-ibbdis-769213 kernel: sp : ffffffc008d73ae0
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x29: ffffffc008d73ae0 x28: ffffffe2563e0d08 x27: ffffff8041e72528
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x26: ffffffe2563e0f68 x25: 0000000040045613 x24: ffffff8053969400
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x23: 0000000000000000 x22: ffffff8046adf028 x21: ffffff8046adf028
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x20: 0000000000000009 x19: ffffff8046adf028 x18: ffffffc088d73787
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x17: 0003bbfdb54d3510 x16: ffffffe278fdeaa0 x15: ffffff80fb7da240
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x14: 0000000000000019 x13: ffffff8045c00028 x12: 0000000000000000
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x11: ffffff8047b3e7b8 x10: ffffff8047b3e6c8 x9 : ffffffe25644a98c
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x8 : ffffff8047b3e6f0 x7 : 0000000000000001 x6 : ffffffffc0000000
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x5 : ffffff8041e7281c x4 : 0000000000000000 x3 : ffffff8045c00008
Sep 08 12:06:19 pi-ibbdis-769213 kernel: x2 : 0000000000000000 x1 : 00000000000051a8 x0 : 0000000000000002
Sep 08 12:06:19 pi-ibbdis-769213 kernel: Call trace:
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  __vb2_queue_cancel+0x200/0x280 [videobuf2_common]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  vb2_core_streamoff+0x2c/0xd0 [videobuf2_common]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  vb2_streamoff+0x24/0x80 [videobuf2_v4l2]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  v4l2_m2m_streamoff+0x80/0x160 [v4l2_mem2mem]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  v4l2_m2m_ioctl_streamoff+0x20/0x30 [v4l2_mem2mem]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  v4l_streamoff+0x28/0x40 [videodev]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  __video_do_ioctl+0x18c/0x3e0 [videodev]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  video_usercopy+0x21c/0x790 [videodev]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  video_ioctl2+0x20/0x40 [videodev]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  v4l2_ioctl+0x48/0x70 [videodev]
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  __arm64_sys_ioctl+0xac/0xf0
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  invoke_syscall+0x50/0x120
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  el0_svc_common.constprop.0+0x104/0x130
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  do_el0_svc+0x38/0xe0
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  el0_svc+0x30/0xa0
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  el0t_64_sync_handler+0xbc/0x140
Sep 08 12:06:19 pi-ibbdis-769213 kernel:  el0t_64_sync+0x18c/0x190
Sep 08 12:06:19 pi-ibbdis-769213 kernel: ---[ end trace 0000000000000000 ]---
Sep 08 12:06:19 pi-ibbdis-769213 kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000829dd968 in active state
Sep 08 12:06:19 pi-ibbdis-769213 kernel: videobuf2_common: driver bug: stop_streaming operation is leaving buf 000000008fbd3223 in active state
Sep 08 12:06:20 pi-ibbdis-769213 kernel: swiotlb_tbl_map_single: 280 callbacks suppressed
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 6885376 bytes), total 32768 (slots), used 94 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 6885376 bytes), total 32768 (slots), used 94 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 1855488 bytes), total 32768 (slots), used 212 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 1855488 bytes), total 32768 (slots), used 212 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 6885376 bytes), total 32768 (slots), used 94 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 6885376 bytes), total 32768 (slots), used 94 (slots)
[... cut ...] rm gpu: swiotlb buffer is full (sz: 1855488 bytes), total 32768 (slots), used 212 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 1855488 bytes), total 32768 (slots), used 212 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 6885376 bytes), total 32768 (slots), used 94 (slots)
Sep 08 12:06:20 pi-ibbdis-769213 kernel: vc4-drm gpu: swiotlb buffer is full (sz: 6885376 bytes), total 32768 (slots), used 94 (slots)

```

6by9, 2023-09-08:
The warnings
```
WARNING: CPU: 0 PID: 427 at drivers/media/common/videobuf2/videobuf2-core.c:2035 __vb2_queue_cancel+0x200/0x280 [videobuf2_common]
videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000829dd968 in active state
```
are just that - warnings, not errors.

If you have an issue, they are symptoms and not the cause.

nicoske, 2023-09-08:
@6by9 Sorry I wasn't specific enough, the video still plays but after a few hours it stops with the following error:

```
Sep 08 13:56:50 pi-ibbdis-769213 kernel: bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep 08 13:56:50 pi-ibbdis-769213 kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Failed disabling o/p port, ret -62
Sep 08 13:56:52 pi-ibbdis-769213 kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_flush_buffers: Timeout waiting for buffers to be returned - 4 outstanding
Sep 08 13:56:55 pi-ibbdis-769213 kernel: bcm2835_mmal_vchiq: timed out waiting for sync completion
Sep 08 13:56:55 pi-ibbdis-769213 kernel: bcm2835-codec bcm2835-codec: bcm2835_codec_stop_streaming: Failed enabling component, ret -62

```

6by9, 2023-09-08:
```timed out waiting for sync completion``` generally means that the firmware has given up totally.
Without a way to reproduce this, we can't really debug what has happened in the firmware. "sudo vclog -m" may give some useful information, but no guarantees.

nicoske, 2023-09-12:
After some verification I've noticed that the allocated GPU memory was only 78Mb. There's some confusion on how to configure it for KMS driver.

Config.txt:

```
[all]
disable_splash=1
kernel=u-boot.bin
disable_overscan=1
dtparam=audio=on
display_auto_detect=1
dtoverlay=disable-wifi
dtoverlay=disable-bt

[pi4]
dtoverlay=vc4-kms-v3d,cma-256
arm_64bit=1
arm_boost=1

```

I thought "cma-256" would dynamically allow memory to the GPU when needed but it seems not the case.
Adding gpu_mem=256 solved the issue..

popcornmix, 2023-09-12:
The default gpu_mem setting should be good for Level 4.1 H.264.
Are you using a higher level, or decoding more than one stream at once?

nicoske, 2023-09-14:
Only one stream, here are the specs:

```
Format                                   : MPEG-4
Format profile                           : Base Media / Version 2
Codec ID                                 : mp42 (mp41/isom)
File size                                : 5.04 MiB
Duration                                 : 15 s 600 ms
Overall bit rate                         : 2 712 kb/s
Frame rate                               : 25.000 FPS
Encoded date                             : 2023-09-07 12:41:48 UTC
Tagged date                              : 2023-09-07 12:41:48 UTC

Video
ID                                       : 1
Format                                   : AVC
Format/Info                              : Advanced Video Codec
Format profile                           : Constrained Baseline@L4
Format settings                          : 2 Ref Frames
Format settings, CABAC                   : No
Format settings, Reference frames        : 2 frames
Format settings, GOP                     : M=1, N=25
Codec ID                                 : avc1
Codec ID/Info                            : Advanced Video Coding
Duration                                 : 15 s 600 ms
Bit rate                                 : 2 711 kb/s
Width                                    : 1 920 pixels
Height                                   : 1 080 pixels
Display aspect ratio                     : 16:9
Frame rate mode                          : Constant
Frame rate                               : 25.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Bits/(Pixel*Frame)                       : 0.052
Stream size                              : 5.04 MiB (100%)
Encoded date                             : 2023-09-07 12:41:48 UTC
Tagged date                              : 2023-09-07 12:41:48 UTC
Codec configuration box                  : avcC

```

I'll have a look to compile vclog to the target device

popcornmix, 2023-09-14:
Is there anything your code is doing that may allocate additional gpu_mem?

Are you confident increasing gpu_mem really fixes the issue completely (rather than making the failure take ~4x longer to occur?)

nicoske, 2024-05-13:
Hi, better late than never. We are still facing the issue with small length videos. When looped continuously seem to cause the buffer not to clean up correctly, eventually leading to a system freeze after several hours. 

Below is the Python script used to instantiate a playlist of videos in a loop:

```
	import gi
	gi.require_version('Gst', '1.0')
	from gi.repository import Gst, GLib
	import json

	# Initialize GStreamer
	Gst.init(None)

	video_pipeline = "playbin uri={url}"

	def load_media_list_from_json(filename):
		with open(filename, 'r') as file:
			return json.load(file)

	def play_media(media):
		loop = GLib.MainLoop()
		pipeline_str = video_pipeline.format(url=media["url"])

		pipeline = Gst.parse_launch(pipeline_str)

		def on_eos(bus, message, loop):
			stop_media(pipeline, loop)

		bus = pipeline.get_bus()
		bus.add_signal_watch()
		bus.connect("message::eos", on_eos, loop)

		print("Playing:", media["url"])
		pipeline.set_state(Gst.State.PLAYING)

		loop.run()

	def stop_media(pipeline, loop):
		print("Stopping playback")
		pipeline.set_state(Gst.State.NULL)
		# Add a short timeout to ensure pipeline state change completes
		GLib.timeout_add_seconds(0.1, lambda: play_next_media(loop))

	def play_next_media(loop):
		loop.quit()

	def main():
		media_list = load_media_list_from_json("media_playlist.json")
		print(len(media_list), "items found in the playlist")
		index = 0
		while True:
			play_media(media_list[index])
			index = (index + 1) % len(media_list)

	if __name__ == "__main__":
		main()
```

The media_playlist.json contains 2 videos:

```
[
    {
        "type": "video",
        "url": "https://github.com/chthomos/video-media-samples/raw/master/big-buck-bunny-1080p-30sec.mp4"
    },
    {
        "type": "video",
        "url": "http://commondatastorage.googleapis.com/gtv-videos-bucket/sample/ForBiggerBlazes.mp4"
    }
[... 111 lines cut ...]
350334.582: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
350390.924: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
350390.958: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
382876.827: video_decode:11:RIL: colourspace changing
413109.060: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
413109.097: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
413151.223: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
413151.243: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
413940.639: video_decode:12:RIL: nothing changing
483379.470: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
483379.503: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
483434.298: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
483434.320: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
515832.229: video_decode:13:RIL: colourspace changing
546077.508: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
546077.546: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
546118.268: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
546118.287: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
546891.816: video_decode:14:RIL: nothing changing
616334.649: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
616334.670: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
616390.928: mmalsrv: mmal_server_do_buffer_from_host: pool exhausted, recyling bust, on comp 64 port 2
616390.963: mmal: mmal_port_event_send: event lost on port 1,0 (buffer header callback not defined)
648827.271: video_decode:15:RIL: colourspace changing
```

nicoske, 2024-05-13:
Forgot to mention that I tested this on RPI3, RPI3B+, RPI4 running on latest kernel 6.6.30-v8 with latest firmware.

nicoske, 2024-05-14:
I managed to retrieve some kernel messages ater the crash/freeze. Unfortunately It's not complete (missing the exact crash timestamp).: This message is repeating itself every minute:

```
[ 1374.677622] rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
[ 1374.683717] rcu:     0-...0: (3 ticks this GP) idle=02fc/1/0x4000000000000002 softirq=156354/156354 fqs=3056780
[ 1374.693550] rcu:     (detected by 1, t=8329707 jiffies, g=334681, q=1924 ncpus=4)
[ 1374.700771] Task dump for CPU 0:
[ 1374.703995] task:swapper/0       state:R  running task     stack:0     pid:0     ppid:0      flags:0x0000000a
[ 1374.713917] Call trace:
[ 1374.716359]  __switch_to+0xe0/0x160
[ 1374.719853]  cpu_number+0x0/0x8
```

nicoske, 2024-09-16:
Could we please find a solution for this issue? Looping videos that are at least 15 seconds each does seem to stabilize things somewhat, but we’re still seeing numerous warnings in the logs. This makes it quite challenging to read and diagnose other potential issues. Any assistance would be greatly appreciated!

6by9, 2024-09-16:
The WARN triggered by GStreamer (and other apps using VIDIOC_CREATE_BUFS) has been fixed with #6348 (merged last week).

nicoske, 2024-09-17:
Thank you @6by9 ! Testing now and indeed the message is gone, no issue so far !