raspberrypi/linux #3795: USB Audio dropouts are back

alexivkin, 2020-08-14:
USB sound DAC pops and drops out periodically. dmesg is filling up with NYET messages:
```
[ 2991.651926] Transfer to device 6 endpoint 0x4 frame 762 failed - FIQ reported NYET. Data may have been lost.
[ 2999.907920] Transfer to device 6 endpoint 0x4 frame 826 failed - FIQ reported NYET. Data may have been lost.
```
Happening on 5.4 kernel, 4.18 is working fine. 

`dwc_otg.fiq_fsm_mask` of 0x5 cleared dmesg, but the cracks and pops are still happening. 0x7 and 0xF seem to have no effect, but that's expected on a [3 bit mask](https://www.raspberrypi.org/forums/viewtopic.php?t=70437). Happens under any load, high and low. `dwc_otg.fiq_fsm_enable=0` is the only reliable fix. 

Seems to be a regression of what was fixed in https://github.com/raspberrypi/linux/issues/586 and https://github.com/raspberrypi/linux/issues/575

**System**
* RPI 3B+ with proper power supply and not overclocked
* 5.4.0-1015-raspi aarch64 kernel. Standard ubuntu build `CONFIG_DMA_CMA=y`, CMA at 5Mb
* USB tree has only the DAC (TI PCM2902) and LAN on the dc_otg hub
* Official RPi touchscreen which I guess is handled by the hid_generic module, not usbhid

P33M, 2020-08-14:
Does the issue occur if you use Raspberry Pi OS?

alexivkin, 2020-08-14:
Not sure how to get that kernel on it. The stock 4.18 does not exhibit that issue.

P33M, 2020-08-14:
From a default install, run `sudo apt update && sudo apt full-upgrade` - this will install, among other things, 5.4 kernel and supporting packages.

Add the line `arm_64bit=1` to /boot/config.txt and reboot.

alexivkin, 2020-08-15:
Reproduced it on Raspberry Pi OS (hmm, that's a lot longer to type than raspbian ;) with `5.4.51-v8+ #1333 SMP PREEMEPT aarch64`
Subjectively there are lot less NYET's in dmesg compared to ubuntu, but that's a fresh Lite install vs the fully loaded Ubuntu.
Another thing that may help is after running many hours with "fiq_fsm_enable=0" the kernel trips on retire_capture_urb callbacks and resets the USB DAC

alexivkin, 2020-08-21:
Tested on RPi4 with the latest EEPROM firmware and no longer see these messages. Could it be that the firmware fixed it? 
BTW I noticed that ubuntu runs `dwc_otg.lpm_enable=0` by default, whereis raspbian does not.

alexivkin, 2020-09-07:
Thank you for the explanation!
It'd be great if the dwc driver code would warn and be off-by-detault on the PRi 3B/B+ in 64bit mode to avoid latency glitches...but that's probably an unrealistic ask, so I'll just close the ticket ;)

closed 2020-09-07 by alexivkin

iucoen, 2020-12-21:
I still have this issue with Pi 1 with the latest Raspberry OS playing 176kHz/32-bit PCM (actually it's DSD64 packed into DSD-over-PCM). There are a constant popping sounds, almost like background static.

With Pi2, it's a much better, but once in a way I still hear some popping noises. setting force_turbo=1 seems to help a bit as well. But I'm now close to maxing out CPU. The thread write PCM samples to hw buffer is using 1.3% CPU. So I guess it has something to do with frequency scaling.