raspberrypi/linux #3796: 64 bit bcmrpi3_defconfig CONFIG_USB_GADGET is not set

Dark-Sky, 2020-08-14:
Is there a reason why CONFIG_USB_GADGET is not enabled after running make bcmrpi3_defconfig.

pelwell, 2020-08-14:
Only that is a user-contributing defconfig and not one that we use. Do you just want "CONFIG_USB_GADGET=m" or "CONFIG_USB_GADGET=y", or is there something else?

Dark-Sky, 2020-08-14:
Guess it does not make any difference. Seems like you have it by default CONFIG_USB_GADGET=y in all of the other defconfigs. I compile all kernels here. My dvb devices will not work with it disabled.

Dark-Sky, 2020-08-14:
As you can see a multitude of dvb modules will not get compiled. There are a ton more not showing in this window. bcmrpi3_defconfig is the right hand pane.

[image]

pelwell, 2020-08-14:
Done - see https://github.com/raspberrypi/linux/commit/96be00073b63c170b6916f35abb58748bc8f6d26.

Dark-Sky, 2020-08-14:
Thanks. Will save me some time in the future building kernels.

closed 2020-08-14 by Dark-Sky