raspberrypi/linux #3802: Support mmc DDR52 timing in the mmc driver for eMMC modules

t-linden, 2020-08-18:
According to page 18 in https://elinux.org/images/9/91/Clement-sd-mmc-high-speed-support-in-linux-kernel_0.pdf
the DDR mode for eMMC was introduced with MMC 4.4 in 2009. It can run at up to 52 MHz and used at 3 V.
This is useful from the point of  view of eMMC modules attached with a passive SD-card adapter to a Raspberry Pi as this would allow the available bandwidth to double without lowering the voltage to 1.8 V. On Raspberrry Pi models preceeding the Raspberry Pi Model 4 B this is important, because they don't support 1.8 V at all. The current mmc driver on the Raspberry Pi uses the timing "mmc high-speed", but using the timing "mmc DDR52" would be needed to achieve the double performance.

**To reproduce**
Insert an eMMC module on a SD-adapter into a Raspberry Pi. The maximum read performance is about 20 MB/s.

**Expected behaviour**
Insert an eMMC module on a SD-adapter into a Raspberry Pi. The expected maximum read performance would be about 40 MB/s.

**Actual behaviour**
Even if a eMMC module supports the timing "mmc DDR52" the mmc driver does not enable this mode.

**System**
See some details in this link at https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=282639

pelwell, 2020-08-18:
If you can point out the change which needs to be made we'll consider it, but without an DDR52-capable EMMC and an SD adaptor our options are limited.

t-linden, 2020-08-18:
AFAIK your colleague timg236 has two units of the RasPiKey which is an integrated eMMC-uSD module which is DDR52 capable according to the manufacturer. If you can get one of those then that could be helpful.
https://github.com/raspberrypi/firmware/issues/1235
http://www.uugear.com/product/raspikey-plug-and-play-emmc-module-for-raspberry-pi/

The needed driver patch should check for the capabilities of the eMMC module and do some checks on those. 
On the Raspberry Pi 4 B the hardware can do 1.8 V so to save power one should run at that voltage when switching to DDR52.
On the previous Raspberry Pi models one would need to run at 3.3 V with DDR52.

There is already a patch that apparently does this, please see:
https://src.fedoraproject.org/rpms/kernel/blob/606a7e904bc8f49a3dcac2c59e3006a55711662b/f/arm-sdhci-esdhc-imx-fixes.patch

timg236, 2020-08-19:
This looks like something which should be added to the upstream sdhci-iproc driver and maybe backported.

CM4 has EMMC support so that would be the best test

pelwell, 2020-08-23:
That's the SD1 interface, used for SDIO to the WiFi controller.

lategoodbye, 2020-08-23:
@t-linden In case you are able to rebuild the Raspberry Pi 4 DTB, please try this patch:

https://gist.github.com/lategoodbye/583ef0d7cd74a39d6c8f7d7f3d2c3613

t-linden, 2020-08-23:
Which driver is then the default one in use on the BCM2711?

lategoodbye, 2020-08-23:
> Which driver is then the default one in use on the BCM2711?

sdhci-iproc for emmc2 interface

t-linden, 2020-08-23:
Thanks! Your patch works!

cat /sys/kernel/debug/mmc0/ios
clock:          52000000 Hz
actual clock:   50000000 Hz
vdd:            21 (3.3 ~ 3.4 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    8 (mmc DDR52)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)

 hdparm -Tt /dev/mmcblk0p2 

/dev/mmcblk0p2:
 Timing cached reads:   1514 MB in  2.00 seconds = 756.95 MB/sec
 HDIO_DRIVE_CMD(identify) failed: Invalid argument
 Timing buffered disk reads: 134 MB in  3.01 seconds =  44.45 MB/sec

From dtc -I fs /proc/device-tree
>        emmc2bus {
                compatible = "simple-bus";
                ranges = < 0x00 0x7e000000 0x00 0xfe000000 0x1800000 >;
                #address-cells = < 0x02 >;
                #size-cells = < 0x01 >;
                phandle = < 0xce >;
                dma-ranges = < 0x00 0xc0000000 0x00 0x00 0x40000000 >;

                emmc2@7e340000 {
                        compatible = "brcm,bcm2711-emmc2";
                        clocks = < 0x07 0x33 >;
                        broken-cd;
                        status = "okay";
                        interrupts = < 0x00 0x7e 0x04 >;
                        mmc-ddr-3_3v;
                        phandle = < 0x3a >;
                        reg = < 0x00 0x7e340000 0x100 >;
                        vmmc-supply = < 0x25 >;
                        vqmmc-supply = < 0x24 >;
                };

source /usr/share/agnostics/sdtest.sh
Run 1
prepare-file;0;0;33625;65
seq-write;0;0;37406;73
rand-4k-write;0;0;22716;5679
rand-4k-read;22459;5614;0;0
Sequential write speed 37406 KB/sec (target 10000) - PASS
Random write speed 5679 IOPS (target 500) - PASS
Random read speed 5614 IOPS (target 1500) - PASS

lategoodbye, 2020-08-23:
In theory i should be possible to add "mmc-ddr-1_8v;".

Yes, it was really an issue in sdhci-iproc. I thought that the mmc_caps were initialized from the sdhci register, but they are left to zero. So the devicetree change isn't the final solution.

t-linden, 2020-08-24:
Thanks for the help with this! What is the sdhci register?

t-linden, 2020-08-24:
I compiled bcm2711-rpi-4-b.dts with mmc-ddr-1_8v; instead. The kernel starts to boot but is repeating the error message:
mmc0: error -110 whilst initialising MMC card

pelwell, 2020-08-24:
The intention is for the common sdhci driver to learn the capabilities of the SDHCI interface by reading the interface's CAPABILITIES register. The fact that declaring the capability in the Device Tree makes it work suggests that mechanism isn't working for some reason. However, I'm willing to apply the DT change to bcm2711-rpi.dtsi in the meantime so that all bcm2711-based devices pick it up.

pelwell, 2020-08-24:
See f1280b6.

t-linden, 2020-08-24:
Thank you very much for committing the patch!

t-linden, 2020-08-25:
If I apply the patch [https://github.com/raspberrypi/linux/commit/5e55b61a665e4f8d7fe939155b7baa32a16d0e06](url) then the booting hangs by repeating the error message
`mmc0: error -110 whilst initialising MMC card`
The eMMC module is a Samsung KLMBG2JETD specified as "eMMC 5.1 | 32 GB | 1.8, 3.3 V / 3.3 V | HS400" [https://www.samsung.com/semiconductor/estorage/emmc/](url).

t-linden, 2020-08-25:
To run the eMMC module with DDR52 and a signal voltage of 1.8 V the supply voltage should still be 3.3 V. I would assume something like this patch is needed: [http://patchwork.ozlabs.org/project/linux-tegra/patch/20180831163817.23970-24-marcel@ziswiler.com/](url)?

What is this `vqmmc-supply = <&sd_io_1v8_reg>;` line supposed to do?

uugear, 2020-08-28:
Hi @t-linden, I am Shawn from UUGear. Have you managed to run the eMMC module with DDR52 mode? We are quite excited to check if RasPiKey can do work with DDR52 too. Are you willing to share more details with us? You can drop me an email: shawn [at] uugear [dot] com, thank you in advance.

t-linden, 2020-08-28:
@uugear, as you can see from the discussion DDR52 mode works on the BCM2711 with 3.3 V. So far it has only been tested with the eMMC module that I have.

t-linden, 2020-09-01:
I tried enabling DDR52 with 3.3 V on a Raspberry 3 B with this change:

diff --git a/a/bcm2708-rpi.dtsi b/b/bcm2708-rpi.dtsi
index 2c440e5..c728ab9 100644
--- a/a/bcm2708-rpi.dtsi
+++ b/b/bcm2708-rpi.dtsi
@@ -23,6 +23,7 @@
 &sdhost {
        pinctrl-names = "default";
        pinctrl-0 = <&sdhost_gpio48>;
+       mmc-ddr-3_3v;
        status = "okay";
 };

The eMMC module is recognized, but when the partitions should be mounted a dump is given by the card driver.
Any ideas why the driver crashes and how to fix that?

Raspberry Pi Bootcode
Read File: config.txt, 1849
Read File: start.elf, 3008032 (bytes)
Read File: fixup.dat, 7288 (bytes)
MESS:00:00:02.102687:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.107644:0: brfs: File read: 1849 bytes
MESS:00:00:02.138037:0: HDMI:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.144108:0: HDMI:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.150358:0: HDMI:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.156609:0: HDMI:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.162859:0: HDMI:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.169109:0: HDMI:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.175359:0: HDMI:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.181609:0: HDMI:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.187859:0: HDMI:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.194109:0: HDMI:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.200117:0: HDMI:EDID giving up on reading EDID block 0
MESS:00:00:02.207463:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.212334:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not define
d
MESS:00:00:02.230275:0: brfs: File read: 1849 bytes
MESS:00:00:02.424499:0: gpioman: gpioman_get_pin_num: pin DISPLAY_DSI_PORT not d
efined
MESS:00:00:02.431943:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not define
d
MESS:00:00:02.437968:0: *** Restart logging
MESS:00:00:02.442989:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.449847:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.456618:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.463388:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.470159:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.476930:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.483701:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.490472:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.497242:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.504013:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.510542:0: hdmi: HDMI:EDID giving up on reading EDID block 0
MESS:00:00:02.516356:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.524065:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.530836:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.537607:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.544378:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.551149:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.557920:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.564691:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.571462:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.578232:0: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.584761:0: hdmi: HDMI:EDID giving up on reading EDID block 0
MESS:00:00:02.590549:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_di
splay_state instead
MESS:00:00:02.599038:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:02.604733:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-i
d 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae:
[... 1456 lines cut ...]
[    4.709798] Segment Routing with IPv6
[    4.730762] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-
1.1, smsc95xx USB 2.0 Ethernet, XX:XX:XX:XX:XX:XX
[    4.750223] systemd[1]: systemd 241 running in system mode. (+PAM +AUDIT +SEL
INUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +
XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybr
id)
[    4.782018] systemd[1]: Detected architecture arm.
[    4.850230] systemd[1]: Set hostname to <raspberrypi>.
[    5.764188] random: systemd: uninitialized urandom read (16 bytes read)
[    5.787895] random: systemd: uninitialized urandom read (16 bytes read)
[    5.798669] systemd[1]: Listening on Journal Socket.
[    5.810528] random: systemd: uninitialized urandom read (16 bytes read)
[    5.828158] systemd[1]: Starting Restore / save the current clock...
[    5.841664] systemd[1]: Reached target Swap.
[    5.858920] systemd[1]: Starting Set the console keyboard layout...
[    5.872929] systemd[1]: Listening on initctl Compatibility Named Pipe.
[    5.887806] systemd[1]: Listening on udev Control Socket.
[    5.901143] systemd[1]: Set up automount Arbitrary Executable File Formats Fi
le System Automount Point.
[    6.120415] i2c /dev entries driver

Raspbian GNU/Linux 10 raspberrypi ttyS0

raspberrypi login:

pelwell, 2020-09-01:
You told Linux that BCM2837 was capable of driving a card using DDR, but it isn't, so when the MMC framework sends the necessary commands to switch to the DDR mode the communication breaks.

t-linden, 2020-09-01:
Is any of the other Raspberry Pi models other than the 4 B and CM4 capable of DDR52 timing? The manual for the BCM2835  https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf says on page 65: 

> 5 External Mass Media Controller
Introduction
The External Mass Media Controller (EMMC) is an embedded MultiMediaTM and SDTM card
interface provided by ArasanTM. It is compliant to the following standards:
• SDTM Host Controller Standard Specification Version 3.0 Draft 1.0
• SDIOTM card specification version 3.0
• SDTM Memory Card Specification Draft version 3.0
• SDTM Memory Card Security Specification version 1.01
• MMCTM Specification version 3.31,4.2 and 4.4

Supporting MMC 4.4 implies supporting DDR52.

pelwell, 2020-09-01:
Show me where it says that MMC 4.4 controllers must support Dual Data Rate transfers, i.e. that it isn't an optional feature.

t-linden, 2020-09-01:
In the BCM2835 manual there is a table on page 82 which reads:

18:16 UHSMODE RW 0x0
Select the speed mode of the SD card:
000 = SDR12
001 = SDR25
010 = SDR50
011 = SDR104
100 = DDR50
other = reserved

and on page 87 there is another register described:

Synopsis
This register is used to delay the card clock when sampling the returning data and
command response from the card. It determines by how many steps the sampling
clock is delayed in DDR mode.

Bit(s) Field Name Description Type Reset
31:6                      Reserved - Write as 0, read as don't care
5:0 Steps              Number of steps (0 to 4)   RW 0x0

Based on this documentation there is DDR50 capability in the hardware on the Raspberry Pi models preceeding the 4 B and CM4. Should that not be enough for an eMMC card to work in DDR mode at 3.3 V in principle?

pelwell, 2020-09-01:
That is SD1 - the modified Arasan MMC controller that we use to drive the SDIO WLAN interface. When I say "modified", cut-down might be more appropriate. Certain restriction exist because it drives generic GPIO pins rather than dedicated fixed-function pins. The register descriptions you see are for the source IP, rather than a description of that which is possible in this reduced implementation. Our GPIOs max-out at about 50MHz, so DDR50 is stretch too far.

I have a version of that same peripheral spec with more detail. It goes into some detail (which I won't share) about what has been omitted, but this sentence is interesting:

> VideoCore™ allows the software to control the SD™ clock delay line directly so it can also be used to compensate for the quite hard timing constraints when using SD™ cards in SDR25 mode.

SDR25 is considered "quite hard", so it's not surprising in a way that DDR50 doesn't get a mention - it's impossible.

t-linden, 2020-09-01:
Thanks @pelwell for this explicit answer.