raspberrypi/linux #3814: USB SSD disconnect after a while

SimoB, 2020-08-24:
My external SSD attached to one of the USB3 ports of my rpi4 is not accessible after some time, from the dmsg log seems to me it get disconnected, but I'm not sure.

The sata/usb thing I am using is one of the jmicron crappy one that needs UAS disabled to work well.

This setup has been ultra stable for months (with 0 problems and no reboots) until I think 2/3 weeks ago I did a sudo update && sudo full-upgrade, maybe some update messed up my system?

**System**
RPI 4 4gb with raspbian lite on sd card
```

simone@rpi-s1:~ $ uname -a
Linux rpi-s1 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

simone@rpi-s1:~ $ sudo rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu 16 Apr 17:11:26 UTC 2020 (1587057086)
 LATEST: Thu 16 Apr 17:11:26 UTC 2020 (1587057086)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad

simone@rpi-s1:~ $ lsusb
Bus 002 Device 003: ID 152d:0561 JMicron Technology Corp. / JMicron USA Technology Corp. JMS551 - Sharkoon SATA QuickPort Duo
Bus 002 Device 002: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

simone@rpi-s1:~ $ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 465.8G  0 disk
└─sda1        8:1    0 465.8G  0 part /mnt/ssd
sdb           8:16   0   2.7T  0 disk
└─sdb1        8:17   0   2.7T  0 part /mnt/blu
sdc           8:32   0   2.7T  0 disk
└─sdc1        8:33   0   2.7T  0 part /mnt/red
mmcblk0     179:0    0  29.7G  0 disk
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29.5G  0 part /
```

**Logs**
```
Linux rpi-s1 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Aug 23 11:01:25 2020 from 192.168.1.100
simone@rpi-s1:~ $ sudo dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.51-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1333 SMP Mon Aug 10 16:51:40 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 1012736
[    0.000000]   DMA zone: 2304 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 816128 pages, LIFO batch:63
[... 516 lines cut ...]
[114295.349167] usb 2-1: USB disconnect, device number 4
[114295.351702] sd 2:0:0:0: [sdb] Synchronizing SCSI cache
[114295.351918] sd 2:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=0x00
[114295.829566] usb 2-1: new SuperSpeed Gen 1 USB device number 5 using xhci_hcd
[114295.860982] usb 2-1: New USB device found, idVendor=152d, idProduct=0578, bcdDevice= 2.04
[114295.860999] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[114295.861014] usb 2-1: Product: External USB 3.0
[114295.861029] usb 2-1: Manufacturer: JMicron
[114295.861043] usb 2-1: SerialNumber: 201703310007F
[114295.863723] usb 2-1: UAS is blacklisted for this device, using usb-storage instead
[114295.863834] usb 2-1: UAS is blacklisted for this device, using usb-storage instead
[114295.863851] usb-storage 2-1:1.0: USB Mass Storage device detected
[114295.864331] usb-storage 2-1:1.0: Quirks match for vid 152d pid 0578: 1800000
[114295.864481] scsi host1: usb-storage 2-1:1.0
[114296.939945] scsi 1:0:0:0: Direct-Access     External USB3.0           0204 PQ: 0 ANSI: 6
[114296.940862] sd 1:0:0:0: Attached scsi generic sg1 type 0
[114296.941421] sd 1:0:0:0: [sdb] 976773168 512-byte logical blocks: (500 GB/466 GiB)
[114296.942362] sd 1:0:0:0: [sdb] Write Protect is off
[114296.942381] sd 1:0:0:0: [sdb] Mode Sense: 47 00 00 08
[114296.943270] sd 1:0:0:0: [sdb] Disabling FUA
[114296.943289] sd 1:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[114297.483006]  sdb: sdb1
[114297.486113] sd 1:0:0:0: [sdb] Attached SCSI disk

```

RyanHir, 2020-08-24:
Have you tried using a powered USB hub? 

I have had this issue before and a powered USB 3.0 hub was able to fix the problem. I know this does not explain the issues showing up after a update but it could be a power delivery issue.

SimoB, 2020-08-24:
> 
> 
> Have you tried using a powered USB hub?
> 
> I have had this issue before and a powered USB 3.0 hub was able to fix the problem. I know this does not explain the issues showing up after a update but it could be a power delivery issue.

No, but because I do weekly auto backups of my system I was able to roll back to a backup before the update and from what I see I was on kernel 4.19.

I'll report back if the problem is present with kernel 4.19 or not.

SimoB, 2020-08-26:
I can confirm that rolling back to kernel 4.19 resolved my problem with usb ssd disconnections.

Titaniumtown, 2020-08-26:
I can not reproduce this issue. I have a raspberry pi 4 8gb running 24/7 w/ usb drives w/o any issues. hmmmm

RyanHir, 2020-08-26:
That's quite interesting, I am not a kernel developer so I am not the most
aware but 4.19 is a LTS so you could just keep backporting the kernel by
recompiling any patches that are added in this repo.

This may be bad advice so please take it with a grain of salt.

RyanHir, 2020-08-26:
From experience, I think the issue is power related

Titaniumtown, 2020-08-26:
I've had issues w/ usb HDDs (as in the spinning kind) pulling to much power and the pi just shuts off, tries to turn back on, the hdd pulls to much power, and it's stuck like that.

wyuenho, 2022-03-29:
same problem here

wyuenho, 2022-03-29:
For what it's worth, I don't think this issue is related to the HDD not having enough power as I run mine on a powered USB hub. Even when I don't the drive is only reported to use a maximum of 896mA of current from `lsusb`, since I only have 1 device attached to any USB ports, and the Pi can supply 1.2A total, I don't think power is an issue.

I'm running a Samsung 980 Pro NVMe drive inside a RTL9210 enclosure, so the NVMe drive is treated as a USB Mass Storage device with UAS enabled. I too see that `fs/fs-writeback.c:2323 __mark_inode_dirty+0x310/0x424` line in the log, especially after getting hit a couple of times by #3404.

slash-spb, 2023-11-03:
` 6382.291188] sd 0:0:0:0: [sda] tag#5 uas_eh_abort_handler 0 uas-tag 3 inflight: CMD OUT 
[ 6382.291214] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x2a 2a 00 19 56 48 00 00 04 00 00
[ 6382.291370] sd 0:0:0:0: [sda] tag#4 uas_eh_abort_handler 0 uas-tag 5 inflight: CMD OUT 
[ 6382.291386] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 53 fd b5 10 00 04 00 00
[ 6382.291501] sd 0:0:0:0: [sda] tag#3 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD OUT 
[ 6382.291514] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x2a 2a 00 19 9d 85 e8 00 02 18 00
[ 6382.291650] sd 0:0:0:0: [sda] tag#2 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD OUT 
[ 6382.291663] sd 0:0:0:0: [sda] tag#2 CDB: opcode=0x2a 2a 00 19 56 50 00 00 01 18 00
[ 6382.291750] sd 0:0:0:0: [sda] tag#1 uas_eh_abort_handler 0 uas-tag 4 inflight: CMD OUT 
[ 6382.291762] sd 0:0:0:0: [sda] tag#1 CDB: opcode=0x2a 2a 00 19 56 4c 00 00 04 00 00
[ 6382.291868] sd 0:0:0:0: [sda] tag#0 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD OUT 
[ 6382.291880] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x2a 2a 00 53 fd b1 10 00 04 00 00
[ 6382.315191] scsi host0: uas_eh_device_reset_handler start
[ 6382.643282] usb 2-1.4: reset SuperSpeed USB device number 3 using xhci-hcd
[ 6382.668578] scsi host0: uas_eh_device_reset_handler success
[ 6392.774725] sd 0:0:0:0: [sda] tag#5 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD 
[ 6392.774747] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x0 00 00 00 00 00 00
[ 6392.774758] scsi host0: uas_eh_device_reset_handler start
[ 6398.150797] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6403.526264] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6403.734231] usb 2-1.4: device not accepting address 3, error -62
[ 6409.158022] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6414.533817] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6414.741819] usb 2-1.4: device not accepting address 3, error -62
[ 6420.165563] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6425.541335] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6425.749298] usb 2-1.4: device not accepting address 3, error -62
[ 6431.173082] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6436.548853] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[ 6436.756805] usb 2-1.4: device not accepting address 3, error -62
[ 6436.757713] scsi host0: uas_eh_device_reset_handler FAILED err -19
[ 6436.757734] sd 0:0:0:0: Device offlined - not ready after error recovery
[ 6436.757744] sd 0:0:0:0: Device offlined - not ready after error recovery
[ 6436.757751] sd 0:0:0:0: Device offlined - not ready after error recovery
[ 6436.757758] sd 0:0:0:0: Device offlined - not ready after error recovery
[ 6436.757765] sd 0:0:0:0: Device offlined - not ready after error recovery
[ 6436.757772] sd 0:0:0:0: Device offlined - not ready after error recovery
[ 6436.757813] sd 0:0:0:0: [sda] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=DRIVER_OK cmd_age=85s
[ 6436.757826] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x2a 2a 00 53 fd b1 10 00 04 00 00
[ 6436.757834] blk_update_request: I/O error, dev sda, sector 1409134864 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[ 6436.768898] sd 0:0:0:0: rejecting I/O to offline device
[ 6436.768924] sd 0:0:0:0: [sda] tag#1 UNKNOWN(0x2003) Result: hostbyte=0x03 driverbyte=DRIVER_OK cmd_age=85s
[ 6436.769264] usb 2-1.4: USB disconnect, device number 3
[ 6436.773512] sd 0:0:0:0: [sda] tag#1 CDB: opcode=0x2a 2a 00 19 56 4c 00 00 04 00 00
[ 6436.773513] blk_update_request: I/O error, dev sda, sector 1409136912 op 0x1:(WRITE) flags 0x0 phys_seg 125 prio class 0
[ 6436.773520] blk_update_request: I/O error, dev sda, sector 425085952 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[ 6436.784331] EXT4-fs warning (device sda1): ext4_end_bio:344: I/O error 10 writing to inode 13238281 starting block 176142238)
[ 6436.795309] EXT4-fs warning (device sda1): ext4_end_bio:344: I/O error 10 writing to inode 13238281 starting block 176142239)
[ 6436.795383] blk_update_request: I/O error, dev sda, sector 429756624 op 0x1:(WRITE) flags 0x0 phys_seg 9 prio class 0
[ 6436.795761] blk_update_request: I/O error, dev sda, sector 429780112 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[ 6436.805817] EXT4-fs warning (device sda1): ext4_end_bio:344: I/O error 10 writing to inode 13238281 starting block 53719587)
[ 6436.816805] blk_update_request: I/O error, dev sda, sector 429774320 op 0x1:(WRITE) flags 0x0 phys_seg 29 prio class 0
[ 6436.816850] Buffer I/O error on device sda1, logical block 53719322
[ 6436.817014] blk_update_request: I/O error, dev sda, sector 429781136 op 0x1:(WRITE) flags 0x0 phys_seg 17 prio class 0
[ 6436.817026] EXT4-fs warning (device sda1): ext4_end_bio:344: I/O error 10 writing to inode 13238281 starting block 53722659)
[ 6436.817043] blk_update_request: I/O error, dev sda, sector 425086976 op 0x1:(WRITE) flags 0x0 phys_seg 35 prio class 0
[ 6436.817051] EXT4-fs warning (device sda1): ext4_end_bio:344: I/O error 10 writing to inode 13238281 starting block 53135906)
[ 6436.817057] EXT4-fs warning (device sda1): ext4_end_bio:344: I/O error 10 writing to inode 13238281 starting block 53135907)
[ 6436.817067] blk_update_request: I/O error, dev sda, sector 429753832 op 0x1:(WRITE) flags 0x0 phys_seg 67 prio class 0
[... cut ...]
[ 6768.850503] EXT4-fs warning (device sda1): htree_dirblock_to_tree:1072: inode #2: lblock 0: comm smbd: error -5 reading directory block
[ 6768.859854] EXT4-fs warning (device sda1): htree_dirblock_to_tree:1072: inode #2: lblock 0: comm smbd: error -5 reading directory block
[ 6775.996954] EXT4-fs warning (device sda1): htree_dirblock_to_tree:1072: inode #2: lblock 0: comm smbd: error -5 reading directory block
`
Goodday!

It's seems i solve the problem aflter replacing short usb cable to long usb cable.

I have ssd usb drive