raspberrypi/linux #3817: spi-bcm2835 fe204000.spi: could not register SPI controller: -16

amahoneyLIT, 2020-08-26:
**Describe the bug**

spi seems to not work with raspberrypi-kernel_1.20200819-1_armhf.deb, when it worked in the last kernel I was using raspberrypi-kernel_1.20200212-1_armhf.deb

```
$ dmesg | grep spi
[    8.732885] spi-bcm2835 fe204000.spi: could not register SPI controller: -16
[    8.743644] spi-bcm2835: probe of fe204000.spi failed with error -16

$ls /dev/spi*
ls: cannot access '/dev/spi*': No such file or directory

$ cat /boot/config.txt | grep spi
dtparam=spi=on
```

**To reproduce**
Build raspbian with pi-gen and up to date local mirrors of repos containing raspberrypi-kernel_1.20200819-1_armhf.deb

**Expected behaviour**
we should have a `/dev/spidev.0.0`

**Actual behaviour**
No spi devices

**System**

* Which model of Raspberry Pi? RPi 4 B Rev 1.1
* Which OS and version - local fork of pi-gen. 
* Which firmware version (`vcgencmd version`)?
```
Aug 19 2020 17:38:16
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)
```
* Which kernel version (`uname -a`)? `Linux thrombolux 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux`

**Logs**
My last pi-gen build was successful with local snapshots of archive.raspbian.org and archive.raspberrypi.org frozen in March 2020 with raspberrypi-kernel_1.20200212-1_armhf.deb and spi worked as expected. Comparing the pi-gen build logs, notably the section of unpacking and setting up raspberrypi-kernel is interesting

See the following line no longer present
```
Removing 'diversion of /boot/overlays/spi0-hw-cs.dtbo to /usr/share/rpikernelhack/overlays/spi0-hw-cs.dtbo by rpikernelhack'
```

dmesg etc
```
$ dmesg | grep spi
[    8.732885] spi-bcm2835 fe204000.spi: could not register SPI controller: -16
[    8.743644] spi-bcm2835: probe of fe204000.spi failed with error -16

$ls /dev/spi*
ls: cannot access '/dev/spi*': No such file or directory

$ cat /boot/config.txt | grep spi
dtparam=spi=on
```

some sections from `dtc -Ifs /proc/device-tree`
```
 spi@7e204c00 {
                        compatible = "brcm,bcm2835-spi";
                        clocks = < 0x07 0x14 >;
                        status = "disabled";
                        #address-cells = < 0x01 >;
                        interrupts = < 0x00 0x76 0x04 >;
                        #size-cells = < 0x00 >;
                        phandle = < 0xb9 >;
                        reg = < 0x7e204c00 0x200 >;
                };

spi@7e204000 {
                        compatible = "brcm,bcm2835-spi";
                        clocks = < 0x07 0x14 >;
                        status = "okay";
                        #address-cells = < 0x01 >;
                        interrupts = < 0x00 0x76 0x04 >;
                        cs-gpios = < 0x0f 0x08 0x01 0x0f 0x07 0x01 >;
                        #size-cells = < 0x00 >;
                        dma-names = "tx\0rx";
                        phandle = < 0x30 >;
                        reg = < 0x7e204000 0x200 >;
                        pinctrl-0 = < 0x0d 0x0e >;
                        dmas = < 0x0b 0x06 0x0b 0x07 >;
                        pinctrl-names = "default";

                        spidev@1 {
                                compatible = "spidev";
                                #address-cells = < 0x01 >;
                                #size-cells = < 0x00 >;
                                phandle = < 0xa5 >;
                                reg = < 0x01 >;
                                spi-max-frequency = < 0x7735940 >;
                        };

                        spidev@0 {
                                compatible = "spidev";
                                #address-cells = < 0x01 >;
                                #size-cells = < 0x00 >;
                                phandle = < 0xa4 >;
                                reg = < 0x00 >;
                                spi-max-frequency = < 0x7735940 >;
                        };
                };

spi@7e204600 {
                        compatible = "brcm,bcm2835-spi";
                        clocks = < 0x07 0x14 >;
                        status = "disabled";
                        #address-cells = < 0x01 >;
                        interrupts = < 0x00 0x76 0x04 >;
                        #size-cells = < 0x00 >;
                        phandle = < 0xb6 >;
                        reg = < 0x7e204600 0x200 >;
                };

spi@7e204a00 {
                        compatible = "brcm,bcm2835-spi";
                        clocks = < 0x07 0x14 >;
                        status = "disabled";
                        #address-cells = < 0x01 >;
                        interrupts = < 0x00 0x76 0x04 >;
                        #size-cells = < 0x00 >;
                        phandle = < 0xb8 >;
                        reg = < 0x7e204a00 0x200 >;
                };

spi@7e204800 {
                        compatible = "brcm,bcm2835-spi";
                        clocks = < 0x07 0x14 >;
                        status = "disabled";
                        #address-cells = < 0x01 >;
                        interrupts = < 0x00 0x76 0x04 >;
                        #size-cells = < 0x00 >;
                        phandle = < 0xb7 >;
                        reg = < 0x7e204800 0x200 >;
                };

spi@7e215080 {
                        compatible = "brcm,bcm2835-aux-spi";
                        clocks = < 0x14 0x01 >;
                        status = "disabled";
                        #address-cells = < 0x01 >;
                        interrupts = < 0x00 0x5d 0x04 >;
                        #size-cells = < 0x00 >;
                        phandle = < 0xaa >;
                        reg = < 0x7e215080 0x40 >;
                };

spi@7e2150c0 {
                        compatible = "brcm,bcm2835-aux-spi";
                        clocks = < 0x14 0x02 >;
                        status = "disabled";
                        #address-cells = < 0x01 >;
                        interrupts = < 0x00 0x5d 0x04 >;
                        #size-cells = < 0x00 >;
                        phandle = < 0xab >;
                        reg = < 0x7e2150c0 0x40 >;
                };

```

**Additional context**
n/a

pelwell, 2020-08-26:
-16 is error -EBUSY. In the 5.4 kernel we've removed the downstream code that converts hardware CS lines to GPIO CSs. An unintended consequence of this has been to prevent the use of the unused SPI pins for other purposes. Do you have something in your configuration - likely an overlay - that could be claiming one of the GPIOs from 7 to 11?

amahoneyLIT, 2020-08-26:
@pelwell thanks for the details. Yes we're using `dtoverlay=i2c-rtc-gpio,rv3028,i2c_gpio_sda=6,i2c_gpio_scl=7`

Is this considered a bug or a permanent design change? I couldn't find any related issues. It would require a hardware design change for me to not use GPIO 7 in my design.

pelwell, 2020-08-26:
It's fixable with another overlay (not yet written) that removes the second CS line on GPIO7.

pelwell, 2020-08-28:
There's a new overlay in the repo - `spi0-1cs` - that does what you want. It also includes a `no_miso` parameter that frees up GPIO 9 if using devices that are write-only. `spi0-cs` has been renamed (with an auto-redirect) to `spi0-2cs` for consistency with this and the `spi1-` and spi2-` overlays, and also gains the `no_miso` parameter.

amahoneyLIT, 2020-08-28:
Thanks, I can report your changes work (at least spi0-1cs). Before applying the new overlays I also tested spi0-cs, which I was unaware of, and moving CS1 to an unused GPIO pin also works for my design.

In your patch, I noticed
```
Name:   spi0-cs
Info:   This overlay has been renamed spi0-1cs, keeping spi0-cs as an
        alias for backwards compatibility.
```

but to me it seems like spi0-cs (with existing parameters cs0_pin and cs1_pin) should be an alias to spi0-2cs, since they have the same params and function. 

Details on how I tested the new overlays:

```
git clone --depth=1 https://github.com/raspberrypi/linux.git
cd linux
make bcm2709_defconfig
make dtbs
sudo cp ./arch/arm/boot/dts/overlays/overlay_map.dtb /boot/overlays
sudo cp ./arch/arm/boot/dts/overlays/spi0-1cs.dtbo /boot/overlays
sudo cp ./arch/arm/boot/dts/overlays/spi0-2cs.dtbo /boot/overlays
sudo mv /boot/overlays/spi0-cs.dtbo /boot/overlays/spi0-cs.dtbo.old
# and add `dtoverlay=spi0-1cs` to /boot/config.txt
```

In my code I tested with, curiously I can no longer open("/dev/spidev0.0", O_RDWR) as the pi user, it returns -1. I have to be root for it to work. The pi user is members of the right groups I think, dialout, gpio, spi. Not sure if this is a change going to the 5.4 kernel. Any ideas?

pelwell, 2020-08-28:
Thanks for spotting the README error - it's somewhere between a typo and a thinko; the overlay_map file is correct.

I'm happy to see you found your udev issue.

closed 2020-08-28 by amahoneyLIT