raspberrypi/linux #3818: librtlsdr performance issues with 5.x kernels from approx 5.4.42 (USB zero-copy related)

mutability, 2020-08-26:
Background thread: https://discussions.flightaware.com/t/fyi-new-pi-kernel-causes-dump1090-fa-to-run-much-harder/67309

Somewhere between approx kernel 5.4.40 and 5.4.42, some applications that use librtlsdr/lbusb to receive data from USB SDR dongles on Pis started seeing greatly increased CPU use. I tracked this down to a large increase in CPU in a path that essentially does nothing but a simple convert-and-copy loop from the buffers provided by librtlsdr (which in turn come from libusb) into a separate heap buffer. The wallclock and CPU time for that loop increased 35x for no apparent reason. This affects at least Pi 3B+ and Pi 4B, possibly other models.

I suspect the underlying cause is this commit: https://github.com/raspberrypi/linux/commit/0432f7632a24f15332ba36228fe9d055f4a5a771

which fixed a bug in USB zero-copy userspace mappings. Previously, zero-copy mappings were nonfunctional and librtlsdr had a test in place to detect this. With an older kernel, this would disable use of zero-copy mappings. With the new kernel, librtlsdr started to use zero-copy.

_however_

The zero-copy performance is, apparently, terrible. Anything other than a plain `memcpy` out of the buffer goes extremely slowly; I assume this is to do with the cache attributes of the mapping.

In practice this means that users of librtlsdr as distributed in Raspbian have a few options:

1) do not upgrade to the 5.x kernel series;
2) upgrade and pay a much larger CPU cost;
3) upgrade, and rebuild librtlsdr by hand to explicity disable zerocopy;
4) patch the application so that it manually copies data out of the librtlsdr-provided buffer into a heap buffer before touching it further. Yes, this defeats the purpose of the zero-copy mapping. Unfortunately this cost must then be paid regardless of kernel version as there's no practical way to detect if librtlsdr is using a zero-copy buffer or not. Here's an example of such a workaround: https://github.com/flightaware/dump1090/commit/653ad6127a57b88af78b30e9790bb97f80f8eff7

Ideally either we need a kernel that avoids the performance problem somehow; or Raspbian needs to provide rebuilt librtlsdr (or possibly libusb) packages with zero-copy support disabled to avoid the kernel issue.

(FWIW, I tried contacting the author of the kernel commit directly but haven't heard back)

P33M, 2020-09-01:
Coherent mappings effectively bypass all caching on our platform, as there are no DMA peripherals that maintain cache coherence with the ARM L2. 

There are generally two ways in which buffers are passed between non-coherent hardware and CPU - the first uses sync_for_{device|cpu} which performs the individual cache flushing/invalidation required for a CPU to release/claim a buffer, The second uses coherent (aka uncached) memory. mmap'ing a buffer implies that the CPU could read or write it at any time, so you have to use a coherent mapping. 

Memcpy performance is less badly affected because optimised routines are used which fill cachelines efficiently - doing word read/writes will cause cache reads/flushes every single time a load or store is executed.

The correct solution is for librtlsdr to not use zerocopy on non-coherent platforms.

pelwell, 2020-09-01:
Or to perform cache maintenance operations around accesses, which would make it less zero-copy-like.

P33M, 2020-09-01:
I don't think that any of the ARMv7 maintenance operations are useable in unprivileged contexts though...

pelwell, 2020-09-01:
I was thinking to use something like dmabuf to manage the sharing.

mutability, 2020-09-01:
> The correct solution is for librtlsdr to not use zerocopy on non-coherent platforms.

How do you detect a non-coherent platform?

mutability, 2020-09-01:
> mmap'ing a buffer implies that the CPU could read or write it at any time, so you have to use a coherent mapping.

In this specific case, there are well defined points where control of the mmap'd region is handed to or from userspace (e.g. when a URB is submitted or completes); could you do the necessary cache invalidation at those points?

P33M, 2020-09-01:
In the non-mmap case the USB stack downstream of usbfs handles DMA mapping/unmapping automatically (which includes flushing cache before poking controller hardware, and invalidating cache before giving you the buffer back). The necessary copies to/from userspace are done between cacheable buffers so this will always be faster than using DMA-coherent mappings.

The only downside is that latency may increase because you must submit requests in chunks - you only get a callback once an entire buffer has been transferred, but you have control over your buffer sizes so this can create an upper bound on latency.

Does dmabuf allow for more granular control than this?

Why does librtlsdr want to use mmap'd buffers over a enqueue/dequeue dataflow?

> > The correct solution is for librtlsdr to not use zerocopy on non-coherent platforms.
> 
> How do you detect a non-coherent platform?

Not easily. I would guess that ARM platforms with coherent USB controllers are far fewer in number than non-coherent ones, though.

mutability, 2020-09-02:
librtlsdr continuously streams data from a USB bulk IN endpoint, typically at up to ~5MB/s. Latency is not a big issue as this is a receive-only SDR. Similar userspace libraries for other SDR hardware are likely to work in a similar way, and at higher data rates.

The mmap actually happens in libusb's [libusb_dev_mem_alloc](http://libusb.sourceforge.net/api-1.0/group__libusb__asyncio.html#ga47c492e750197f1a28874c7a0dee40e3) which claims to do this:

> Attempts to allocate a block of persistent DMA memory suitable for transfers against the given device. If successful, will return a block of memory that is suitable for use as "buffer" in libusb_transfer against this device. Using this memory instead of regular memory means that the host controller can use DMA directly into the buffer to increase performance, and also that transfers can no longer fail due to kernel memory fragmentation.

> Note that this means you should not modify this memory (or even data on the same cache lines) when a transfer is in progress, although it is legal to have several transfers going on within the same memory block.

The data flow via libusb is the same in both cases: librtlsdr keeps a pool of buffers for USB transfers, allocated either on the heap or via `libusb_dev_mem_alloc`, and uses them to submit URBs to read from the bulk endpoint (`libusb_submit_transfer`). libusb polls for completed URBs and tells librtlsdr when a transfer completes. librtlsdr processes the completed transfer buffer contents (this is the bit that is going slow) and resubmits the transfer. Neither libusb nor librtlsdr touches the buffer before the transfer completes.

So the tl;dr is that librtlsdr tries to use mmap because libusb claims it'll make things go faster / avoid fragmentation. Is libusb getting the wrong end of the stick here?

6by9, 2020-09-02:
dmabufs have the [DMA_BUF_IOCTL_SYNC](https://elixir.bootlin.com/linux/latest/source/include/uapi/linux/dma-buf.h#L41) ioctl that you are obliged to use to denote start and end of CPU access to the buffer. Those then trigger the relevant cache operations to make everything coherent with the dma mappings.