raspberrypi/linux #3819: usb 1-1.2: Not enough bandwidth for new device state.  usb 8 channel soundcard

atuctuc, 2020-08-27:
hallo,

only on the raspberry 4 the  GIGAPort HD+ soudcard does not work properly. 
alsamixer is showing up the sondcard. (raspi 2,3+ it works well!!!) 

aplay stops with the following:

aplay -D plughw:HD hund.wav 
Wiedergabe: WAVE 'hund.wav' : Signed 16 bit Little Endian, Rate: 44100 Hz, stereo
aplay: set_params:1403: Fehler beim Setzen der Hardware-Parameter:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 32
CHANNELS: 2
RATE: 44100
PERIOD_TIME: (125011 125012)
PERIOD_SIZE: 5513
PERIOD_BYTES: 22052
PERIODS: (3 4)
BUFFER_TIME: 500000
BUFFER_SIZE: 22050
BUFFER_BYTES: 88200
TICK_TIME: 0

syslog shows:
Aug 27 15:55:40 raspberrypi kernel: [  953.021599] usb 1-1.2: Not enough bandwidth for new device state.
Aug 27 15:55:40 raspberrypi kernel: [  953.021708] usb 1-1.2: Not enough bandwidth for altsetting 1
Aug 27 15:55:40 raspberrypi kernel: [  953.021727] usb 1-1.2: 1:1: usb_set_interface failed (-28)

any idea?

P33M, 2020-09-01:
Please follow the instructions in the first post of this thread: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=273027

atuctuc, 2020-09-02:
wao, after the pieeprom update, it works,
thank you!!!!

closed 2020-09-02 by atuctuc