raspberrypi/linux #3821: Is 'earlycon' useful on a Pi?

juergh, 2020-08-28:
I've added 'earlycon=pl011,0xfe201000' and now I get the following on the serial console from a Pi 4B rev1.1:

<SNIP>
Read start4.elf bytes  2283936 hnd 0x00011d23 hash ddddf81164f250ad
Read fixup4.dat bytes     5422 hnd 0x0002975a hash fdfbb390f4a2c93c
0x00c03111 0x00000000 0x0000001f
MEM GPU: 76 ARM: 948 TOTAL: 1024
FIXUP src: 128 256 dst: 948 1024
Starting start4.elf @ 0xfec00200

[    2.181617] printk: console [ttyS0] enabled
[    2.190088] printk: bootconsole [pl11] disabled
[    2.199881] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
[    2.208062] mmc-bcm2835 fe300000.mmcnr: mmc_debug:0 mmc_debug2:0
[    2.214203] mmc-bcm2835 fe300000.mmcnr: DMA channel allocated
<SNIP>

pi@rpi-4b-rev1d1-17cf:/boot $ cat /boot/cmdline.txt 
root=PARTUUID=cc55dfa7-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles console=tty1 console=ttyAMA0,115200 earlycon=pl011,0xfe201000

pi@rpi-4b-rev1d1-17cf:~ $ uname -a
Linux rpi-4b-rev1d1-17cf 5.4.59-v8+ #1337 SMP PREEMPT Wed Aug 26 11:10:18 BST 2020 aarch64 GNU/Linux

Logging seems to start late so it behaves more like 'latecon' ;-) Is that expected behavior? Is 'earlycon' even useful on a Pi? Or am I misconfigured somehow?

Thanks
...Juerg

pelwell, 2020-08-28:
Yes, earlycon is useful. The problem is visible in your logging output:
```
[ 2.181617] printk: console [ttyS0] enabled
[ 2.190088] printk: bootconsole [pl11] disabled
```
You're not seeing the output you are expecting because you are using the wrong UART. By default, UART0/ttyAMA0/pl011 is used for Bluetooth, with UART1/ttyS0/mini-8250 being used for the console (if I had a pound for every time I've written that...). 

You need to either add `dtoverlay=disable-bt` to config.txt or use the 8250 for earlycon - `earlycon=uart8250,mmio32,0xfe215040`:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.51-v8+ (dom@buildbot) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.9)) #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe215040 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
...
[    0.012853] printk: console [tty1] enabled
[    0.017016] printk: bootconsole [uart8250] disabled
```

juergh, 2020-08-28:
Thanks for that info, Phil! But you must admit it is convoluted, no? Maybe it would help to add earlycon information to https://www.raspberrypi.org/documentation/configuration/uart.md? Or is it in some other place that I simply haven't found?

pelwell, 2020-08-28:
I thought to expand the documentation following this question.

pelwell, 2020-08-28:
As if by magic, a documentation PR appeared.

juergh, 2020-08-28:
Very nice, thanks!

closed 2020-09-01 by juergh