raspberrypi/linux #3823: 5.4.59-v7l+ #1337 Can't overclock over 2000mhz anymore

myzar, 2020-08-28:
I've upgraded to this kernel hoping to fix my m2 ssd disconnecting randomly introduced in latest stable kernel released few days ago. That is fixed but with this kernel i can't overclock over 2000mhz , anything above that is refused and the cpu is stuck at 600169920 

relevant logs:

[ 1.134005] cpufreq: cpufreq_online: CPU0: Running at unlisted freq: 2100000 KHz
[ 1.134076] cpufreq: cpufreq_online: CPU0: Unlisted initial frequency changed to: 2294967 KHz

[ 1.124136] cpufreq: cpufreq_online: CPU0: Running at unlisted freq: 2147000 KHz
[ 1.124206] cpufreq: cpufreq_online: CPU0: Unlisted initial frequency changed to: 2294967 KHz

pi@retropie:/sys/devices/system/cpu/cpu0/cpufreq $ uname -a
Linux retropie 5.4.59-v7l+ #1337 SMP Wed Aug 26 11:03:21 BST 2020 armv7l GNU/Linux

pi@retropie:/sys/devices/system/cpu/cpu0/cpufreq $ cat ./scaling_available_frequencies
2294967
pi@retropie:/sys/devices/system/cpu/cpu0/cpufreq $ cat ./cpuinfo_max_freq
2294967

pi@retropie:/sys/devices/system/cpu/cpu0/cpufreq $ cat ./scaling_max_freq
2294967
pi@retropie:/sys/devices/system/cpu/cpu0/cpufreq $ cat ./scaling_available_governors
conservative ondemand userspace powersave performance schedutil
pi@retropie:/sys/devices/system/cpu/cpu0/cpufreq $ cat ./scaling_governor
performance
pi@retropie:/sys/devices/system/cpu/cpu0/cpufreq $ vcgencmd measure_clock arm
frequency(48)=600169920

Titaniumtown, 2020-08-28:
I have an issue like this, I just reboot a few times, and it works. It's super weird.

popcornmix, 2020-08-28:
Can you both report output of:
```
vcgencmd version
uname -a
```
and show your modified config.txt settings.
@Titaniumtown are you using PiOS (raspbian) or retropie?

Titaniumtown, 2020-08-28:
I'm using arch linux arm. I switched to kernel 5.8, and it works fine.

myzar, 2020-08-28:
Aug 24 2020 18:50:56
Copyright (c) 2012 Broadcom
version 8d19c1935e67d8ac2fe87c639d4e8b8ef250813b (clean) (release) (start)

Linux retropie 5.4.59-v7l+ #1337 SMP Wed Aug 26 11:03:21 BST 2020 armv7l GNU/Linux

Revision        : c03112

over_voltage=6
v3d_freq=750
arm_freq=2150
avoid_pwm_pll=1
hdmi_enable_4kp60=1
dtoverlay=disable-bt
dtoverlay=disable-wifi

I'm using retropie , it used to happen randomly with the old 5.4.51+ like one in ten , but with this it happens always

Titaniumtown, 2020-08-28:
@popcornmix I can send you the output of those commands later, I am doing school rn.

popcornmix, 2020-08-28:
@myzar  can you confirm if the issue started with the firmware or kernel being updated?
(I don't really know what is in "Linux retropie 5.4.59-v7l+", but I do know what changed in latest firmware).

myzar, 2020-08-28:
@popcornmix I've use rpi-update yesterday i think it updated the kernel only because i see /lib/firmware/raspberrypi/bootloader/beta/pieeprom-2020-07-31.bin . I did that hoping to fix my m2 ssd disconnecting when my wifi dongle was heavily used, and it indeed fixed that but then i noticed the cpu being stuck @600mhz

popcornmix, 2020-08-28:
There's more than one meaning for firmware, but in relation to rpi-update it is the `start*.elf`/`fixup*.dat` files (and version returned from vcgencmd version).
There have been changes related to arm clocking in both firmware and kernel recently.
If uname returns "Linux retropie 5.4.59-v7l+" then rpi-update didn't update the kernel.
are there "kernel=" lines in config.txt?

myzar, 2020-08-28:
@popcornmix it updated kernel for sure , kernel is ver linux-31d197d1b31942d46d882732447485737c69072a i had to use rpi get source to compile the wifi module and the joypad , previous kernel was 5.4.51+ , i've deleted the old dirs manually so i'm 200% sure it updated the kernel

popcornmix, 2020-08-28:
Ah yes, that kernel build number does match rpi-update. Getting thrown off by the retropie hostname.

myzar, 2020-08-28:
to be even more specific

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.59-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1337 SMP Wed Aug 26 11:03:21 BST 2020

Titaniumtown, 2020-08-28:
On kernel 5.8, I just got the same issue. I guess it came back, lol.

popcornmix, 2020-09-01:
rpi-update firmware has a fix for this - please test.

Titaniumtown, 2020-09-01:
Ok, thanks! When I get time to test, I will make sure to! :)

myzar, 2020-09-01:
fixed  vcgencmd measure_clock arm                        frequency(48)=2150507776

Titaniumtown, 2020-09-01:
@myzar reboot a few times and make sure, bc before this firmware update (haven't tested yet) it as a random thing.

Titaniumtown, 2020-09-20:
I am still getting this issue, this is super annoying.

popcornmix, 2020-09-21:
@Titaniumtown I believe this issue is fixed.
Are you using initial_turbo? Might this be your issue? https://github.com/raspberrypi/firmware/issues/1469

Titaniumtown, 2020-09-21:
I am not using initial_turbo

popcornmix, 2020-09-21:
You'll have to explain your issue. The issue reported here has been confirmed fixed by the author.
Your use case is different and you haven't explained it. 

Ideally show the minimum changes to config.txt and commands run to show the issue in a standard RPiOS image.