raspberrypi/linux #3825: 100baseT ethernet controller support

benjamin2044, 2020-09-01:
We asked whether RaspberryPi can include driver support for ENC424J600 as it is already supporting ENC28J600. The only problem is speed ENC28j600 runs at 10mbps whereas ENC424j600 runs at 100mbps.
We had already started the discussion on the topic, and the solution provided was to modify the kernel. However, due to the complexity of modifying the kernel we were advised to create an issue linking to the following thread.
https://www.raspberrypi.org/forums/viewtopic.php?t=180491
Is it possible that RaspberryPi can include driver support for ENC424J600 in the future, so that the users can just go to /boot/config.txt and just add a line dtoverlay=enc424j600.
[https://www.raspberrypi.org/forums/viewtopic.php?t=180491](url)

pelwell, 2020-09-01:
I can provide an overlay that might work, but I have no way of testing it; in particular I'm not sure how fast we can push the SPI bus before it becomes unreliable - 50MHz or so. You will have to build your own kernel and test it yourself - the instructions are here: https://www.raspberrypi.org/documentation/linux/kernel/building.md
And the guide to customising the configuration (you'll need to add `CONFIG_ENCX24J600=m`) is here: https://www.raspberrypi.org/documentation/linux/kernel/configuring.md

If you feel up to that, the overlay is probably going to look like this:
```
// Overlay for the Microchip ENCX24J600 Ethernet Controller
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			status = "okay";

			eth1: encx24j60@0{
				compatible = "microchip,encx24j600";
				reg = <0>; /* CE0 */
				pinctrl-names = "default";
				pinctrl-0 = <&eth1_pins>;
				interrupt-parent = <&gpio>;
				interrupts = <25 0x2>; /* falling edge */
				spi-max-frequency = <50000000>;
				status = "okay";
			};
		};
	};

	fragment@1 {
		target = <&spidev0>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@2 {
		target = <&gpio>;
		__overlay__ {
			eth1_pins: eth1_pins {
				brcm,pins = <25>;
				brcm,function = <0>; /* in */
				brcm,pull = <0>; /* none */
			};
		};
	};

	__overrides__ {
		int_pin = <&eth1>, "interrupts:0",
		          <&eth1_pins>, "brcm,pins:0";
		speed   = <&eth1>, "spi-max-frequency:0";
	};
};
```

bcecchinato, 2021-07-21:
@pelwell Thanks for the fragment ;) We are also using a CM3+ in our projet with a slightly different dts :
```
// Overlay for the Microchip encx24j600 Ethernet Controller
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2708";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			status = "okay";

			eth1: encx24j600@0{
				compatible = "microchip,encx24j600";
				reg = <0>; /* CE0 */
				pinctrl-names = "default";
				pinctrl-0 = <&eth1_pins>;
				interrupt-parent = <&gpio>;
				interrupts = <25 0x2>; /* falling edge */
				spi-max-frequency = <12000000>;
				status = "okay";
			};
		};
	};

	fragment@1 {
		target = <&spidev0>;
		__overlay__ {
			status = "disabled";
		};
	};

	fragment@2 {
		target = <&gpio>;
		__overlay__ {
			eth1_pins: eth1_pins {
				brcm,pins = <25>;
				brcm,function = <0>; /* in */
				brcm,pull = <0>; /* none */
			};
		};
	};

	__overrides__ {
		int_pin = <&eth1>, "interrupts:0",
		          <&eth1_pins>, "brcm,pins:0";
		speed   = <&eth1>, "spi-max-frequency:0";
	};
};
```

I'm able to perform some tests easily to push the SPI. I've set `speed=50000000` in /boot/config.txt, the eth interface is not mounting. I've tried with `speed=24000000`, the interface is available (the value is totally arbitrary, i have no clue on what I can use).

Let me know if you I can help (avoiding to rebuild the kernel for this module would be a huge step for us) ;)

benjamin2044, 2021-07-22:
I tried but failed.
Try to follow this excellent tutorial.
https://blog.stabel.family/raspberry-pi-4-device-tree/

and also this.
https://www.cnblogs.com/qiengo/p/5915508.html