raspberrypi/linux #3826: Kernel logo

spaccabbomm, 2020-09-01:
Hi All,
after a firmware update that jump kernel from 4.9.40 to 5.4.58 now i dont see the kernel logo even if it is enabled in kernel config.
any help?

popcornmix, 2020-09-01:
Are you talking about the 4 (or one on pi0/pi1) raspberries?

spaccabbomm, 2020-09-01:
Working with Compute module 3

pelwell, 2020-09-01:
So which platform(s) is it not working with?

spaccabbomm, 2020-09-01:
The platform is raspberry CM3 with raspbian jessie. i've a customized boot kernel (4.9.40) logo that is visible, now after rpi-update i have the kernel 5.4.y but now i cant see any logo. now i should see the 4 raspberry icon but dont.

popcornmix, 2020-09-01:
On a Pi4, booting to console I get raspberries for the initial part of boot (using the dumb framebuffer) and they disappear when fkms switches in.

@spaccabbomm are you booting to console or desktop. Have you enabled fkms/kms? Any non standard settings in config.txt or cmdline.txt?

spaccabbomm, 2020-09-01:
I'm booting on desktop, here is my cmdline.txt 

dwc_otg.lpm_enable=0 console=serial0,115200  console=tty3 root=PARTUUID=fa46bcd2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait loglevel=3 quiet splash plymouth.ignore-serial-consoles vt.global_cursor_default=0 dwc_otg.microframe_schedule=1 sdhci-bcm2708.enable_llm=1 dwc_otg.speed=1 smsc95xx.turbo_mode=N

here the config.txt

# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
gpu_mem=256
disable_splash=1
display_rotate=1
lcd_rotate=1
dtoverlay=i2c-rtc,pcf8523

dtparam=act_led_trigger=mmc0
start_debug=1

# Enabling watchdog.
dtparam=watchdog=on

spaccabbomm, 2020-09-01:
Here is my boot video. the "loading" text is the kernel logo.

http://mutech.homepc.it/rpi.mp4

BigMuscle85, 2020-10-21:
Since Linux kernel 5, the boot logo is not displayed when "quiet" boot is enabled. see commit 10993504d647356196a04b3022d645ec92e00159

spaccabbomm, 2020-10-21:
Now i get the logo after set "loglevel=5"