raspberrypi/linux #3828: Unknown dtparam 'eth_led0'/'eth_led1' - ignored

luskaner, 2020-09-01:
**Describe the bug**
Unable to set the `dtparam` flags `eth_led0` and `eth_led1` following the instructions of [README](https://raw.githubusercontent.com/raspberrypi/firmware/master/boot/overlays/README) in order to disable the amber and green leds of network ethernet.

**To reproduce**

1. Set the following flags in in `/boot/config.txt`:
 * `dtparam=eth_led0=14`
 * `dtparam=eth_led1=14`
2. Reboot the machine

**Expected behaviour**

- The flags are recognized and the ethernet leds are disabled
- `vcdbg log msg` shows:
```
001717.026: dtparam: eth_led0=14
001719.949: Unknown dtparam 'eth_led0' - ignored
001719.980: dtparam: eth_led1=14
001722.858: Unknown dtparam 'eth_led1' - ignored
```

**Actual behaviour**
The flag is ignored and behaviour of ethernet leds is unchanged.

**System**
**[Full Pastebin](https://pastebin.com/3Zhc98G9)**

* Which model of Raspberry Pi? 
**Raspberry Pi 3 Model B**
* Which OS and version (`cat /etc/rpi-issue`)? 
```
Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Aug 19 2020 17:40:15
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
`Linux raspberrypi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux`

**Logs**
_See System -> Full Pastebin_

**Additional context**
_None_

pelwell, 2020-09-01:
The README says:
```
        eth_led0                Set mode of LED0 - amber on Pi3B+ (default "1"),
                                green on Pi4 (default "0").
                                The legal values are:

                                Pi3B+

                                0=link/activity          1=link1000/activity
                                2=link100/activity       3=link10/activity
                                4=link100/1000/activity  5=link10/1000/activity
                                6=link10/100/activity    14=off    15=on

                                Pi4

                                0=Speed/Activity         1=Speed
                                2=Flash activity         3=FDX
                                4=Off                    5=On
                                6=Alt                    7=Speed/Flash
                                8=Link                   9=Activity
```
This parameter does not exist for the 3B.

luskaner, 2020-09-02:
Nevermind @pelwell missed the '+' I thought this would work, thanks!

closed 2020-09-02 by luskaner