raspberrypi/linux #3829: 5.4.61-v7+ #1339 I2C is unusable

zorbasoft, 2020-09-02:
I present the configuration that **worked before the update**
*config.txt*
> dtparam=i2c_vc=off
> dtparam=i2c_arm=on
> dtoverlay=i2c1-bcm2708,sda1_pin=44,scl1_pin=45,pin_func=6

everything works **before** the update:
```
~ $ raspi-gpio get
GPIO 2: level=1 fsel=0 func=INPUT
GPIO 3: level=1 fsel=0 func=INPUT
GPIO 44: level=1 fsel=6 alt=2 func=SDA1
GPIO 45: level=1 fsel=6 alt=2 func=SCL1
```
problem **after** the update:
```
GPIO 2: level=1 fsel=4 alt=0 func=SDA1
GPIO 3: level=1 fsel=4 alt=0 func=SCL1
GPIO 44: level=1 fsel=0 func=INPUT
GPIO 45: level=1 fsel=0 func=INPUT
```
**I can't turn off**  I2C on GPIO 2-3
and **can't turn on** I2C on GPIO 44-45

any solution ?

pelwell, 2020-09-02:
Which platform is this? Why are you making that change?

6by9, 2020-09-02:
Is this potentially a collision with the i2c0_mux on 44&45? That shouldn't be enabled when i2c_vc=off, but it's possible.

Which kernel version are you counting as "before the update"?

There's also the issue that ```i2c1-bcm2708.dtbo``` won't exist on the 5.4 branch.
As noted in the [README](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/README#L1374)
```
Name:   i2c1-bcm2708
Info:   Deprecated, legacy version of i2c1.
Load:   <Deprecated>
```

zorbasoft, 2020-09-03:
I have raspi **CM3** with Raspbian  (buster).
Up to version **4.19.97-v7+** all working properly.
After updating `aptitude upgrade`  to version 5.4.61-v7+ it was not working.

important information:
- config.txt is same as before upgrading
- all required *modules* are loaded `i2c-dev` and `i2c-bcm2708`
- `i2c1-bcm2708.dtbo` and others exists in *overlays*

Regardless of configuration `dtoverlay=i2c1-bcm2708,sda1_pin=44,scl1_pin=45,pin_func=6`,
I'm more concerned that doesn't work turning off `dtparam=i2c_vc=off` !

pelwell, 2020-09-03:
As @6by9 said, the `i2c1-bcm2708` overlay should not be used any more:
```
pi@raspberrypi:~$ dtoverlay -h i2c1-bcm2708
Name:   i2c1-bcm2708

Info:   Deprecated, legacy version of i2c1.

Usage:  <Deprecated>
```
You should be using the i2c1 overlay:
```
pi@raspberrypi:~$ dtoverlay -h i2c1
Name:   i2c1

Info:   Change i2c1 pin usage. Not all pin combinations are usable on all
        platforms - platforms other then Compute Modules can only use this
        to disable transaction combining.

Usage:  dtoverlay=i2c1,<param>=<val>

Params: pins_2_3                Use pins 2 and 3 (default)
        pins_44_45              Use pins 44 and 45
        combine                 Allow transactions to be combined (default
                                "yes")
```
In your case that means:
```
dtoverlay=i2c1,pins_44_45
```

zorbasoft, 2020-09-03:
Thank you @6by9 @pelwell for support.
*Deprecated* :)
Everything is back to normal.
Best regards

zorbasoft, 2020-09-03:
all ok

closed 2020-09-03 by zorbasoft