raspberrypi/linux #3831: GPT SD card booting fails with 5.4 kernel

reraikes, 2020-09-03:
Booting an SD card which has a GPT partition table fails with the 5.4 kernel (2020-08-20-raspios-buster-armhf-lite.img).  A Hybrid MBR is present for the boot partition to allow booting on the Raspberry Pi 3B+.

The same installation boots properly with the 4.19 kernel (2020-05-27-raspios-buster-lite-armhf.img) until you do an apt update/full-upgrade which upgrades the kernel to 5.4.

The problem is equally reproducible on a Raspberry Pi 4 or a Raspberry Pi 3B+.

With the 4.19 kernel, booting completes successfully.

With the 5.4 kernel, the appropriate start.elf is found, loaded, and executed without error, but shortly thereafter, booting halts with the green led repeatedly blinking 7 times.

The same installation on USB devices boots properly with either the 4.19 kernel or the 5.4 kernel.

Simply replacing the 5.4 start.elf files with these 4.19 start.elf files makes the GPT SD card boot properly:

bootcode.bin      52,480   06/06/2020 22:38
fixup.dat          6,746   06/06/2020 22:38
fixup4.dat         5,405   06/06/2020 22:38
fixup4cd.dat       3,146   06/06/2020 22:38
fixup4db.dat       8,417   06/06/2020 22:38
fixup4x.dat        8,419   06/06/2020 22:38
fixup_cd.dat       2,663   06/06/2020 22:38
fixup_db.dat       9,820   06/06/2020 22:38
fixup_x.dat        9,818   06/06/2020 22:38
start.elf      2,884,708   06/06/2020 22:38
start4.elf     2,272,992   06/06/2020 22:38
start4cd.elf     816,124   06/06/2020 22:38
start4db.elf   3,774,532   06/06/2020 22:38
start4x.elf    3,031,652   06/06/2020 22:38
start_cd.elf     694,052   06/06/2020 22:38
start_db.elf   4,861,512   06/06/2020 22:38
start_x.elf    3,799,144   06/06/2020 22:38

It appears something got broken in the 5.4 start.elf files that only affects GPT SD card booting.

timg236, 2020-09-03:
Sound like a firmware rather than a linux issue (wrong repo), a description of the first / last rpi-update releases which for this would help

reraikes, 2020-09-03:
Sorry if I'm in the wrong place.  I'll repost elsewhere if you need me to (just point me there please).

I don't know the exact release where the problem began,  I suppose I could go through the list of rpi-update hashes trying them one at a time to isolate the exact point it broke.  The files listed above are the ones used during the early RPi 4 MSD boot testing.

timg236, 2020-09-03:
I think we could probably get by with full details of the partition table.

Before 06/06/2020 the never parsed GPT partitions so there was no support for GPT. Hybrid may have worked because the start.elf ignored the GPT part and failed over to the next primary/hybrid partition.
Now, it will find the GPT partition table and if it doesn't find a FAT/EFI partition it might stop.

reraikes, 2020-09-03:
I just regressed to 4.19.108 with rpi-update and it booted fine.  I'll see if I can home in on the exact place it broke.

In the meantime, here is the partition table info:

Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 3864BC3D-2CE2-4EDA-821A-C21C60699167

Device            Start      End  Sectors  Size Type
/dev/mmcblk0p1     8192   532479   524288  256M Microsoft basic data
/dev/mmcblk0p2   532480 21503999 20971520   10G Linux filesystem

Disk /dev/mmcblk0: 29 GiB, 31104958464 bytes, 60751872 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device         Boot Start    End Sectors  Size Id Type
/dev/mmcblk0p1       8192 532479  524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2          1   8191    8191    4M ee GPT