raspberrypi/linux #3834: Kernel 5.9.0-rc3 compile issues

Dark-Sky, 2020-09-04:
Using pi4 8G compiling native with gcc 10.2.0 on 64bit OS. Used make bcm2711_defconfig to generate the .config I get this error and the compile aborts:

```
sound/soc/codecs/i-sabre-codec.c:273:18: error: initialization of ‘int (*)(struct snd_soc_dai *, int,  int)’ from incompatible pointer type ‘int (*)(struct snd_soc_dai *, int)’ [-Werror=incompatible-pointer-types]
  273 |  .mute_stream  = i_sabre_codec_dac_mute,
      |                  ^~~~~~~~~~~~~~~~~~~~~~
sound/soc/codecs/i-sabre-codec.c:273:18: note: (near initialization for ‘i_sabre_codec_dai_ops.mute_stream’)
cc1: some warnings being treated as errors
```
I have gotten "cc1: some warnings being treated as errors" in the past due to gcc version issues but I compiled the i-sabre modules yesterday with the same gcc 10.2.0 yesterday with kernel 5.8.5 and had no issues.

I disabled CONFIG_SND_BCM2708_SOC_I_SABRE_Q2M & CONFIG_SND_SOC_I_SABRE_CODEC modules and the kernel compiled ok but I did notice some "-z norelro" warnings I have not seen before between the first and second stage of the module building:

```
  LD      .tmp_vmlinux.kallsyms1
ld: warning: -z norelro ignored
  KSYM    .tmp_vmlinux.kallsyms1.o
  LD      .tmp_vmlinux.kallsyms2
ld: warning: -z norelro ignored
  KSYM    .tmp_vmlinux.kallsyms2.o
  LD      vmlinux
ld: warning: -z norelro ignored
  SORTTAB vmlinux
  SYSMAP  System.map
```

pelwell, 2020-09-04:
Fixed by bb8ca0e (until the next rebase).

popcornmix, 2020-09-04:
Be aware that 5.9 has a lot of conflicts with downstream (as partial commits are appearing upstream).
It's in a state where you should be able to boot with working display, network and usb.
But kms, codecs, camera, isp, vcsm are still being worked on and may not be functional currently.

Dark-Sky, 2020-09-04:
Thanks a million. It compiles now.

closed 2020-09-04 by pelwell