raspberrypi/linux #3835: i2s pcm1794 support for 384kHz sample rate

bitkeeper, 2020-09-04:
In 2017 the i2s driver `rate_max` is increased from 192000 to 384000Hz.
Despite the pcm1794 also supports 384kHz, the codec pcm1794a is never updated to reflect this.

Is there a specific reason for, or could the the codec be updated to support 384kHz ?

Matching patch:
```
diff --git a/sound/soc/codecs/pcm1794a.c b/sound/soc/codecs/pcm1794a.c
index 36b5b6c26655..271e79feba6b 100644
--- a/sound/soc/codecs/pcm1794a.c
+++ b/sound/soc/codecs/pcm1794a.c
@@ -26,7 +26,7 @@ static struct snd_soc_dai_driver pcm1794a_dai = {
        .playback = {
                .channels_min = 2,
                .channels_max = 2,
-               .rates = SNDRV_PCM_RATE_8000_192000,
+               .rates = SNDRV_PCM_RATE_8000_384000,
                .formats = SNDRV_PCM_FMTBIT_S16_LE |
                           SNDRV_PCM_FMTBIT_S24_LE
        },
```

pelwell, 2020-09-04:
Have you tested this yourself?

bitkeeper, 2020-09-04:
From my side it is more a question. 
I only have tested that it currently isn't support:

It concerns a DDDAC 1794 NOS dac. No digital filters are enabled.
It seems that some have it working at that rate with other i2s sources (https://www.diyaudio.com/forums/digital-line-level/224108-nos-192-24-dac-pcm1794-waveio-usb-input-712.html#post5909393)

pelwell, 2020-09-04:
It's really easy to build your own kernel - follow our instructions here: https://www.raspberrypi.org/documentation/linux/kernel/building.md

HiassofT, 2020-09-04:
The [PCM1794 datasheet](https://www.ti.com/lit/gpn/pcm1794) states the supported samplerates are 10-200kHz.

What makes you think exceeding the specs by almost 100% won't result in (possibly severely) degraded performance?

pelwell, 2020-09-04:
That does look like a stretch.

@bitkeeper What makes you say "the pcm1794 also supports 384kHz"?

bitkeeper, 2020-09-04:
Thanks, the proof is in the pudding so I will try to build the kernel and report back for better or worse .

The spec is applies to 8x oversampling with the digital filter.
In this case the PCM1974A is used in a gray array; 8x oversampling and digital filters are off. And the system clock supports the higher rates. I think that is why others succeeded with other i2s sources.

In the end I would like to have the following audio chain: Raspberry Pi >i2s > Fifopi >i2s> Dac.
Digital Filtering is then moved to the software on the pi.

pelwell, 2020-09-04:
Unfortunately, like the baking of puddings, IC manufacture does not produce precisely the same results every time. Designers have to over-engineer/under-sell parts so that parts at the extremes of the range still function correctly.

bitkeeper, 2020-09-07:
Tested the custom kernel with [moOde](https://moodeaudio.org/) and it works flawless:
[image]

vinhx, 2021-01-05:
I'm very new in play with i2s interfave, there big question with me how to wire pcm1794 i2s pin to Raspberry pi, may you give me your schematic that you using, thank alot